Xilinx-CNS/onload #6: CentOS 8.3.2011 build is broken

galuha, 2020-12-18:
Hello! 

System: CentOS 8.3.2011
To reproduce:
1 Install the latest CentOS
2 Clone this repo
3 Run scripts/onload_install

Please see this log:

/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c: In function ‘efx_tc_indr_setup_cb’:
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:4776:15: error: implicit declaration of function ‘flow_indr_block_cb_alloc’; did you mean ‘flow_indr_block_call’? [-Werror=implicit-function-declaration]
    block_cb = flow_indr_block_cb_alloc(efx_tc_block_cb, binding,
               ^~~~~~~~~~~~~~~~~~~~~~~~
               flow_indr_block_call
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:4776:13: warning: assignment to ‘struct flow_block_cb *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
    block_cb = flow_indr_block_cb_alloc(efx_tc_block_cb, binding,
             ^
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:4804:4: error: implicit declaration of function ‘flow_indr_block_cb_remove’; did you mean ‘flow_indr_block_cb_register’? [-Werror=implicit-function-declaration]
    flow_indr_block_cb_remove(block_cb, tcb);
    ^~~~~~~~~~~~~~~~~~~~~~~~~
    flow_indr_block_cb_register
  CC [M]  /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/farch.o
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c: In function ‘efx_init_tc’:
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:4951:30: error: passing argument 1 of ‘flow_indr_dev_register’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  rc = flow_indr_dev_register(efx_tc_indr_setup_cb, efx);
                              ^~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/sch_generic.h:21,
                 from ./include/linux/filter.h:25,
                 from ./include/net/sock.h:64,
                 from ./include/net/inet_sock.h:26,
                 from ./include/net/ip.h:31,
                 from /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.h:60,
                 from /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:12:
./include/net/flow_offload.h:539:55: note: expected ‘int (*)(struct net_device *, void *, enum tc_setup_type,  void *)’ but argument is of type ‘int (*)(struct net_device *, void *, enum tc_setup_type,  void *, void *, void (*)(struct flow_block_cb *))’
 int flow_indr_dev_register(flow_indr_block_bind_cb_t *cb, void *cb_priv);
                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c: In function ‘efx_fini_tc’:
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:4972:28: error: passing argument 1 of ‘flow_indr_dev_unregister’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   flow_indr_dev_unregister(efx_tc_indr_setup_cb, efx, efx_tc_block_unbind);
                            ^~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/sch_generic.h:21,
                 from ./include/linux/filter.h:25,
                 from ./include/net/sock.h:64,
                 from ./include/net/inet_sock.h:26,
                 from ./include/net/ip.h:31,
                 from /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.h:60,
                 from /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:12:
./include/net/flow_offload.h:540:58: note: expected ‘int (*)(struct net_device *, void *, enum tc_setup_type,  void *)’ but argument is of type ‘int (*)(struct net_device *, void *, enum tc_setup_type,  void *, void *, void (*)(struct flow_block_cb *))’
 void flow_indr_dev_unregister(flow_indr_block_bind_cb_t *cb, void *cb_priv,
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:4972:55: error: passing argument 3 of ‘flow_indr_dev_unregister’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   flow_indr_dev_unregister(efx_tc_indr_setup_cb, efx, efx_tc_block_unbind);
                                                       ^~~~~~~~~~~~~~~~~~~
In file included from ./include/net/sch_generic.h:21,
                 from ./include/linux/filter.h:25,
                 from ./include/net/sock.h:64,
                 from ./include/net/inet_sock.h:26,
                 from ./include/net/ip.h:31,
                 from /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.h:60,
                 from /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:12:
./include/net/flow_offload.h:541:27: note: expected ‘int (*)(enum tc_setup_type,  void *, void *)’ but argument is of type ‘void (*)(void *)’
          flow_setup_cb_t *setup_cb);
          ~~~~~~~~~~~~~~~~~^~~~~~~~
  CC [M]  /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/siena.o
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c: In function ‘efx_tc_netdev_event’:
/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.c:5017:10: error: passing argument 3 of ‘__flow_indr_block_cb_register’ from incompatible pointer type [-Werror=incompatible-pointer-types]
          efx_tc_indr_setup_cb, efx);
          ^~~~~~~~~~~~~~~~~~~~
In file included from ./include/net/sch_generic.h:21,
[... 34 lines cut ...]
  CC [M]  /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/aoe.o
  CC [M]  /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/debugfs.o
  CC [M]  /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/dump.o
  CC [M]  /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/driverlink.o
  LD [M]  /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/sfc_driverlink.o
cc1: some warnings being treated as errors
make[8]: *** [scripts/Makefile.build:315: /root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc/tc.o] Error 1
make[7]: *** [Makefile:1544: _module_/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc] Error 2
make[7]: Leaving directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
make[6]: *** [/root/onload/onload/src/driver/linux_net/drivers/net/ethernet/sfc/Makefile:260: modules] Error 2
make[6]: Leaving directory '/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet/sfc'
make[5]: *** [/root/onload/onload/src/driver/linux_net/drivers/net/ethernet/mmake.mk:8: all] Error 2
make[5]: Leaving directory '/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net/ethernet'
make[4]: *** [/root/onload/onload/src/driver/linux_net/drivers/net/mmake.mk:8: all] Error 2
make[4]: Leaving directory '/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers/net'
make[3]: *** [/root/onload/onload/src/driver/linux_net/drivers/mmake.mk:8: all] Error 2
make[3]: Leaving directory '/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net/drivers'
make[2]: *** [/root/onload/onload/src/driver/linux_net/mmake.mk:8: all] Error 2
make[2]: Leaving directory '/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver/linux_net'
make[1]: *** [/root/onload/onload/src/driver/mmake.mk:19: all] Error 2
make[1]: Leaving directory '/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64/driver'
make: *** [/root/onload/onload/src/mmake.mk:25: all] Error 2
make: Leaving directory '/root/onload/onload/build/x86_64_linux-4.18.0-240.1.1.el8_3.x86_64'
onload_build: ERROR: Failed to build driver components.
onload_install: ERROR: Build failed.  Not installing.

shirshen12, 2020-12-18:
You need to install `kernel-devel` and `kernel-headers` package. Make sure to match them exactly with kernel source, else it wont compile.

galuha, 2020-12-18:
I did it. In other case it would not’ve built at all, would have complained
in the beginnin

shirshen12, 2020-12-18:
Can you post the details of NiC and its driver ?

galuha, 2020-12-18:
The card is SF10-050020

galuha, 2020-12-18:
Just found out that onload is not supported by this card. But drivers still should build

galuha, 2020-12-18:
How do i actually detect my card model by the look of it and from the console? Never could figure it out actually with SF cards. 

From lspci perspective they are mostly the same:
# lspci | grep Sol
81:00.0 Ethernet controller: Solarflare Communications SFC9220 10/40G Ethernet Controller (rev 02)
81:00.1 Ethernet controller: Solarflare Communications SFC9220 10/40G Ethernet Controller (rev 02)

# lspci | grep Sol
01:00.0 Ethernet controller: Solarflare Communications SFC9250 10/25/40/50/100G Ethernet Controller (rev 01)
01:00.1 Ethernet controller: Solarflare Communications SFC9250 10/25/40/50/100G Ethernet Controller (rev 01)

And on the card that is in my hands now it is written: Solarflare 2016 SF10-050020
But the seller states that it is SFN8522PLUS

How can i be sure?

abower-xilinx, 2020-12-21:
Hi @galuha, thanks for trying out this development branch! Please contact support-nic@xilinx.com or use resources at https://support-nic.xilinx.com/ if you need support for your Solarflare adapter or to download supported versions of Onload.

abower-xilinx, 2020-12-21:
Thanks for raising the build error, which is fixed by e9d90b290e498c8aa9073b03ebdce52b1e17b732.

closed 2020-12-21 by abower-xilinx

abower-xilinx, 2020-12-21:
Andrei,

> lspci | grep Sol
> 81:00.0 Ethernet controller: Solarflare Communications SFC9220 10/40G Ethernet Controller (rev 02)
> 81:00.1 Ethernet controller: Solarflare Communications SFC9220 10/40G Ethernet Controller (rev 02)

This output gives the ASIC details not the NIC model. If you use `lspci -v -d1924:` then you can see the adapter type after 'Subsystem'. You can also see this information in `dmesg` when the `sfc` driver initializes. The `sfreport.pl` tool from https://support-nic.xilinx.com/wp/drivers is useful for extracting more information about the networking subsystem or when reporting issues with supported hardware or software.

galuha, 2020-12-21:
Thank you very much,
Now i can see that my card is:
Subsystem: Solarflare Communications SFN8522-R2 8000 Series 10G Adapter