Xilinx-CNS/onload #7: sfc driver does not load in a bare metal with i40e driver

shirshen12, 2021-01-02:
Hello @maciejj-xilinx 

I have been trying to get Onload working on a: 

40GBe NiC (X710, 4 x 10 bonded)
i40e driver, version: 2.8.20-k (its a stock version, same issue with 2.13 i40e driver update as well)
64 core Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz
Centos 8, 4.18.0-240.1.1.el8_3.x86_64

So onload compiles well, but when I go to reload it, I see this error:
[image]

Please see dmesg as well below:
[image]

Any help is appreciated.

maciejj-xilinx, 2021-01-07:
I have not been able to reproduce this issue with rhel8.3 4.18.0-240.el8.x86_64 kernel and fad45221166929c0467deed6f8b1927618948552
I have not tried centos 8 4.18.0-240.1.1.el8_3.x86_64 specifically.

It looks like dependency on mtd module. Is mtd module inseted, would `modprobe mtd` help?

shirshen12, 2021-01-10:
Hi @maciejj-xilinx It did not help `modprobe mtd` also results in the below error:
`[root@shir-bare01 ~]# modprobe mtd`
`modprobe: FATAL: Module mtd not found in directory /lib/modules/4.18.0-240.1.1.el8_3.x86_64`

I am not certain why would we need a Memory Tech Device for sfc driver. Else this should have also affected _ixgbe_ based system as well.

Are you reproducing the error on a _i40e_ based system ?

maciejj-xilinx, 2021-01-11:
Thanks for trying.
It looks unfortunately to be kernel compatibility issue of our driver suite. I have raised the issue internally.

shirshen12, 2021-01-17:
Hi @maciejj-xilinx ,

I tried to make i40e driver with onload and the same issue persists as before. I checked the commit @abower-xilinx made and hence made an attempt.

This is an FYI

shirshen12, 2021-01-27:
Hello @maciejj-xilinx ,

I have tried to compile the latest Onload on top of 140e driver again and it looks like we are still hitting the same issue, as described in this ticket.

Can you please let me know the status of the issue you have raised internally ?

maciejj-xilinx, 2021-02-03:
I have raised the priority of the issue. We depend on other internal team to have this fixed.

shirshen12, 2021-02-16:
Hello @maciejj-xilinx any updates on this issue ? Just checking, if any updates, I will test it again.

maciejj-xilinx, 2021-02-25:
Hey @shirshen12 

We can reproduce the issue with 4.18.0-240.15.1.el8_3.x86_64 on rhel8.3
It is not fixed yet unfornately.

Regards,
Maciej

shirshen12, 2021-02-26:
Thanks for sharing the exact kernel version, can you also share the exact driver version @maciejj-xilinx for i40e ? I am updating the driver version to obtain zerocopy support, are you also updating the driver ?

maciejj-xilinx, 2021-03-03:
My earlier survey of the standalone i40 driver code brought me to believe that zerocopy gets disabled when built against redhat/centos 8 kernels. The distro driver code did not seem to suffer from this issue.

shirshen12, 2022-06-10:
Closing this ticket since i40e driver is compatible now with Onload.

closed 2022-06-10 by shirshen12