Xilinx-CNS/onload #9: unable to load onload module on Ubuntu20.10

ShayAgros, 2021-01-18:
Hi, first of all thank you for sharing this code.

I'm trying to run on a virtual machine for some testing but not able to load onload driver to the kernel.

I use stock Ubuntu 20.10 distribution and I followed the steps in [DEVELOPING.md](https://github.com/Xilinx-CNS/onload/blob/master/DEVELOPING.md). Namely these commands:
```
export PATH="$PWD/scripts:$PATH"

mmakebuildtree --driver
mmakebuildtree --user

make -C "$(mmaketool --toppath)/build/$(mmaketool --driverbuild)"
make -C "$(mmaketool --toppath)/build/$(mmaketool --userbuild)"

"$(mmaketool --toppath)/build/$(mmaketool --driverbuild)/driver/linux/load.sh" onload
```

The output I keep getting is
```
unload.sh: /sbin/rmmod virtual_bus
unload.sh: /sbin/rmmod sfc_driverlink
NET_OPT is
CHAR_OPT is
modprobe: FATAL: Module mtdchar not found in directory /lib/modules/5.8.0-38-generic
ERROR: Did not find sfc_control in /proc/devices
load.sh: no Solarflare NICs detected in this machine
```

While it makes sense that no Solarflare NIC is detected, a module called mtdchar doesn't seem to exist based on what I see in the kernel source.

Same behaviour was observed on Ubuntu 20.04. Any guesses to what am I doing wrong ?

maciejj-xilinx, 2021-01-28:
Hello,
It looks load.sh script does not work nicely without Solarflare/Xilinx NICs. We'll have a patch shortly.

I suggest trying customer build/install script:
`
scripts/onload_install
onload_tool reload
`

ShayAgros, 2021-02-02:
@maciejj-xilinx Thanks (: It works now.

Tested on both Ubuntu 20.04 and 20.10

closed 2021-02-02 by ShayAgros