Xilinx-CNS/onload #11: Driver code build issue on Ubuntu 18.04.5 LTS

cpaolini, 2021-02-21:
make -C "$(mmaketool --toppath)/build/$(mmaketool --driverbuild)"
.
.
.
/home/paolini/onload/build/x86_64_linux-4.15.0-135-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.c: In function ‘flow_rule_alloc’:
/home/paolini/onload/build/x86_64_linux-4.15.0-135-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.c:642:17: error: implicit declaration of function ‘struct_size’; did you mean ‘bd_set_size’? [-Werror=implicit-function-declaration]
  rule = kzalloc(struct_size(rule, action.entries, num_actions),
                 ^~~~~~~~~~~
                 bd_set_size
/home/paolini/onload/build/x86_64_linux-4.15.0-135-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.c:642:35: error: ‘action’ undeclared (first use in this function); did you mean ‘tc_action’?
  rule = kzalloc(struct_size(rule, action.entries, num_actions),
                                   ^~~~~~
                                   tc_action
/home/paolini/onload/build/x86_64_linux-4.15.0-135-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.c:642:35: note: each undeclared identifier is reported only once for each function it appears in
cc1: all warnings being treated as errors
scripts/Makefile.build:330: recipe for target '/home/paolini/onload/build/x86_64_linux-4.15.0-135-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.o' failed
make[8]: *** [/home/paolini/onload/build/x86_64_linux-4.15.0-135-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.o] Error 1

maciejj-xilinx, 2021-02-26:
Hi cpaolini,

Thank you for raising this.

The fix for the issue has been identified and we are hoping to be able to incorporate it in matter of days.

Regards,
Maciej Jablonski

maciejj-xilinx, 2021-03-03:
Hi,

The fixed has been incorporated,

Regards,
Maciej Jablonski

closed 2021-03-03 by maciejj-xilinx

cpaolini, 2021-03-03:
Hi, 

I just pulled all updates, but I still get an issue with implicit declaration of function struct_size in kernel_compat.c:

  CC [M]  /home/paolini/onload/build/x86_64_linux-4.15.0-136-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.o
/home/paolini/onload/build/x86_64_linux-4.15.0-136-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.c: In function ‘flow_rule_alloc’:
/home/paolini/onload/build/x86_64_linux-4.15.0-136-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.c:642:17: error: implicit declaration of function ‘struct_size’; did you mean ‘bd_set_size’? [-Werror=implicit-function-declaration]
  rule = kzalloc(struct_size(rule, action.entries, num_actions),
                 ^~~~~~~~~~~
                 bd_set_size
/home/paolini/onload/build/x86_64_linux-4.15.0-136-generic/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.c:642:3

Thank you,

Chris