Xilinx-CNS/onload #15: "Efsend_pio" - Performance test does not meet expectations

huoyingyangjie, 2021-03-15:
I used "efsend_pio" to send 32 udp packets. It took about 70us. Now I don’t know where the problem occurred. Can you give me a test data?

OS: centos7.6
CPU:i9-9980XE
NIC:X2522

abower-xilinx, 2021-03-18:
Hi, have you tried a supported release of Onload? For tuning advice with ef_vi on your NIC please contact support-nic@xilinx.com.

huoyingyangjie, 2021-03-19:
How can i get a supported release of Onload?

huoyingyangjie, 2021-03-19:
I have been using [openonload](https://github.com/majek/openonload).

abower-xilinx, 2021-03-19:
> How can i get a supported release of Onload?

Please see: https://support-nic.xilinx.com/wp/welcome

huoyingyangjie, 2021-03-19:
I found three [onloads](https://support-nic.xilinx.com/wp/onload). Which one of them can achieve the expected performance ?

abower-xilinx, 2021-03-19:
> I found three [onloads](https://support-nic.xilinx.com/wp/onload). Which one of them can achieve the expected performance ?

Please e-mail support-nic@xilinx.com for advice with tuning your application on your NIC!