Xilinx-CNS/onload #20: is onload must run on Solarflare network interfaces ?

h2cw2l, 2021-04-09:
when run my app, i see error as below: 
'''
This error can occur if no Solarflare network interfaces
are active/UP, or they are running packed stream
firmware, are disabled or lack Onload activation keys.
Please check your configuration. To obtain activation
keys, please contact your sales representative.
'''

LD_PRELOAD="$(mmaketool --toppath)/build/$(mmaketool --userbuild)/lib/transport/unix/libcitransport0.so" sockperf server -i 10.194.138.99 -p 1500 --tcp
oo:sockperf[62239]: netif_tcp_helper_alloc_u: ENODEV.
This error can occur if no Solarflare network interfaces
are active/UP, or they are running packed stream
firmware, are disabled or lack Onload activation keys.
Please check your configuration. To obtain activation
keys, please contact your sales representative.
sockperf: == version #3.7-1.gitb741ab3c60b1 ==
sockperf: [SERVER] listen on:
[ 0] IP = 10.194.139.66   PORT =  1500 # TCP
sockperf: Warmup stage (sending a few dummy messages)...
sockperf: [tid 62239] using recvfrom() to block on socket(s)
^Csockperf: Test end (interrupted by user)
sockperf: No messages were received on the server.
sockperf: cleanupAfterLoop() exit

abower-xilinx, 2021-04-09:
Did you register your non-Solarflare interface with Onload for AF_XDP as per [the quick start guide](https://github.com/Xilinx-CNS/onload#installation-and-quick-start-guide)?
```sh
echo ens2f0 > /sys/module/sfc_resource/afxdp/register
```

h2cw2l, 2021-04-09:
> Did you register your non-Solarflare interface with Onload for AF_XDP as per [the quick start guide](https://github.com/Xilinx-CNS/onload#installation-and-quick-start-guide)?
> 
> ```shell
> echo ens2f0 > /sys/module/sfc_resource/afxdp/register
> ```

closed 2021-04-09 by h2cw2l

h2cw2l, 2021-04-09:
> Did you register your non-Solarflare interface with Onload for AF_XDP as per [the quick start guide](https://github.com/Xilinx-CNS/onload#installation-and-quick-start-guide)?
> 
> ```shell
> echo ens2f0 > /sys/module/sfc_resource/afxdp/register
> ```

1、yes i have excuted this command
echo ens2f0 > /sys/module/sfc_resource/afxdp/register
2、infos of my NIC as follow
[root@A03-R05-I139-66-FVP3HP2 linux-4.18.0-240.10.1.el8_3]# ethtool -i eth0
driver: ixgbe
version: 5.1.0-k-rh8.2.0
firmware-version: 0x8000090c, 18.3.6
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes
3、os version 
[root@A03-R05-I139-66-FVP3HP2 linux-4.18.0-240.10.1.el8_3]# cat /etc/centos-release
CentOS Linux release 8.2.2004 (Core)
[root@A03-R05-I139-66-FVP3HP2 linux-4.18.0-240.10.1.el8_3]# uname -a
Linux A03-R05-I139-66-FVP3HP2.JD.LOCAL 4.18.0-240.10.1.el8_3.x86_64 #1 SMP Mon Jan 18 17:05:51 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
[root@A03-R05-I139-66-FVP3HP2 linux-4.18.0-240.10.1.el8_3]#