Xilinx-CNS/onload #28: OpenOnload on non solarflare adapter

hjastenger, 2021-05-29:
I feel like an idiot for asking this here, could not find a conclusive answer anywhere. Is it possible to run OpenOnload on non solarflare / Xilinx hardware. Aka if a setup my hosting from AWS, is it possible to configure OpenOnload to work on the ENA driver that automatically comes with that machine?

rhughes-xilinx, 2021-06-01:
Hi. Yes, Onload has an AF_XDP backend, which makes it possible to run it on any NIC.

AF_XDP support comes in 3 flavours, in increasing performance: generic, in-driver, and zero-copy. Last time I looked (admittedly a while ago) the ena driver supported neither in-driver nor zero-copy, so you're stuck with the slowest option, provided generically by Linux. Onload should still be faster than native sockets, though.

hjastenger, 2021-06-01:
Tried a couple of hours today to get the setup working. Noticed that the default ENA provided by Amazon was a bit out of date. After updating, compiling from this repository and running it I had the onload_cp_server running. Adding my eth0 yielded the following issue in dmesg  ```[  905.128865] [sfc efhw] xdp_set_link: eth0 does not support XDP```. Im a bit stuck currently, any idea why this is happening?

I was under the impression that after updating both the kernel (Red Hat) and the ENA driver I should have XDP support. Also I bit unsure how to check if my current setup supports XDP, grepping for `CONFIG_XDP_SOCKET` in my boot config yields it should be enabled. Was following this guide to check / enable XDP on ena https://trying2adult.com/what-is-xdp-and-how-do-you-use-it-in-linux-amazon-ec2-example/

Any idea why it looks like my eth0 isn't supporting XDP? Any help is appreciated, bit stuck atm. Thanks in advance

rhughes-xilinx, 2021-06-01:
I'm sorry, it looks like I gave you bad info - re-checking the code it looks like we do currently need in-driver support for XDP. As I see it you have a few options:

1) Give up
2) Enhance Onload to make it work with the generic XDP. It's possible that this might prove difficult if, for example, the necessary kernel APIs do not exist for Onload to get the details it needs.
3) Add XDP support to the ena driver. A minimal implementation may not be a massive amount of work
4) Switch to an instance type with an Intel NIC. See https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/enhanced-networking.html
5) Switch to Azure or GCP, both of which also offer NICs with better drivers

rhughes-xilinx, 2021-06-01:
OK, perhaps I ought to stop talking, since every time I (figuratively) open my mouth I give you bad advice, but I'm going to try again:

The current ena driver (in Linus's tree) does have in-driver support for XDP, so it should work with Onload. The specific error logging you gave (`eth0 does not support XDP`) can only be coming out if you have an ena driver without support, so I suggest you check that.

hjastenger, 2021-06-02:
Hi Richard, first of all thanks for the detailed response! Really appreciate you reaching out on this GH issue! Lots of information already for me to try out. Point of background on the previous mentioned points:

For the ENA driver im currently running version 2.5.0g, which should have support for XDP. So I'm either doing something when compiling and configuring the ENA or something went south with that specific version. Ill recheck on the amazon driver repository for any directions.

One last thing I noticed is the following when starting up open onload and loading the drivers into the kernel:
```
unload.sh: /sbin/rmmod onload
unload.sh: /sbin/rmmod sfc_char
unload.sh: /sbin/rmmod sfc_resource
unload.sh: /sbin/rmmod sfc_driverlink
NET_OPT is
CHAR_OPT is
modprobe: FATAL: Module mtdchar not found in directory /lib/modules/4.18.0-305.el8.x86_64
ERROR: Did not find sfc_control in /proc/devices
sfc is a RELEASE driver
RESOURCE_OPT is
CHAR_OPT is
ONLOAD_OPT is
```
The process appears to be running fine, just not sure whether this `FATAL` & `ERROR` logs might be a potential culprit.

hjastenger, 2021-06-02:
So from a different issue on the amazon driver repository (https://github.com/amzn/amzn-drivers/issues/173) I read the following: 

> The driver doesn't support AF_XDP yet and the application falls back to SKB mode (kernel's generic implementation for XDP processing) which implies copying.

So even though XDP is supported by the ENA driver, it doesn't support AF_XDP yet? Might be the reason why its currently failing.

eugeneia, 2021-06-02:
@hjastenger Fwiw (Hi from https://github.com/amzn/amzn-drivers/issues/173), the XDP related APIs should work on EC2 assuming you have recent kernel and ENA versions (I have tested a 5.11 kernel recently with ENA 2.5.0). Its just that its not particular fast at this time due to the issue you quoted.

hjastenger, 2021-06-02:
Hi @eugeneia, thanks for the info. I'm currently using RHEL 4.18, latest kernel from RH. I've tried upgrading the kernel using something like Elrepo to 5.12.9 but this results in not being able to build the onload directory properly. Any idea?

rhughes-xilinx, 2021-06-02:
> not sure whether this FATAL & ERROR logs might be a potential culprit

We really ought to find some way of hiding those messages. They confuse everybody and they're absolutely harmless.

The `eth0 does not support XDP` comes out specifically iff `dev->netdev_ops->ndo_bpf == NULL`. You can see at https://github.com/amzn/amzn-drivers/blob/1c30884cea0704df0850019b15045961eca975a6/kernel/linux/ena/ena_netdev.c#L4076 that this isn't the case, so my first guess would be that `ENA_XDP_SUPPORT` has somehow ended up off. Perhaps try sticking a `#error` at that line to be sure it's getting compiled-in.

ol-alexandra, 2021-06-03:
> I've tried upgrading the kernel using something like Elrepo to 5.12.9 but this results in not being able to build onload properly.

Onload build and works with linux-5.12 if CONFIG_VDPA is not set.  I've filed an internal bug for CONFIG_VDPA issue.  It usually takes a few weeks to fix such an issue.  In the best case we'll get a fix next Tuesday,

hjastenger, 2021-06-03:
Thanks for the replies all! Super helpful. Reverted my kernel upgrade back to 4.18 and swapped out the ENA for Intel (). Added my eth0 interface doesn't yield me the `eth0 does not support XDP` anymore. It currently tells me:
```
[  450.666568] [sfc efrm] efrm_nondl_register_device: register eth0
[  450.673665] [sfc efrm] Using VI range 0+(0-1)<<0
[  450.673667] [sfc efrm] eth0 type=4:
[  450.678906] [sfc efrm] efrm_driverlink_resume:
[  450.691265] [sfc efrm] ? hardware init failed (-22, attempt 1 of 1)
[  450.691266] [sfc efrm] ?: ERROR: hardware init failed rc=-22
[  450.697517] [sfc efrm] eth0 index=0 ifindex=2
[  450.702383] [onload] oo_nic_add: ifindex=2 oo_index=0
```
I hope hardware init failed due to the fact that im currently not using one of the Solarflare NIC's? Running something like `onload ping 8.8.8.8` still doesn't display the 'success' banner for me. Its still printing
```
oo:ping[6050]: netif_tcp_helper_alloc_u: ENODEV.
This error can occur if no Solarflare network interfaces
are active/UP, or they are running packed stream
firmware, are disabled or lack Onload activation keys.
Please check your configuration. To obtain activation
keys, please contact your sales representative.
oo:ping[6050]: __citp_netif_alloc: failed to construct netif (19)
oo:ping[6050]: citp_netif_alloc_and_init: failed to create netif (19)
oo:ping[6050]: citp_udp_socket: failed (errno:19) - PASSING TO OS
```

ol-alexandra, 2021-06-03:
Have you typed `echo enp1s0f0 |sudo tee /sys/module/sfc_resource/afxdp/register` for your NIC?

hjastenger, 2021-06-03:
You are referring to the step 'adding your interface' with `echo ens2f0 > /sys/module/sfc_resource/afxdp/register` right? Which I did for my eth0 interface, found the entry in dmesg `[  450.666568] [sfc efrm] efrm_nondl_register_device: register eth0`

ol-alexandra, 2021-06-03:
Do you see `Accelerating ens2f0: RX 1 TX 1` message in syslog from `onload_cp_server`?  Have you brought the interface up?

hjastenger, 2021-06-03:
So I guess you only add `ens2f0` if you have that interface, so thats why im adding `eth0` instead of the placeholder from the docs. syslog does contain that entry

```
xxx onload_cp_server[2870]: Accelerating eth0: RX 1 TX 1
```

ol-alexandra, 2021-06-03:
Cool.  Is the interface up?

hjastenger, 2021-06-03:
Yes the interface is up
```
[xx]# cat /sys/class/net/eth0/operstate
up
```

hjastenger, 2021-06-03:
Im still seeing a lot of context switches with `perf stat -e 'sched:sched_switch' -a -A --timeout 10000` when running something with onload, so I have the impression onload isn't functioning properly?

rhughes-xilinx, 2021-06-03:
The canonical way to determine whether Onload is functioning properly is with the `onload_stackdump` tool. `onload_stackdump dump` while the Onloaded app is still running should print out some info for each socket that's accelerated, each under a heading vaguely akin to `TCP 0:2036 lcl=192.168.138.3:48087 rmt=192.168.138.2:12345 ESTABLISHED`.

Note that `ping`'s sockets will never be accelerated by Onload (we have yet to find a use-case where anybody needed ICMP to be very efficient). For minimal testing I tend to use netcat; for performance testing iperf or its ilk are fine.

hjastenger, 2021-06-03:
> Note that ping's sockets will never be accelerated by Onload (we have yet to find a use-case where anybody needed ICMP to be very efficient).
Was only running it with ping to verify onload was running by checking the banner message. Read somewhere on the onload forum that you could easily check onload was functioning by checking for the sucess info banner.

> The canonical way to determine whether Onload is functioning properly is with the onload_stackdump tool. onload_stackdump dump while the Onloaded app is still running should print out some info for [...]
Have been trying out the code used by Cloudflare in one of their blogs 'https://blog.cloudflare.com/how-to-receive-a-million-packets/', but ill try to validate this with the `onload_stackdump` you describe, thanks again for the effort all!

ronenhamias, 2021-06-03:
this thread is really helpful so thanks! 

just small question where i can find `onload_stackdump`

hjastenger, 2021-06-04:
@ronenhamias I think (assumption) he might be referring to `onload_tcpdump`?

rhughes-xilinx, 2021-06-04:
No, I meant `onload_stackdump`. It should be in `build/gnu_x86_64/tools/ip`

ronenhamias, 2021-06-04:
thank you @rhughes-xilinx @hjastenger

monergeim, 2021-06-06:
Good day, gentlemens!
Regarding  onload_stackdump, I'm using awslinux 2 with kernel 5.4.117 and ena ver 2.5.0, but getting some strange error during onload install, which led me to nowhere:
```
make[2]: Entering directory `/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools'
make -C /usr/src/kernels/5.4.117-58.216.amzn2.x86_64 NDEBUG=1 GCOV= CC=cc  symverfile=/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools/Module.symvers KBUILD_EXTMOD=/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools _module_/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools
make[3]: Entering directory `/usr/src/kernels/5.4.117-58.216.amzn2.x86_64'
make[4]: *** No rule to make target `_module_/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools'.  Stop.
make[3]: *** [sub-make] Error 2
make[3]: Leaving directory `/usr/src/kernels/5.4.117-58.216.amzn2.x86_64'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools'
```
May somebody know the reason why it could appear?

ronenhamias, 2021-06-07:
Hello again 

following this thread - when i am running  `onload_stackdump dump` as suggested

i am getting:

`ci onload_stackdump should not itself be run under onload acceleration.`

am i missing anything?

rhughes-xilinx, 2021-06-07:
> `ci onload_stackdump should not itself be run under onload acceleration.`

Is the question "I definitely think I'm not running this command under Onload", "I don't understand what this is saying" or "how can I tell whether I am or not?"? My first guess would be that somewhere way up your terminal you ran `onload bash` and forgot.

ronenhamias, 2021-06-08:
actually was trying to verify that the installation works.

but i made a mistake and was trying to install it from `https://support-nic.xilinx.com/wp/onload?sd=SF-109585-LS-35&pe=SF-122921-DH-4`
and this version probably is for solarflare NIC only. and this is why its not working.

but when trying to build it from source the build fails with this error:

`./openonload/scripts/onload_install`
```
make[2]: Entering directory `/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools'
make -C /usr/src/kernels/5.4.117-58.216.amzn2.x86_64 NDEBUG=1 GCOV= CC=cc  symverfile=/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools/Module.symvers KBUILD_EXTMOD=/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools _module_/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools
make[3]: Entering directory `/usr/src/kernels/5.4.117-58.216.amzn2.x86_64'
make[4]: *** No rule to make target `_module_/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools'.  Stop.
make[3]: *** [sub-make] Error 2
make[3]: Leaving directory `/usr/src/kernels/5.4.117-58.216.amzn2.x86_64'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/home/ec2-user/openonload/build/x86_64_linux-5.4.117-58.216.amzn2.x86_64/lib/citools'
```

still trying to figure out why it fails to build - any ideas?

ol-alexandra, 2021-06-08:
Any ideas how I can download that Amazon's kernel `5.4.117-58.216.amzn2`?  Or at least provide full build log, please.  Sometimes there are some hints before the fatal error.

NB Onload is tested with Ubuntu's `5.4.0-73-generic`, so I'd expect that any linux-5.4 works.  Or at least builds.  Probably Amazon patches their kernel, and I have no idea how.  Google refused to provide me any details or download link for that Amazon's kernel.  May be my google-foo is not sufficient.

ceeaspb, 2021-11-13:
Doesn't fully anwser your question, as I am not sure 5.4 is available now, but you can download the kernel source for a running instance:
`yumdownloader --source kernel-5.10.75-79.358.amzn2`
It includes the upstream and patches.
May be of use to test the Amazon 2 image as it rolls forward.

aneagoe, 2022-01-19:
@hjastenger have you had any success with this? Are you able to share an update of the current state of things and a rough guide on getting this up and running?

aneagoe, 2022-01-19:
Ok, so I managed to compile, install and enable the latest code (commit ` 7b76edae6b7f7b7276c1496b2a6ed1420fab5306`) against kernel `4.18.0-348.7.1.el8_5.x86_64` on a centos8 ec2 instance. I had to lower the MTU and adjust the channels (ie `ethtool -L eth0 4`) after which I was able to successfully enable onload:
```Jan 19 15:20:46 localhost kernel: Solarflare driverlink driver v5.3.12.1010 API v33.0
Jan 19 15:20:46 localhost kernel: Solarflare NET driver v5.3.12.1010
Jan 19 15:20:46 localhost kernel: Efx driverlink registering resource driver
Jan 19 15:20:46 localhost kernel: [onload] Onload 7b76edae 2022-01-19 master 
Jan 19 15:20:46 localhost kernel: [onload] Copyright 2019-2022 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks 
Jan 19 15:20:46 localhost onload_cp_server[156356]: Spawned daemon process 156374
Jan 19 15:20:46 localhost onload_cp_server[156374]: Onload Control Plane server 7b76edae 2022-01-19 master  started: id 0, pid 156374 
Jan 19 15:20:51 localhost kernel: [sfc efrm] efrm_nondl_register_device: register eth0
Jan 19 15:20:51 localhost kernel: [sfc efrm] eth0 type=4:\x000 ifindex=2
Jan 19 15:20:51 localhost kernel: ena 0000:00:05.0 eth0: XDP program is set, changing the max_mtu from 9216 to 3498
Jan 19 15:20:51 localhost kernel: [sfc efrm] eth0 index=0 ifindex=2
Jan 19 15:20:51 localhost kernel: [onload] oo_nic_add: ifindex=2 oo_index=0 
Jan 19 15:20:51 localhost onload_cp_server[156374]: Accelerating eth0: RX 1 TX 1
```

Onload is then seemingly accelerating applications. However, when inspecting `onload_stackdump lots`, I noticed there are no `u_polls` and only `k_polls` is increasing. Any idea why?

@rhughes-xilinx @hjastenger @eugeneia

ol-alexandra, 2022-01-19:
Please share `onload_stackdump lots` to start with.

aneagoe, 2022-01-19:
@ol-alexandra please see below. There are no accelerated sockets... Comparatively, running the same application (just a udpsender) on a machine with X2522 card shows accelerated socket and u_polls is increasing as expected. 

```
[root@localhost ~]# onload_stackdump lots 
============================================================
ci_netif_dump_to_logger: stack=0 name=
  cplane_pid=156374
  namespace=net:[4026531992]
  Onload 7b76edae 2022-01-19 master  uid=0 pid=174172 ns_flags=a0
  creation_time=2022-01-19 15:59:15 (delta=27secs)
  lock=0 UNLOCKED  nics=1 primed=1
  sock_bufs: max=8192 n_allocated=2048 free=2048
  aux_bufs: free=0
  aux_bufs[syn-recv state]: n=0 max=2048
  aux_bufs[syn-recv bucket]: n=0 max=8192
  aux_bufs[epoll3 state]: n=0 max=8192
  aux_bufs[unknown]: n=0 max=8192
  pkt_sets: pkt_size=2048 set_size=1024 max=32 alloc=32
  pkt_set[0]: free=493 current
  pkt_set[1]: free=1024
  pkt_set[2]: free=1024
  pkt_set[3]: free=1024
  pkt_set[4]: free=1024
  pkt_set[5]: free=1024
  pkt_set[6]: free=1024
  pkt_set[7]: free=1024
  pkt_set[8]: free=1024
  pkt_set[9]: free=1024
  pkt_set[10]: free=1024
  pkt_set[11]: free=1024
  pkt_set[12]: free=1024
  pkt_set[13]: free=1024
  pkt_set[14]: free=1024
  pkt_set[15]: free=1024
  pkt_set[16]: free=1024
  pkt_set[17]: free=1024
  pkt_set[18]: free=1024
  pkt_set[19]: free=1024
  pkt_set[20]: free=1024
  pkt_set[21]: free=1024
  pkt_set[22]: free=1024
  pkt_set[23]: free=1024
  pkt_set[24]: free=1024
  pkt_set[25]: free=1024
  pkt_set[26]: free=1024
  pkt_set[27]: free=1024
  pkt_set[28]: free=1024
  pkt_set[29]: free=1024
  pkt_set[30]: free=1024
  pkt_set[31]: free=1024
  pkt_bufs: max=32768 alloc=32768 free=32237 async=0
  pkt_bufs: rx=531 rx_ring=499 rx_queued=0 pressure_pool=32
  pkt_bufs: tx=0 tx_ring=0 tx_oflow=0
  pkt_bufs: in_loopback=0 in_sock=0
  pkt_bufs: rx_reserved=0
  time: netif=541ad3 poll=541ad3 now=541aea (diff=0.032sec)
  active cache: hit=0 avail=0 cache=EMPTY pending=EMPTY
  passive scalable cache: cache=EMPTY pending=EMPTY
ci_netif_dump_vi: stack=0 intf=0 dev=? hw=4
  vi=0 pd_owner=1 channel=0 tcpdump=off vi_flags=a00000 oo_vi_flags=0
  evq: cap=-1 current=0 is_32_evs=1 is_ev=0
  evq: sync_major=ffffffff sync_minor=0 sync_min=0
  evq: sync_synced=0 sync_flags=0
  rxq: cap=511 lim=511 spc=12 level=499 total_desc=45
  txq: cap=511 lim=511 spc=511 level=0 pkts=0 oflow_pkts=0
  txq: pio_buf_size=0 tot_pkts=0 bytes=0
  txq: ts_nsec=40000000
  clk: 
  last_rx_stamp: 0:0
  ctpio: max_frame_len=0 frame_len_check=0 ct_thresh=4294940120
[... 577 lines cut ...]
   ci_ip_time_tick2ms: 1.398118
   ci_ip_time_ms2tick: 0.715247 (b71a7400)
                  khz: 2999965
time constants for this CPU
  rto_initial: 715ticks (1000ms)
  rto_min: 144ticks (200ms)
  rto_max: 85829ticks (120000ms)
  delack: 35ticks (50ms)
  idle: 35ticks (50ms)
  keepalive_time: 169513ticks (237000ms)
  keepalive_intvl: 9298ticks (13000ms)
  keepalive_probes: 7
  zwin_max: 85829ticks (120000ms)
  paws_idle: 1483136993ticks (2073600000ms)
  PMTU slow discover: 407691ticks (570000ms)
  PMTU fast discover: 85829ticks (120000ms)
  PMTU recover: 21457ticks (30000ms)
  RFC 5961 challenge ack limit: 2 per tick, 1000 per sec

  Time between ACKs sent as a response to invalid incoming TCP packets: 358ticks (500ms)

  Intrumentation: 1ticks (0ms)
--------------------- process env --------------------------
--------------------------------------------
```

aneagoe, 2022-01-19:
@ol-alexandra ok, I managed to find something more. When I run the application with onload, I can see the following on `/var/log/messages`:
```
Jan 19 16:11:11 localhost kernel: [onload] [1]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
Jan 19 16:11:11 localhost kernel: ena 0000:00:05.0 eth0: Command parameter 46 is not supported
Jan 19 16:11:11 localhost kernel: [sfc efhw] rxclass_get_dev_info: rxclass: Cannot get RX class rule count 
Jan 19 16:11:11 localhost kernel: [onload] __oof_socket_add_wild: 1:2047 ERROR: FILTER UDP 172.29.28.32:65500 0.0.0.0:0 failed (-95)
Jan 19 16:11:14 localhost kernel: [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
Jan 19 16:11:14 localhost kernel: [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
```

ol-alexandra, 2022-01-19:
`ethtool -i  eth0` please

aneagoe, 2022-01-19:
@ol-alexandra
```
[root@localhost ~]# ethtool -i eth0
driver: ena
version: 4.18.0-348.7.1.el8_5.x86_64
firmware-version:
expansion-rom-version:
bus-info: 0000:00:05.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

ol-alexandra, 2022-01-19:
`ena` does not support n-tuple filtering, which is required for Onload-over-AF_XDP.

@abower-xilinx @maciejj-xilinx I do not see n-tuple filtering being mentioned in https://github.com/Xilinx-CNS/onload/blob/master/README.md

aneagoe, 2022-01-20:
@ol-alexandra thanks a lot for clarifying. Indeed, the driver doesn't seem to support n-tuple filtering, as per https://lore.kernel.org/lkml/1486646499-13682-2-git-send-email-netanel@annapurnalabs.com/

maciejj-xilinx, 2022-01-20:
In principle, n-tuple supports would not be required if the interface was configured with single channel.
 * {{ethtool -l <ifname>}} output would show how many RX channels there is and {{ethtool -L}} could be used to set it.

Currently, Onload strictly requires n-tuple support. However, it would be pretty simple change to allow it to run without filters when single channel is set on the NIC. Would this be your usecase?

aneagoe, 2022-01-20:
@maciejj-xilinx it's definitely worth a shot. Our goal is to optimize the ec2 network stack on top of ENA adapters by using kernel bypass. It's very difficult to asses how big of an improvement we'd be looking at without testing first. If this is an easy fix, I'm happy to test and report back.

aneagoe, 2022-02-24:
@maciejj-xilinx is there any chance to drop n-tuple hard requirement in case interface is configured with single channel?

maciejj-xilinx, 2022-02-28:
Hi aneagoe,

Thanks for getting in touch.

I have submitted a change that allows disabling use of ntuple filters.

Disabling ntuple filter can be achieved with
```echo 0 > /sys/module/sfc_resource/parameters/enable_af_xdp_flow_filters  ```

We have not tested the change specifically with ENA, but we are hopeful that this should get you further.

Hope that helps and let us know whether this works for you.

Maciej

aneagoe, 2022-02-28:
@maciejj-xilinx thanks a lot for this, I'll test with it shortly.

aneagoe, 2022-03-02:
That definitely did the trick and now I see u_polls being ~10x k_polls. So acceleration definitely works. However, it seems to be slower that without onload by ~1us (just observational testing with udpserver/udpclient and isolcpus/pinning). In both cases (with and without onload) I've used the same channel configuration (1).

maciejj-xilinx, 2022-03-08:
We have not characterized Onload on ENA specifically, however there are some general tuning guidelines to optimize Onload for latency.

On onload side the most important latency-wise tunables are:

EF_POLL_USEC=100000
EF_UL_EPOLL=3

From our observations with Onload over AF_XDP latency was better than with kernel stack. Though this would be largely depending on internals of particular network driver and its setting such as interrupt moderation.

In general we saw really good results with applications such as Redis or Nginx proxy With these the focus is throughput, or latency under load.

Can I ask what sort of application you are working with?

aneagoe, 2022-03-08:
The only testing that I've done (where I observed the difference in performance) was using udpclient/udpserver from here: https://github.com/majek/dump/tree/master/how-to-receive-a-packet. I haven't tested with `EF_UL_EPOLL=3` but since onload is using the latency profile by default `EF_POLL_USEC=100000` was set (and confirmed via `onload_stackdump lots`). 
In contrast, testing the same but on two back to back servers with X2522 adapters showed an improvement of about ~3.5us when running with onload:
```
[root@aloha ~]# taskset -c 5 udpclient 10.0.3.3:10000 --polling --timestamp
[*] Sending to 10.0.3.3:10000, polling=1, src_port=65500
pps= 47340 avg= 20.310us dev=  1.583us min= 0.000us  (packet=1.184us/0.333)  
pps= 47674 avg= 20.147us dev=  1.419us min=12.651us  (packet=1.185us/0.358)  
pps= 47691 avg= 20.139us dev=  1.456us min=12.631us  (packet=1.180us/0.330)  
pps= 47550 avg= 20.195us dev=  1.345us min=12.645us  (packet=1.198us/0.289)  
pps= 49298 avg= 19.442us dev=  1.519us min=12.234us  (packet=1.182us/0.334)  
pps= 48211 avg= 19.912us dev=  1.408us min=12.673us  (packet=1.182us/0.340)   
^C
[root@aloha ~]# taskset -c 5 onload udpclient 10.0.3.3:10000 --polling --timestamp
[*] Sending to 10.0.3.3:10000, polling=1, src_port=65500
oo:udpclient[1565109]: Using OpenOnload 7.1.1.75 [6]
oo:udpclient[1565109]: Copyright 2019-2021 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
pps= 65540 avg= 14.303us dev=  0.530us min= 0.000us  (packet=0.205us/0.115)  
pps= 64646 avg= 14.702us dev=  1.230us min= 9.197us  (packet=0.280us/0.099)  
pps= 64635 avg= 14.717us dev=  1.046us min= 9.121us  (packet=0.307us/0.250)  
pps= 66688 avg= 14.243us dev=  0.521us min= 8.918us  (packet=0.292us/0.166)  
pps= 65226 avg= 14.575us dev=  1.109us min= 9.080us  (packet=0.296us/0.076)  
pps= 64746 avg= 14.674us dev=  0.964us min= 9.156us  (packet=0.299us/0.096)  
```
Both on AWS and on the physical servers I've used isolcpus and kernel parameter optimizations.
I can imagine that running in single channel mode has some detrimental effects but even when not using onload the adapter was still set in single channel mode, so the comparison is quite fair.

shirshen12, 2022-03-23:
@aneagoe I have been able to run Onload-on-AF_XDP on Mellanox NiCs (Connect X 4 Lx and beyond with mlx5_core and AF_XDP ZC support) and Intel NiCs (ixgbe/i40e) in Openstack and bare-metals (Azure in progress) and the numbers are amazing for Redis and Memcached.

Redis: 2X vis-a-vis kernel
Memcached: 35% better vis-a-vis kernel

mikhail-lapshin, 2022-05-17:
Hi [maciejj-xilinx](https://github.com/maciejj-xilinx)

Trying the latest Onload with ENA, and this seems not working for me. 
> Disabling ntuple filter can be achieved with
> `echo 0 > /sys/module/sfc_resource/parameters/enable_af_xdp_flow_filters `

```
$ ethtool -S ens5 | grep xdp
     queue_0_rx_xdp_aborted: 3281
     queue_0_rx_xdp_drop: 0
     queue_0_rx_xdp_pass: 388380
     queue_0_rx_xdp_tx: 0
     queue_0_rx_xdp_invalid: 0
     queue_0_rx_xdp_redirect: 495933
```
queue_0_rx_xdp_redirect is growing when I run some traffic. queue_0_rx_xdp_pass growing as well. Does this imply that it works?

dmesg still have **FIXME AF_XDP**
```
[ 5902.191771] [onload] Onload  
[ 5902.194635] [onload] Copyright 2019-2022 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
[ 5902.304378] onload_cp_server[36188]: Spawned daemon process 36208
[ 5907.916669] [sfc efrm] efrm_nondl_register_device: register ens5
[ 5907.920690] [sfc efrm] ens5 type=4:\x000 ifindex=2
[ 5907.922772] ena 0000:00:05.0 ens5: XDP program is set, changing the max_mtu from 9216 to 3498
[ 5907.922786] [sfc efrm] ens5 index=0 ifindex=2
[ 5907.922798] [onload] oo_nic_add: ifindex=2 oo_index=0
[ 5984.419463] [onload] [0]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
[ 5994.476638] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
[ 5994.480669] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
```
My measurements with and without onload are the same, so I strongly believe onload is not working in my case. 

How to trouble shoot this?

shirshen12, 2022-05-18:
Ignore those FIXME_AF_XDP. You need to see if XDP_REDIRECT clause is implemented in the ENA driver or not. This is the same problem I see in Azure.

Onload works on all drivers where the XDP_REDIRECT clause is implemented like Intel, Mellanox.

You can try in AWS with EC2 instances which give PCI passthrough to Intel 82599 NiCs

mikhail-lapshin, 2022-05-18:
EC2 with Intel NIC is not working either. 
It fails on registration
`echo ens3 | sudo tee /sys/module/sfc_resource/afxdp/register
`
```
[ 1839.705182] [onload] Onload  
[ 1839.707833] [onload] Copyright 2019-2022 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
[ 1839.819177] onload_cp_server[35494]: Spawned daemon process 35514
[ 1862.899986] [sfc efrm] nondl_register_store: cannot detect number of channels for device ens3 assuming 1
[ 1862.899991] [sfc efrm] efrm_nondl_register_device: register ens3
[ 1862.906456] [sfc efrm] ens3 type=4:\x000 ifindex=2
[ 1862.906732] [sfc efrm] ?: ERROR: hardware init failed rc=-22
[ 1862.912687] [sfc efrm] ens3 index=0 ifindex=2
[ 1862.912742] [onload] oo_nic_add: ifindex=2 oo_index=0
```
And it's not just a log line to ignore, onload reports the same error.
```
$ sudo onload  iperf -c 172.31.36.232 -M 64 -t 10 -i 1
oo:iperf[35753]: netif_tcp_helper_alloc_u: ENODEV.
This error can occur if:
 - no Solarflare network interfaces are active/UP, or they are running
   packed stream firmware or are disabled, and
 - there are no AF_XDP interfaces registered with sfc_resource
Please check your configuration.
WARNING: attempt to set TCP maxmimum segment size to 64 failed.
Setting the MSS may not be implemented on this OS.
```

Still would prefer to sort the issue with ENA. These Intel NIC instances don't look good for our needs.

shirshen12, 2022-05-19:
let me post the entire Intel 82599 Instruction set:

`NiC Intel 82599 - 10GBe
Driver: ixgbe
OS: Ubuntu 21.04 LTS

# upgrade to latest OS kernel
apt update -y
apt upgrade -y
apt full-upgrade -y

# reboot into new kernel
reboot

apt install build-essential net-tools unzip libcap-dev linux-tools-common linux-tools-generic netperf libevent-dev libnl-route-3-dev tk bison tcl libnl-3-dev flex libnl-route-3-200 dracut python2 libpcap-dev -y
apt install initramfs-tools -y

# build the intel driver, ixgbe
wget https://downloadmirror.intel.com/682680/ixgbe-5.13.4.tar.gz
tar zxf ixgbe-5.13.4.tar.gz
cd ixgbe-5.13.4/src/
make install

# build the intel driver, i40e
wget http://downloadmirror.intel.com/709707/i40e-2.17.15.tar.gz
tar zxf i40e-2.17.15.tar.gz
cd i40e-2.17.15/src/
make install

# The binary will be installed as:
/lib/modules/<KERNEL VER>/updates/drivers/net/ethernet/intel/ixgbe/ixgbe.ko

# Load the ixgbe module using the modprobe command.
rmmod ixgbe; modprobe ixgbe

# Load the i40e module using the modprobe command.
rmmod i40e; modprobe i40e

# update the initrd/initramfs file to prevent the OS loading old versions of the ixgbe driver.
update-initramfs -u

# reboot again, just for safety
reboot

# Install Onload:
git clone https://github.com/Xilinx-CNS/onload.git
cd onload
scripts/onload_mkdist --release
cd onload-<version>/scripts/
./onload_install
./onload_tool reload

# register the NiC with AF_XDP driver interface
echo enp1s0 > /sys/module/sfc_resource/afxdp/register

# turn on Intel Flow Director
ethtool --features enp1s0 ntuple on`

shirshen12, 2022-05-19:
```
NiC Intel 82599 - 10GBe
Driver: ixgbe
OS: Ubuntu 21.04 LTS

### upgrade to latest OS kernel
apt update -y
apt upgrade -y
apt full-upgrade -y

### reboot into new kernel
reboot

apt install build-essential net-tools unzip libcap-dev linux-tools-common linux-tools-generic netperf libevent-dev libnl-route-3-dev tk bison tcl libnl-3-dev flex libnl-route-3-200 dracut python2 libpcap-dev -y
apt install initramfs-tools -y

### build the intel driver, ixgbe
wget https://downloadmirror.intel.com/682680/ixgbe-5.13.4.tar.gz
tar zxf ixgbe-5.13.4.tar.gz
cd ixgbe-5.13.4/src/
make install

### build the intel driver, i40e
wget http://downloadmirror.intel.com/709707/i40e-2.17.15.tar.gz
tar zxf i40e-2.17.15.tar.gz
cd i40e-2.17.15/src/
make install

### The binary will be installed as:
/lib/modules/<KERNEL VER>/updates/drivers/net/ethernet/intel/ixgbe/ixgbe.ko

### Load the ixgbe module using the modprobe command.
rmmod ixgbe; modprobe ixgbe

### Load the i40e module using the modprobe command.
rmmod i40e; modprobe i40e

### update the initrd/initramfs file to prevent the OS loading old versions of the ixgbe driver.
update-initramfs -u

### reboot again, just for safety
reboot

### Install Onload:
git clone https://github.com/Xilinx-CNS/onload.git
cd onload
scripts/onload_mkdist --release
cd onload-<version>/scripts/
./onload_install
./onload_tool reload

### register the NiC with AF_XDP driver interface
echo enp1s0 > /sys/module/sfc_resource/afxdp/register

### turn on Intel Flow Director
ethtool --features enp1s0 ntuple on
```

Ventus5566, 2022-06-06:
@shirshen12 

Hi 
which AWS ec2 type you used?
on AWS ec2, I use Ubuntu 20.04 LTS, but network adapter is ixgbevf, not ixgbe

there is three command has error message 
1.  ./onload_tool reload
error messge like this 
"modprobe: ERROR: could not insert 'sfc': Unknown symbol in module, or unknown parameter (see dmesg)
modprobe: ERROR: ../libkmod/libkmod-module.c:990 command_do() Error running install command '/sbin/modprobe --ignore-install sfc  &&             if [ -e /lib/modules/$(uname -r)/extra/sfc_aoe.ko ]; then               /sbin/modprobe sfc_aoe;             fi' for module sfc: retcode 1
modprobe: ERROR: could not insert 'sfc': Invalid argument
onload_tool: FAILED: /sbin/modprobe sfc

2. echo enp1s0 > /sys/module/sfc_resource/afxdp/register
after I do this command , register file is still empty
also I do "ulimit -l unlimited" , but look like not helpful

3. ethtool --features ens4 ntuple on
error message shows
"Cannot change ntuple-filters"
"Could not change any device features"
and "ethtool -k ens4" shows "ntuple-filters: off [fixed]"

shirshen12, 2022-06-06:
@Ventus5566 yes its ixgbevf and not ixgbe direct! You will need PCI passthrough access to Intel NiCs at minimum, which means ixgbe is required. I am not sure of ixgbevf. @rhughes-xilinx once said that ixgbevf needs patches.

I have tested on:

1. Bare-metal Intel 82599 and XL710 (ixgbe and i40e)
2. Bare-metal Mellanox Connect X 4Lx, 5X and 6X (mlx5_core 5.4.1 onwards)
3. VMs with PCI passthrough/SR-IOV access to Intel 82599 and XL710 (ixgbe and i40e and Mellanox Connect X 4Lx, 5X and 6X (mlx5_core 5.4.1 onwards)

cpp77, 2022-08-24:
I've just started on AWS for first time and wondered if someone here could be so kind to summarise the current state of play regarding using Onload on AWS? Is it hit & miss depending on which Network interface card you end up with? Are certain Linux distros better? Obviously we're all missing the Solarflare NICs as AWS doesn't use them.

lparkersc, 2024-06-25:
Hi all. This post is a bit stale. I've run into a few issues with setting up onload with ena on aws. I've created a new issue [here](https://github.com/Xilinx-CNS/onload/issues/231). Any help would be much appreciated.