Xilinx-CNS/onload #30: Possible regression as per journalctl

haelix888, 2021-06-19:
I'm seeing a possible regression introduced with one of these 2 commits on the 9th of June.

4267b166ea37d4d780160003a65029422fbd476a
2f1ba59e5ae77c31c67f53a52f326fc1f3c58e30

All else being equal, reverting the 2 commits gives me clean journalctl and getting back ~5% of performance.
The regression is in the usermode files not kernel ones.

```
: Using Onload <dev-snapshot> [0]
: Copyright 2019-present Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
: netif: ***** EVENT QUEUE OVERFLOW *****
: ... message limit reached

Jun 19 22:05:12 kkperf2 kernel: R10: 0000000000000002 R11: 0000000000000018 R12: ffff9ac8aac04348
Jun 19 22:05:12 kkperf2 kernel: R13: 0000000000000001 R14: ffffb95082208198 R15: ffff9ac8aac04028
Jun 19 22:05:12 kkperf2 kernel: FS:  0000000000000000(0000) GS:ffff9acaca280000(0000) knlGS:0000000000000000
Jun 19 22:05:12 kkperf2 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 19 22:05:12 kkperf2 kernel: CR2: 000055c8ef8dd2e8 CR3: 0000000242810002 CR4: 00000000007706e0
Jun 19 22:05:12 kkperf2 kernel: PKRU: 55555554
Jun 19 22:05:12 kkperf2 kernel: Call Trace:
Jun 19 22:05:12 kkperf2 kernel:  ? ci_netif_rx_post+0x1fc/0x740 [onload]
Jun 19 22:05:12 kkperf2 kernel:  ? ci_netif_poll_n+0x7f9/0x19a0 [onload]
Jun 19 22:05:12 kkperf2 kernel:  ? newidle_balance+0x352/0x3b0
Jun 19 22:05:12 kkperf2 kernel:  ? __switch_to_asm+0x42/0x70
Jun 19 22:05:12 kkperf2 kernel:  ? linux_tcp_helper_periodic_timer+0x138/0x1e0 [onload]
Jun 19 22:05:12 kkperf2 kernel:  ? process_one_work+0x1df/0x370
Jun 19 22:05:12 kkperf2 kernel:  ? worker_thread+0x50/0x400
Jun 19 22:05:12 kkperf2 kernel:  ? process_one_work+0x370/0x370
Jun 19 22:05:12 kkperf2 kernel:  ? kthread+0x11b/0x140
Jun 19 22:05:12 kkperf2 kernel:  ? kthread_associate_blkcg+0xa0/0xa0
Jun 19 22:05:12 kkperf2 kernel:  ? ret_from_fork+0x1f/0x30
Jun 19 22:05:12 kkperf2 kernel: ---[ end trace f4b2f513f2d0c08a ]---
Jun 19 22:05:12 kkperf2 kernel: ------------[ cut here ]------------
Jun 19 22:05:12 kkperf2 kernel: WARNING: CPU: 10 PID: 1655 at /home/oo/onload/build/x86_64_linux-5.10.42-1-lts/lib/ciul/ef10_vi.c:686 ef10_ef_vi_receive_init+0xcf/0xf0 [onload]
Jun 19 22:05:12 kkperf2 kernel: Modules linked in: xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc overlay intel_rapl_msr intel_rapl_>
Jun 19 22:05:12 kkperf2 kernel:  ghash_clmulni_intel sfc(OE) snd_pcm_dmaengine aesni_intel snd_pcm cec r8169 crypto_simd cryptd virtual_bus(OE) intel_gtt snd_timer glue_helper syscopyarea mdio snd realtek sysfillrect vfat vdpa ipmi_devintf mdio_devres sysimgblt fat rapl>
Jun 19 22:05:12 kkperf2 kernel: CPU: 10 PID: 1655 Comm: kworker/u41:1 Tainted: G        W  OE     5.10.42-1-lts #1
Jun 19 22:05:12 kkperf2 kernel: Hardware name: ASUS System Product Name/Pro WS W480-ACE, BIOS 2004 02/08/2021
Jun 19 22:05:12 kkperf2 kernel: Workqueue: onload-wq:0 linux_tcp_helper_periodic_timer [onload]
Jun 19 22:05:12 kkperf2 kernel: RIP: 0010:ef10_ef_vi_receive_init+0xcf/0xf0 [onload]
Jun 19 22:05:12 kkperf2 kernel: Code: 31 c0 c3 48 63 51 10 48 03 81 c0 00 00 00 48 b9 00 00 00 00 00 00 ff ff 48 85 ca 75 14 48 c1 e2 30 48 09 f2 48 89 10 31 c0 c3 <0f> 0b e9 70 ff ff ff 0f 0b eb e8 0f 0b eb 91 0f 0b eb 9c 0f 0b eb
Jun 19 22:05:12 kkperf2 kernel: RSP: 0018:ffffb9508196bcf8 EFLAGS: 00010213
Jun 19 22:05:12 kkperf2 kernel: RAX: 0000000000000188 RBX: 0000000000000008 RCX: ffff9ac8aac04348
Jun 19 22:05:12 kkperf2 kernel: RDX: 0000000000000384 RSI: 00080000161c20c0 RDI: ffffb950822077b8
Jun 19 22:05:12 kkperf2 kernel: RBP: ffffb950822081ac R08: ffffb9508220714c R09: ffffb95082208440
Jun 19 22:05:12 kkperf2 kernel: R10: 0000000000000002 R11: 0000000000000018 R12: ffff9ac8aac04348
Jun 19 22:05:12 kkperf2 kernel: R13: 0000000000000001 R14: ffffb95082208198 R15: ffff9ac8aac04028
Jun 19 22:05:12 kkperf2 kernel: FS:  0000000000000000(0000) GS:ffff9acaca280000(0000) knlGS:0000000000000000
Jun 19 22:05:12 kkperf2 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 19 22:05:12 kkperf2 kernel: CR2: 000055c8ef8dd2e8 CR3: 0000000242810002 CR4: 00000000007706e0
```

ol-alexandra, 2021-06-21:
(1) My patch "2f1ba59 ON-13268: ON-13147: avoid iput() when alloc_file_pseudo() fails" does not look related to the snipped of a Call Trace above.  @haelix888, are you sure it is necessary to revert it to remove the issue?
(2) Our testing does not show such an issue.  Please provide full Call trace from the beginning.  The beginning is usually a few line before the **first** "cut here" line.
(3) If possible, please provide a reproducer.  Are you using SFC NIC or AF_XDP?  What kind of application do you run?  How do you tune Onload with EF_* environment variables?  Do you use debug or ndebug Onload build?
(4) @mseymour-xilinx your patch "4267b16 ON-13173: ef_vi rx polling for EFCT" looks possibly related.  Do you have any ideas?  I understand it is hard to get any ideas without the beginning of the Call Trace, but I hope we'll get more details soon.

rhughes-xilinx, 2021-06-21:
One possible idea (that's easy to rule out quickly) is that you've created a frankenbuild, affected by the change in the `struct ef_vi` layout. Try cleaning everything and building again.

haelix888, 2021-06-25:
I'm sorry, you are right. I had not upgraded the kernel files on the host properly, it seems.
I see it working nicely now, as well as more recent commits.

closed 2021-06-28 by ol-alexandra