Xilinx-CNS/onload #31: out of bounds error

wushuhsu, 2021-06-24:
CC      /root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/netif_tx.o
/root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/netif_tx.c: In function ‘ci_netif_dmaq_shove_plugin’:
/root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/netif_tx.c:203:38: error: array subscript 1 is above array bounds of ‘ef_vi[1]’ [-Werror=array-bounds]
  203 |   ef_vi* vi = &ni->nic_hw[intf_i].vis[q_id];
      |                ~~~~~~~~~~~~~~~~~~~~~~^~~~~~
In file included from /root/src/onload-master/src/include/ci/internal/ip.h:58,
                 from /root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/ip_internal.h:19,
                 from /root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/netif_tx.c:16:
/root/src/onload-master/src/include/ci/internal/ip_types.h:40:30: note: while referencing ‘vis’
   40 |   ef_vi                      vis[CI_MAX_VIS_PER_INTF];
      |                              ^~~
/root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/netif_tx.c:203:38: error: array subscript 1 is above array bounds of ‘ef_vi[1]’ [-Werror=array-bounds]
  203 |   ef_vi* vi = &ni->nic_hw[intf_i].vis[q_id];
      |                ~~~~~~~~~~~~~~~~~~~~~~^~~~~~
In file included from /root/src/onload-master/src/include/ci/internal/ip.h:58,
                 from /root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/ip_internal.h:19,
                 from /root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/netif_tx.c:16:
/root/src/onload-master/src/include/ci/internal/ip_types.h:40:30: note: while referencing ‘vis’
   40 |   ef_vi                      vis[CI_MAX_VIS_PER_INTF];
      |                              ^~~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:334: /root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip/netif_tx.o] Error 1
make[4]: *** [Makefile:1316: _module_/root/src/onload-master/build/x86_64_linux-3.10.0-1160.31.1.el7.x86_64/lib/transport/ip] Error 2

rhughes-xilinx, 2021-06-24:
What compiler and/or compile flags are you using?

We can obviously fix this (probably by ifdeffing that function out in such configs) but I'm interested in what you're doing differently to us.

wushuhsu, 2021-06-24:
I'm using devtools-10 on CentOS 7.

% uname -a

Linux redacted-hostname 3.10.0-1160.31.1.el7.x86_64 #1 SMP Thu Jun 10
13:32:12 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

% gcc -v

Using built-in specs.

COLLECT_GCC=gcc

COLLECT_LTO_WRAPPER=/opt/rh/devtoolset-10/root/usr/libexec/gcc/x86_64-redhat-linux/10/lto-wrapper

Target: x86_64-redhat-linux

Configured with: ../configure --enable-bootstrap
--enable-languages=c,c++,fortran,lto
--prefix=/opt/rh/devtoolset-10/root/usr
--mandir=/opt/rh/devtoolset-10/root/usr/share/man
--infodir=/opt/rh/devtoolset-10/root/usr/share/info --with-bugurl=
http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix
--enable-checking=release --enable-multilib --with-system-zlib
--enable-__cxa_atexit --disable-libunwind-exceptions
--enable-gnu-unique-object --enable-linker-build-id
--with-gcc-major-version-only --with-linker-hash-style=gnu
--with-default-libstdcxx-abi=gcc4-compatible --enable-plugin
--enable-initfini-array
--with-isl=/builddir/build/BUILD/gcc-10.2.1-20200804/obj-x86_64-redhat-linux/isl-install
--disable-libmpx --enable-gnu-indirect-function --with-tune=generic
--with-arch_32=x86-64 --build=x86_64-redhat-linux

Thread model: posix

Supported LTO compression algorithms: zlib

gcc version 10.2.1 20200804 (Red Hat 10.2.1-2) (GCC)

ol-alexandra, 2021-06-25:
We saw it before.  And we decided that we don't want to support Frankenstein of new gcc + old kernel.  You are welcome to provide a patch.

wushuhsu, 2021-06-25:
*diff --git a/src/lib/transport/ip/netif_tx.c
b/src/lib/transport/ip/netif_tx.c*

*index cd19432..2cfadaf 100644*

*--- a/src/lib/transport/ip/netif_tx.c*

*+++ b/src/lib/transport/ip/netif_tx.c*

@@ -198,6 +198,7 @@ void ci_netif_dmaq_shove2(ci_netif* ni, int intf_i, int
is_fresh)

 }

+#if CI_CFG_TCP_OFFLOAD_RECYCLER

 void ci_netif_dmaq_shove_plugin(ci_netif* ni, int intf_i, int q_id)

 {

   ef_vi* vi = &ni->nic_hw[intf_i].vis[q_id];

@@ -206,7 +207,7 @@ void ci_netif_dmaq_shove_plugin(ci_netif* ni, int
intf_i, int q_id)

     __ci_netif_dmaq_shove(ni, &ni->state->nic[intf_i].dmaq[q_id], vi,
intf_i,

                           0 /*is_fresh*/);

 }

-

+#endif

 void __ci_netif_send(ci_netif* netif, ci_ip_pkt_fmt* pkt)

 {

ol-alexandra, 2021-06-25:
Sorry, it is unreadable.  Please make up a pull request.