Xilinx-CNS/onload #32: Kernel panic when running onload on Ubuntu veth

adrienlemasle, 2021-06-25:
Hi,

I'm trying to run onload on a linux veth interface pair on Ubuntu 20.10:
`Linux ubuntu 5.8.0-44-generic #50-Ubuntu SMP Tue Feb 9 06:29:41 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

I create a veth pair and register the interfaces as follows:
```ip link add veth0 type veth peer name veth1
ip addr add 10.1.0.1/24 dev veth0
ip addr add 10.1.0.2/24 dev veth1
ip link set veth0 up
ip link set veth1 up
ip addr add 10.1.0.2/24 dev veth1^C
echo veth0 > /sys/module/sfc_resource/afxdp/register
echo veth1 > /sys/module/sfc_resource/afxdp/register```
```
When trying a simple command:
```/home/adrien/onload/scripts/onload netcat -l 10.1.0.1 10000```

I get the following kernel panic:

```[  636.641548] sfc_driverlink: loading out-of-tree module taints kernel.
[  636.641583] sfc_driverlink: module verification failed: signature and/or required key missing - tainting kernel
[  636.642225] Solarflare driverlink driver v5.3.8.1011 API v32.0
[  636.675918] Solarflare NET driver v5.3.8.1011
[  637.294107] [sfc efrm] init_sfc_resource: RESOURCE driver starting
[  637.294108] [sfc efrm] efrm_driver_ctor: driver created
[  637.294237] [sfc efrm] efrm_driverlink_register:
[  637.294238] Efx driverlink registering resource driver
[  637.368676] [onload] Onload <dev-snapshot>
[  637.368680] [onload] Copyright 2019-present Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
[  637.476205] onload_cp_server[31076]: Spawned daemon process 31080
[  716.488784] IPv6: ADDRCONF(NETDEV_CHANGE): veth1: link becomes ready
[  716.488927] IPv6: ADDRCONF(NETDEV_CHANGE): veth0: link becomes ready
[  742.890243] [sfc efrm] nondl_register_store: cannot detect number of channels for device veth0 assuming 1
[  742.890245] [sfc efrm] efrm_nondl_register_device: register veth0
[  742.890249] [sfc efrm] Using VI range 0+(0-1)<<0
[  742.890250] [sfc efrm] veth0 type=4:\x000 ifindex=4
[  742.890277] [sfc efrm] efrm_driverlink_resume:
[  742.890971] [sfc efrm] veth0 index=0 ifindex=4
[  742.890977] [onload] oo_nic_add: ifindex=4 oo_index=0
[  745.858300] [sfc efrm] nondl_register_store: cannot detect number of channels for device veth1 assuming 1
[  745.858302] [sfc efrm] efrm_nondl_register_device: register veth1
[  745.858308] [sfc efrm] Using VI range 0+(0-1)<<0
[  745.858309] [sfc efrm] veth1 type=4:\x000 ifindex=3
[  745.858340] [sfc efrm] efrm_driverlink_resume:
[  745.858935] [sfc efrm] veth1 index=1 ifindex=3
[  745.858942] [onload] oo_nic_add: ifindex=3 oo_index=1
[  791.998897] [sfc efrm] efrm_vi_rm_delayed_free: 0000000058301cf0
[  791.998907] [sfc efrm] efrm_vi_rm_delayed_free: 0000000058301cf0
[  792.000153] [onload] [0]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
[  792.049776] BUG: kernel NULL pointer dereference, address: 0000000000000050
[  792.049783] #PF: supervisor read access in kernel mode
[  792.049784] #PF: error_code(0x0000) - not-present page
[  792.049786] PGD 0 P4D 0 
[  792.049790] Oops: 0000 [#1] SMP PTI
[  792.049794] CPU: 0 PID: 31163 Comm: netcat Tainted: G           OE     5.8.0-44-generic #50-Ubuntu
[  792.049796] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 07/22/2020
[  792.049823] RIP: 0010:efrm_filter_insert+0x55/0x360 [sfc_resource]
[  792.049826] Code: 81 16 01 00 48 89 c7 49 89 c4 e8 36 35 00 00 48 85 c0 0f 84 ec 02 00 00 49 89 c5 48 8b 80 50 05 00 00 48 c7 45 c0 00 00 00 00 <4c> 8b 70 50 4d 85 f6 75 03 4c 8b 30 48 c7 c7 48 aa 9f c0 e8 93 64
[  792.049828] RSP: 0018:ffffadf5c1d038d0 EFLAGS: 00010286
[  792.049830] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000001027
[  792.049832] RDX: 0000000000000001 RSI: 00000000fffffe01 RDI: ffffffffc09dd3f9
[  792.049833] RBP: ffffadf5c1d03910 R08: 0000000000000008 R09: 0000000000000262
[  792.049834] R10: ffffadf5c1d0393c R11: 0000000000000000 R12: ffffa013909c6c00
[  792.049835] R13: ffffa013b4cbf000 R14: 0000000000000000 R15: ffffadf5c1d0393c
[  792.049838] FS:  00007f85a4017300(0000) GS:ffffa013b9e00000(0000) knlGS:0000000000000000
[  792.049839] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  792.049841] CR2: 0000000000000050 CR3: 0000000082626002 CR4: 00000000001606f0
[  792.049842] Call Trace:
[  792.049882]  oo_hw_filter_set_hwport+0x2dd/0x860 [onload]
[  792.049899]  oo_hw_filter_add_hwports+0x1a0/0x240 [onload]
[  792.049916]  oo_hw_filter_set+0x1f/0x40 [onload]
[  792.049931]  __oof_hw_filter_set.constprop.0+0x18b/0x770 [onload]
[  792.049948]  __oof_socket_add_wild+0xfa/0x180 [onload]
[  792.049964]  oof_socket_add_wild.constprop.0+0x174/0x640 [onload]
[  792.049969]  ? kmalloc_order+0x63/0x80
[  792.049972]  ? kmalloc_order_trace+0x24/0xa0
[  792.049998]  oof_socket_add+0x1654/0x1bf0 [onload]
[  792.050002]  ? ima_file_check+0x5b/0x80
[  792.050016]  tcp_helper_endpoint_set_filters+0x338/0x1190 [onload]
[  792.050031]  efab_ep_filter_set+0x38/0x80 [onload]
[  792.050041]  oo_fop_unlocked_ioctl+0x213/0x3b0 [onload]
[  792.050045]  ? _cond_resched+0x1a/0x50
[  792.050050]  ksys_ioctl+0x8e/0xc0
[  792.050052]  __x64_sys_ioctl+0x1a/0x20
[  792.050055]  do_syscall_64+0x49/0xc0
[  792.050058]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  792.050060] RIP: 0033:0x7f85a415231b
[  792.050063] Code: 89 d8 49 8d 3c 1c 48 f7 d8 49 39 c4 72 b5 e8 1c ff ff ff 85 c0 78 ba 4c 89 e0 5b 5d 41 5c c3 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 1d 3b 0d 00 f7 d8 64 89 01 48
[  792.050065] RSP: 002b:00007fff444bbbd8 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  792.050067] RAX: ffffffffffffffda RBX: 00007f85a4016698 RCX: 00007f85a415231b
[  792.050068] RDX: 00007fff444bbbec RSI: 00000000400c5a23 RDI: 0000000000000005
[  792.050070] RBP: 0000000000000000 R08: 00000000005ffb40 R09: 0000000000000007
[  792.050071] R10: 00007f85a31dbc00 R11: 0000000000000246 R12: 0000000000000005
[  792.050072] R13: 00000000000007ff R14: 00007f85a43d27f8 R15: 00000000001ffc00
[  792.050074] Modules linked in: veth onload(OE) sfc_char(OE) sfc_resource(OE) sfc(OE) virtual_bus(OE) sfc_driverlink(OE) nf_flow_table nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c vdpa mtd 8021q garp mrp stp llc mdio hwmon_vid i2c_algo_bit crc32_generic mii rfcomm bnep vsock_loopback vmw_vsock_virtio_transport_common vmw_vsock_vmci_transport vsock nls_iso8859_1 intel_rapl_msr intel_rapl_common crct10dif_pclmul ghash_clmulni_intel aesni_intel crypto_simd cryptd glue_helper rapl vmw_balloon snd_ens1371 snd_ac97_codec gameport ac97_bus snd_pcm snd_seq_midi joydev input_leds snd_seq_midi_event serio_raw btusb btrtl snd_rawmidi btbcm btintel bluetooth snd_seq uvcvideo ecdh_generic videobuf2_vmalloc ecc videobuf2_memops videobuf2_v4l2 snd_seq_device videobuf2_common snd_timer videodev mc snd soundcore vmw_vmci mac_hid sch_fq_codel vmwgfx ttm drm_kms_helper cec rc_core fb_sys_fops syscopyarea sysfillrect sysimgblt parport_pc ppdev lp drm parport ip_tables x_tables autofs4 hid_generic
[  792.050119]  usbhid hid crc32_pclmul psmouse mptspi mptscsih mptbase e1000 scsi_transport_spi ahci libahci i2c_piix4 pata_acpi
[  792.050130] CR2: 0000000000000050
[  792.050132] ---[ end trace 5b17494d5fe46b2e ]---
[  792.050142] RIP: 0010:efrm_filter_insert+0x55/0x360 [sfc_resource]
[  792.050144] Code: 81 16 01 00 48 89 c7 49 89 c4 e8 36 35 00 00 48 85 c0 0f 84 ec 02 00 00 49 89 c5 48 8b 80 50 05 00 00 48 c7 45 c0 00 00 00 00 <4c> 8b 70 50 4d 85 f6 75 03 4c 8b 30 48 c7 c7 48 aa 9f c0 e8 93 64
[  792.050146] RSP: 0018:ffffadf5c1d038d0 EFLAGS: 00010286
[  792.050148] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000001027
[  792.050149] RDX: 0000000000000001 RSI: 00000000fffffe01 RDI: ffffffffc09dd3f9
[  792.050150] RBP: ffffadf5c1d03910 R08: 0000000000000008 R09: 0000000000000262
[  792.050151] R10: ffffadf5c1d0393c R11: 0000000000000000 R12: ffffa013909c6c00
[  792.050152] R13: ffffa013b4cbf000 R14: 0000000000000000 R15: ffffadf5c1d0393c
[  792.050154] FS:  00007f85a4017300(0000) GS:ffffa013b9e00000(0000) knlGS:0000000000000000
[  792.050156] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  792.050157] CR2: 0000000000000050 CR3: 0000000082626002 CR4: 00000000001606f0
[  792.050812] [onload] Stack [0] released with lock stuck (0x4000000020000000)```

I've also tried on an Ubuntu 20.04 VM running on AWS and I am seeing the same issue.

Thanks for your help

Adrien

ol-alexandra, 2021-06-25:
Which Onload revision do you use?

adrienlemasle, 2021-06-25:
I'm using latest from master (b23f804).

ol-alexandra, 2021-06-25:
If I get the numbers correctly, then it crashes in `efrm_filter_check(): name = dev_name(dev);`.  It means that `net_dev->dev ==NULL`.  Which is plausible for vmware's veth.

I have no idea whether veth supports AF_XDP; we definitely have not tested it.

@sianj-xilinx I guess it may be a recent regression from your "do everything via driverlink" patches.

ol-alexandra, 2021-06-25:
@adrienlemasle Please try the following patch:
```
diff --git a/src/driver/linux_resource/filter.c b/src/driver/linux_resource/filter.c
index ebd6ca3d09..ceb6a4e6f3 100644
--- a/src/driver/linux_resource/filter.c
+++ b/src/driver/linux_resource/filter.c
@@ -2187,7 +2187,7 @@ int efrm_filter_insert(struct efrm_client *client,
 {
        struct efhw_nic *efhw_nic = efrm_client_get_nic(client);
        struct net_device* net_dev;
-       int rc;
+       int rc = 0;
 
 #if CI_CFG_IPV6
        /* FIXME: add IPv6 support to firewall rules (bug 85208) */
@@ -2202,7 +2202,8 @@ int efrm_filter_insert(struct efrm_client *client,
           block it. */
        net_dev = efhw_nic_get_net_dev(efhw_nic);
        if( net_dev ) {
-               rc = efrm_filter_check( net_dev->dev.parent, spec );
+               if( net_dev->dev )
+                       rc = efrm_filter_check( net_dev->dev.parent, spec );
                dev_put(net_dev);
                if ( rc >= 0 )
                        rc = efhw_nic_filter_insert( efhw_nic, spec, replace );
```

adrienlemasle, 2021-06-25:
This does not compile.

```
/home/adrien/onload/build/x86_64_linux-5.8.0-59-generic/driver/linux_resource/filter.c:2205:7: error: used struct type value where scalar is required
 2205 |   if( net_dev->dev )
```

I tried `if( &net_dev->dev )` but still same kernel panic.

adrienlemasle, 2021-06-25:
XDP is support for veth FYI: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=d445516966dcb2924741b13b27738b54df2af01a

adrienlemasle, 2021-06-25:
@ol-alexandra so I tried `if(net_dev->dev.parent)` and this did the trick.

However the call to efhw_nic_filter_insert returns and EOPNOTSUPP error:
```[  143.755021] [onload] __oof_socket_add_wild: 1:2047 ERROR: FILTER TCP 10.1.0.1:10000 0.0.0.0:0 failed (-95)```

I have not idea what this call is doing but if I comment it out, then it seems to be working fine. Any idea if this call is needed for virtual ethernet interface?

ol-alexandra, 2021-06-28:
Thank you for fixing my stupid mistake!  It was Friday evening (a lame excuse, I know).  I'll push this fix.

The `FILTER failed` error is important.  "It works fine" without it because it passes everything to Linux; Onload does nothing in such a case.
@rhughes-xilinx You mentioned which property of AF_XDP support we need for filtering.  Can you comment here, please?

adrienlemasle, 2021-06-28:
@ol-alexandra no worries :)

Let me quickly describe what I'm trying to achieve overall. Basically we have some applications that use onload delegated sends for TCP and we have a limited number of test machines with Solarflare NICs. What we want is to be able to create test setups which don't require NICs, hence why I'm trying to get this to work with Virtual ethernet NICs. For our test setups we don't care about performance, so even if onload passes everything to Linux, as long as the delegated send API works, this would be enough for us.

rhughes-xilinx, 2021-06-28:
It was `ndo_bpf`/`XDP_SETUP_PROG`, which veth (in Linux master) does support. I don't think we ever got around to implementing softwareish filtering for XDP, however. @maciejj-xilinx would know.

kieranm-xilinx, 2021-06-28:
>  For our test setups we don't care about performance, so even if onload passes everything to Linux, as long as the delegated send API works, this would be enough for us.

Unfortunatley if Onload can't accelerate a socket and it is passed to Linux then Onload has no visibility of the socket's state and can't support the delegated send API.  I.e. to use the delegated send API the socket must be accelerated.

adrienlemasle, 2021-06-28:
> Unfortunatley if Onload can't accelerate a socket and it is passed to Linux then Onload has no visibility of the socket's state and can't support the delegated send API. I.e. to use the delegated send API the socket must be accelerated.

@kieranm-xilinx and @ol-alexandra what I'm confused about is that even after I've removed the call to efhw_nic_filter_insert that sets up the TCP filter, the socket still seems to somehow get accelerated cause I can see its state with onload_stackdump...

ol-alexandra, 2021-06-28:
It's really strange.  I do not understand how Onload gets the TCP packets if there is no filtering...  But if a TCP socket is shown in stackdump in the ESTABLISHED state, then it is an accelerated socket, sure.  And delegated send should work.

adrienlemasle, 2021-06-28:
It works once after loading the kernel modules:

```
  TCP 0:2046 lcl=192.168.228.5:42481 rmt=192.168.228.3:42643 ESTABLISHED
  lock: 0 
  rx_wake=00000001     tx_wake=00010437     flags: OS_BACKED 
  ul_poll: 0 spin cycles 0 usec
  uid=0 s_flags: REUSE FILTER BOUND PMTU_DO 
  rcvbuf=131400 sndbuf=131072
  rcvtimeo_ms=0 sndtimeo_ms=0 sigown=0 cmsg=
  bindtodev=0(0,0x0:0) ttl=-1 BOUND_ADDR 
  rx_errno=0 tx_errno=0 so_error=0 os_sock=0,TX
  tcpflags: TSO WSCL SACK ESTAB TLP_TIMER  local_peer: -1
  snd: up=e50f91e5 una-nxt-max=e50f91e5-e50fc695-e52ab0e5 enq=e50fc695
  snd: send=0(4) send+pre=4 inflight=18288(10) wnd=1777408 unused=1759120
  snd: cwnd=8570928+0 used=0 ssthresh=8388480 bytes_acked=6900240 Open
  snd: timed_seq a478d954 timed_ts 4dbcd
  snd: sndbuf_pkts=136 Onloaded(Valid) if=4 mtu=1500 intf_i=0 vlan=0 encap=100 verinfo 206-2
  snd: limited rwnd=0 cwnd=4 nagle=8 more=0 app=131594
  rcv: nxt-max=d29a5390-d29c54d8 wnd adv=131400 cur=131400  FASTSTART FAST
  rcv: isn=d29a5390 up=ffffffff urg_data=0000 q=recv1
  rcv: bytes=0 tot_pkts=40a24 rob_pkts=0 q_pkts=0+0 usr=0
  eff_mss=1448 smss=1460 amss=1460  used_bufs=1 wscl s=7 r=2
  srtt=01 rttvar=000 rto=110 zwins=0,0
  curr_retrans=0 total_retrans=0 dupacks=0 congrecover=0
  rtos=0 frecs=0 seqerr=0,0 ooo_pkts=0 ooo=0
  tx: defer=1451 nomac=0 warm=0 warm_aborted=0
  tmpl: send_fast=0 send_slow=0 active=0
  timers: 
rto(200ms[4e42d])
```

Then next time I try to rerun I get an EBUSY error:

```
onload netperf -L 192.168.228.5 -H 192.168.228.3
oo:netperf[5006]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-16)
See kernel messages in dmesg or /var/log/syslog for more details of this failure
```

ol-alexandra, 2021-06-28:
I'm afraid the answer is "live with it".  Or use a NIC with proper filtering support.

adrienlemasle, 2021-06-28:
Well I'm not sure what NIC has proper filtering support. In fact I've just tried the ENA driver on AWS and I get exactly the same filtering error.

ol-alexandra, 2021-06-28:
I know that Xilinx tested in with one of Intel NICs, only.  I do not remember which one.

adrienlemasle, 2021-06-28:
And here with Intel 82599 with ixgbefv driver :)

```[ 3284.243019] ------------[ cut here ]------------
[ 3284.243024] Driver unsupported XDP return value 4, expect packet loss!
[ 3284.243055] WARNING: CPU: 0 PID: 0 at net/core/filter.c:7130 bpf_warn_invalid_xdp_action+0x3d/0x50
[ 3284.243056] Modules linked in: onload(OE) sfc_char(OE) sfc_resource(OE) sfc(OE) virtual_bus(OE) mtd(OE) sfc_driverlink(OE) mdio vdpa ixgbevf(OE) dummy veth xt_multiport ip6table_filter ip6_tables xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 bpfilter br_netfilter bridge stp llc aufs rpcsec_gss_krb5 auth_rpcgss nfsv4 nfs lockd grace fscache overlay binfmt_misc dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua input_leds serio_raw sch_fq_codel parport_pc ppdev lp parport drm sunrpc ip_tables x_tables autofs4 btrfs blake2b_generic raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel crypto_simd cryptd glue_helper psmouse floppy [last unloaded: virtual_bus]
[ 3284.243093] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           OE     5.8.0-1038-aws #40~20.04.1-Ubuntu
[ 3284.243094] Hardware name: Xen HVM domU, BIOS 4.2.amazon 08/24/2006
[ 3284.243097] RIP: 0010:bpf_warn_invalid_xdp_action+0x3d/0x50
[ 3284.243099] Code: 48 c7 c0 26 ed 69 92 89 fa 48 c7 c6 1e ed 69 92 48 c7 c7 40 ed 69 92 c6 05 c9 20 e9 00 01 48 0f 46 f0 48 89 e5 e8 2b da 18 00 <0f> 0b 5d c3 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 0f 1f 44
[ 3284.243100] RSP: 0018:ffffb1ff80003e18 EFLAGS: 00010286
[ 3284.243102] RAX: 0000000000000000 RBX: ffffb1ff80317588 RCX: 0000000000000027
[ 3284.243102] RDX: 0000000000000027 RSI: 0000000000000086 RDI: ffffa07cffa18cd8
[ 3284.243103] RBP: ffffb1ff80003e18 R08: ffffa07cffa18cd0 R09: 0000000000000004
[ 3284.243104] R10: 0000000000000000 R11: 0000000000000001 R12: ffffa07ce4bfa3b0
[ 3284.243104] R13: ffffb1ff802b5000 R14: 0000000000000000 R15: ffffa07be20b0c00
[ 3284.243106] FS:  0000000000000000(0000) GS:ffffa07cffa00000(0000) knlGS:0000000000000000
[ 3284.243107] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 3284.243107] CR2: 00007fedb51ac014 CR3: 00000003e4b90001 CR4: 00000000001606f0
[ 3284.243109] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 3284.243110] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 3284.243111] Call Trace:
[ 3284.243113]  <IRQ>
[ 3284.243123]  ixgbevf_poll+0xe54/0x1380 [ixgbevf]
[ 3284.243128]  net_rx_action+0x142/0x390
[ 3284.243133]  __do_softirq+0xe1/0x2d6
[ 3284.243135]  asm_call_irq_on_stack+0x12/0x20
[ 3284.243136]  </IRQ>
[ 3284.243143]  do_softirq_own_stack+0x3d/0x50
[ 3284.243146]  irq_exit_rcu+0xa4/0xb0
[ 3284.243152]  sysvec_xen_hvm_callback+0x3b/0x90
[ 3284.243154]  asm_sysvec_xen_hvm_callback+0x12/0x20
[ 3284.243155] RIP: 0010:native_safe_halt+0xe/0x10
[ 3284.243157] Code: 7b ff ff ff eb bd cc cc cc cc cc cc e9 07 00 00 00 0f 00 2d 06 a0 47 00 f4 c3 66 90 e9 07 00 00 00 0f 00 2d f6 9f 47 00 fb f4 <c3> cc 0f 1f 44 00 00 55 48 89 e5 41 55 41 54 53 e8 7d fd 58 ff 65
[ 3284.243157] RSP: 0018:ffffffff92803d48 EFLAGS: 00000246
[ 3284.243158] RAX: 0000000000004000 RBX: 0000000000000001 RCX: 0000000000000034
[ 3284.243159] RDX: 4ec4ec4ec4ec4ec5 RSI: ffffffff92977d20 RDI: ffffa07cfd32d064
[ 3284.243160] RBP: ffffffff92803d50 R08: 000002fcac19866e R09: 00000000000004c6
[ 3284.243160] R10: ffffa07cffa2b3a4 R11: ffffa07cffa2b384 R12: ffffa07cfd32d000
[ 3284.243161] R13: ffffa07cfd32d064 R14: 0000000000000001 R15: 0000000000000001
[ 3284.243166]  ? tick_nohz_next_event+0x7f/0x190
[ 3284.243167]  ? acpi_safe_halt+0x1f/0x30
[ 3284.243168]  acpi_idle_do_entry+0x3d/0x40
[ 3284.243171]  acpi_idle_enter+0xed/0x2b0
[ 3284.243174]  ? xen_sched_clock+0x15/0x20
[ 3284.243177]  cpuidle_enter_state+0x82/0x3b0
[ 3284.243178]  cpuidle_enter+0x2e/0x40
[ 3284.243180]  call_cpuidle+0x23/0x40
[ 3284.243182]  do_idle+0x1db/0x240
[ 3284.243183]  cpu_startup_entry+0x20/0x30
[ 3284.243185]  rest_init+0xae/0xb0
[ 3284.243189]  arch_call_rest_init+0xe/0x1b
[ 3284.243190]  start_kernel+0x6c7/0x6ed
[ 3284.243192]  x86_64_start_reservations+0x24/0x26
[ 3284.243193]  x86_64_start_kernel+0x8b/0x8f
[ 3284.243195]  secondary_startup_64+0xb6/0xc0
[ 3284.243197] ---[ end trace 89f74a6cdcbe1b36 ]---
[ 3284.249898] [sfc efhw] rxclass_get_dev_info: rxclass: Cannot get RX class rule count
[ 3284.249906] [onload] __oof_socket_add_wild: 0:2047 ERROR: FILTER TCP 10.57.10.213:10000 0.0.0.0:0 failed (-95)
[ 3284.267987] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
[ 3284.276784] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
```

maciejj-xilinx, 2021-06-30:
Of Intel NICs we tested with ixgbe and i40e drivers.
I to not think drivers for older Intel NICs will have the necessary support.

We might have done very simple tests (in early stages) with veth and assessed that some level of kernel support for AF_XDP is there. Certainly, we have found it exists for macvlan.

ol-alexandra, 2021-06-30:
I've fixed the first Call Trace above.  The last one looks like an issue in ixgbefv driver, not in Onload.

adrienlemasle, 2021-06-30:
> I've fixed the first Call Trace above. The last one looks like an issue in ixgbefv driver, not in Onload.

Thanks! Yes you are right about ixgbefv. In addition to this, not a lot of drivers have this ntuple filters support . Is there any other more widely supported way that could be used to achieve this filtering?

ol-alexandra, 2021-06-30:
You can run `grep ntuple drivers/net/ethernet/ -r -m1` in the Linux source tree and enjoy a very short list.  Interestingly ixgbe (mentioned by Maciej above) is not in the list.  And i40e is probably the cheapest entry.

adrienlemasle, 2021-06-30:
Ok I think I'll stick to trying to get something out of veth. What can limit the number of onload stacks that can be created? I seems that with veth I can only create a single stack before I get an EBUSY error.

ol-alexandra, 2021-06-30:
You can't use 2 stacks when you have no hardware filtering.  As far as I understand, the first stack gets all the traffic in your case.  I wonder do you get ARP resolved when an Onload stack exists?  We have a way to pass unmatched traffic back to kernel, but it requires to be root IIRC, or to tune some module parameter.

adrienlemasle, 2021-06-30:
Even the single stack doesn't work if I am not root actually.

ol-alexandra, 2021-06-30:
Oh yes, AF_XDP requires to be root!  Then ARP (and any other non-Onload traffic) should be forwarded to kernel properly, but you have no chance of a second stack.

adrienlemasle, 2021-06-30:
Ok funnily enough though if I use the AWS ena driver, I can have as many stacks as I want even after commenting out the filtering error to progress further.

There is something else I don't understand though. When a Solarflare NIC (or in fact a NIC supporting XDP) is installed, you allow accelerating sockets on 127.0.0.1. Why not allow accelerating them anyway without requiring a NIC?

ol-alexandra, 2021-06-30:
Historical reasons, really.  It can be fixed if someone asks kindly enough and/or pays money to Xilinx.  I'll file an internal bug for this, but I guess it won't be a priority.

adrienlemasle, 2021-06-30:
Maybe I should also raise this via Xilinx support also as my company has enterprise onload licenses, many SF NICs and many Xilinx FPGA :)

ol-alexandra, 2021-06-30:
In such a case Xilinx support is the preferred way.  I'll wait for it getting a priority (very likely) and then I'll fix it.

adrienlemasle, 2021-06-30:
Thanks! I asked my questions here originally cause none of these XDP features are in the official Onload yet.

ol-alexandra, 2021-07-01:
So the original issue have been solved, and all the other issues will be moved to official support.  Let's close this.

closed 2021-07-01 by ol-alexandra