Xilinx-CNS/onload #33: Is it possible/useful to build with march/mtune?

haelix888, 2021-06-25:
I looked inside onload_build script and didn't find a "proper" way, so I tried doing the following before invoking onload_build:

```
export CFLAGS='-march=skylake'
export CXXFLAGS='-march=skylake'
also: replace '-mtune=native' with '-march=skylake' inside onload/mk/platform/gnu_x86_64.mk
```

Probably the wrong way as they are redundant, but I do see the arguments picked up in some of the compiler command lines (not all, sometimes I still see them being called with mtune=x86_64).

haelix888, 2021-06-25:
I should add, my experiments so far showed a slowdown, not improvement.

rhughes-xilinx, 2021-06-28:
I can't comment for the others, but it's been a few years since I last tried this personally, and then my results were the same as yours. My hypothesis at the time was that the gcc team don't spend as much time tuning the non-default tunings so, although the optimisation might be mathematically better due to more accurate instruction performance numbers, it turns out to be overall slower due to more complex interactions with the CPU pipeline and caching.

Your experiments aren't really encouraging us to spend a lot of time playing around in this area again.

haelix888, 2021-06-28:
Thanks for the reply.

I have a different theory which I'm hoping to validate, time permitting, in the coming days.
The default onload build already uses `-march=native` which is not a "default" setting - the default for gcc is x86-64. Unlike this gcc default, `march=native` may already produce highly optimised code depending on where you compile onload. In my case I compile on recent hardware and run on even more recent hardware (both Intel skylake but different), so it's possible that by saying `-march=skylake` I have in fact reverted part of the optimisation potential. Skylake is a family of CPU comprising 4-5 subfamilies and spanning 6 years of development, so it's possible that `march=native` results in more optimal code than just `march=skylake`.

So I think the question still stands as it would be great to better expose the `-march` setting to person building onload.

maciejj-xilinx, 2021-08-11:
Worth noting that that -march=skylake (or native and compiling on skylake) will most likely enable MMX/SSE/AVX. These instructions slow down core clock.

haelix888, 2021-09-26:
> Your experiments aren't really encouraging us to spend a lot of time playing around in this area again.

As said previously, it would be great to expose `-march` (and `-O` while at it) to the person building Onload, and let them play with the settings. I was not able to find the toggles in the build scripts.

maciejj-xilinx, 2021-09-27:
Currently, this should be achievable with `export MMAKE_CFLAGS="-march=native"`