Xilinx-CNS/onload #34: error during compilation

antonellocaroli, 2021-06-27:
I get this error after the command:

`make -C "$(mmaketool --toppath)/build/$(mmaketool --userbuild)"`

```
nice cc -I. -I./../../include -I/build/onload/src/include -I/build/onload/src/lib/citools    -fPIC  -m64  -Werror -Wall -Wundef -Wpointer-arith -Wstrict-prototypes -Wnested-externs -Wno-stringop-truncation -Wno-format-truncation -Wimplicit-fallthrough=2 -Wno-array-bounds -Wno-stringop-overflow -Wno-deprecated-declarations -DTRANSPORT_CONFIG_OPT_HDR='<ci/internal/transport_config_opt_extra.h>' -mpclmul -msse4.1 -O2 -g $cflags $cppflags -c /build/onload/src/lib/citools/ippacket.c -o ci_tools_ippacket.o
nice cc -I. -I./../../include -I/build/onload/src/include -I/build/onload/src/lib/citools    -fPIC  -m64  -Werror -Wall -Wundef -Wpointer-arith -Wstrict-prototypes -Wnested-externs -Wno-stringop-truncation -Wno-format-truncation -Wimplicit-fallthrough=2 -Wno-array-bounds -Wno-stringop-overflow -Wno-deprecated-declarations -DTRANSPORT_CONFIG_OPT_HDR='<ci/internal/transport_config_opt_extra.h>' -mpclmul -msse4.1 -O2 -g $cflags $cppflags -c /build/onload/src/lib/citools/namespace.c -o ci_tools_namespace.o
nice cc -I. -I./../../include -I/build/onload/src/include -I/build/onload/src/lib/citools    -fPIC  -m64  -Werror -Wall -Wundef -Wpointer-arith -Wstrict-prototypes -Wnested-externs -Wno-stringop-truncation -Wno-format-truncation -Wimplicit-fallthrough=2 -Wno-array-bounds -Wno-stringop-overflow -Wno-deprecated-declarations -DTRANSPORT_CONFIG_OPT_HDR='<ci/internal/transport_config_opt_extra.h>' -mpclmul -msse4.1 -O2 -g $cflags $cppflags -c /build/onload/src/lib/citools/get_cpu_khz.c -o ci_tools_get_cpu_khz.o
nice cc -I. -I./../../include -I/build/onload/src/include -I/build/onload/src/lib/citools    -fPIC  -m64  -Werror -Wall -Wundef -Wpointer-arith -Wstrict-prototypes -Wnested-externs -Wno-stringop-truncation -Wno-format-truncation -Wimplicit-fallthrough=2 -Wno-array-bounds -Wno-stringop-overflow -Wno-deprecated-declarations -DTRANSPORT_CONFIG_OPT_HDR='<ci/internal/transport_config_opt_extra.h>' -mpclmul -msse4.1 -O2 -g $cflags $cppflags -c /build/onload/src/lib/citools/log_fn.c -o ci_tools_log_fn.o
/build/onload/src/lib/citools/log_fn.c: In function ‘ci_log_stderr’:
/build/onload/src/lib/citools/log_fn.c:30:3: error: ignoring return value of ‘writev’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
   30 |   writev(STDERR_FILENO, v, 2);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/onload/src/lib/citools/log_fn.c: In function ‘ci_log_stdout’:
/build/onload/src/lib/citools/log_fn.c:43:3: error: ignoring return value of ‘writev’ declared with attribute ‘warn_unused_result’ [-Werror=unused-result]
   43 |   writev(STDOUT_FILENO, v, 2);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [/build/onload/mk/after.mk:151: ci_tools_log_fn.o] Error 1
make[2]: Leaving directory '/build/onload/build/gnu_x86_64/lib/citools'
make[1]: *** [/build/onload/src/lib/mmake.mk:19: all] Error 2
make[1]: Leaving directory '/build/onload/build/gnu_x86_64/lib'
make: *** [/build/onload/src/mmake.mk:25: all] Error 2
make: Leaving directory '/build/onload/build/gnu_x86_64
```

ol-alexandra, 2021-06-28:
We disable this warning for Ubuntu, see https://github.com/Xilinx-CNS/onload/blob/master/scripts/mmakebuildtree#L306
You can add a similar fix for your distro, or you can use `make MMAKE_LIBERAL=1` to disable `-Werror`.

Probably we can add `-Wno-unused-result` in all the cases, because we don't support old gcc which does not understand such a parameter... I'll check.

ol-alexandra, 2021-06-29:
No, we do not want to add `-Wno-unused-result`, because we really want such warnings.  However some unreasonable distros require us to check `write()` result.  We are trying to log an error.  What can we do when logging fails?..

So this issue is not going to be fixed unless you provide a patch which sets `W_NO_UNUSED_RESULT` in your particular case, similar to Ubuntu.