Xilinx-CNS/onload #39: Application offloaded to Onload on AF_XDP on Mellanox NiC in Azure does not process traffic and bounces off traffic to kernel stack

shirshen12, 2021-07-19:
Hello Onload Team,

I was able to get Onload to work on Mellanox NiC (SR_IOV mode) in Azure. Please see: https://github.com/Xilinx-CNS/onload/issues/37 for details.

While the app is now being offloaded, we see no traffic being processed by the stack. Please see the following command update:

```
[root@sriov-onload1 ~]# onload_stackdump 
#stack-id stack-name      pids
6         -               -

[root@sriov-onload1 ~]# onload_stackdump stats | grep polls
k_polls: 3789944
u_polls: 0
ioctl_evq_polls: 0
periodic_polls: 2695
interrupt_polls: 3787248
deferred_polls: 0
timeout_interrupt_polls: 0
```
As can be seen, u_polls are ZERO and k_polls are incrementing, meaning userstack poll-mode driver is not being exercised. 

Help is appreciated.

maciejj-xilinx, 2021-07-19:
Hi Shirshendu,

Can you share dmesg to allow assessing whether there is any problem?

I am not familiar wit xdpdump, does "ip link" output confirms no program?

Would you be able to perform the following test for xdp support on hv_netvsc/c5?
 * install a sample xdp program successfully
  * is shown by the diagnostic tools such as xdpdump and "ip link" and
  * actually takes effect?

Kind Regards,
Maciej

shirshen12, 2021-07-21:
Hi @maciejj-xilinx ,

The VMs with the Mellanox NiC do have XDP enabled programs running. I use the xdp-tutorial, packet rewrite program to verify XDP functionality.

Setup the XDP environment and clone the tutorials repo and compile the program
```
sudo yum install clang llvm kernel-headers bpftool
git clone --recurse-submodules https://github.com/xdp-project/xdp-tutorial.git
cd xdp-tutorial
cd packet02-rewriting/
make
```

Load the program and verify if its running:
```
sudo ./xdp_loader -d enP28349s1
Success: Loaded BPF-object(xdp_prog_kern.o) and used section(xdp_port_rewrite)
 - XDP prog attached on device:enP28349s1(ifindex:3)
 - Pinning maps in /sys/fs/bpf/enP28349s1/
 
ip link show dev enP28349s1
3: enP28349s1: <BROADCAST,MULTICAST,SLAVE,UP,LOWER_UP> mtu 1500 xdp qdisc mq master eth0 state UP mode DEFAULT group default qlen 1000
    link/ether 00:0d:3a:7d:9c:68 brd ff:ff:ff:ff:ff:ff
    prog/xdp id 26 
    
xdpdump -D
Interface        Prio  Program name      Mode     ID   Tag               Chain actions
--------------------------------------------------------------------------------------
lo                     <No XDP program loaded!>
eth0                   <No XDP program loaded!>
enP28349s1             xdp_port_rewrite_func native   26   3b185187f1855c4c 

```
As can be seen from above we see xdp program loaded with tag 26

Now we loaded xdp_stats program to see packet stats:

```
./xdp_stats -d enP28349s1
Collecting stats from BPF map
 - BPF map (bpf_map_type:6) id:16 name:xdp_stats_map key_size:4 value_size:16 max_entries:5
XDP-action  
XDP_ABORTED            0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:0.250298
XDP_DROP               0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:0.250233
XDP_PASS               0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:0.250234
XDP_TX                 0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:0.250236
XDP_REDIRECT           0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:0.250239

XDP-action  
XDP_ABORTED            0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:2.000304
XDP_DROP               0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:2.000304
XDP_PASS               0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:2.000304
XDP_TX                 0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:2.000304
XDP_REDIRECT           0 pkts (         0 pps)           0 Kbytes (     0 Mbits/s) period:2.000304

```

maciejj-xilinx, 2021-07-23:
Thanks for all that. As XDP functionality is supported in the system,
I'd expect onload to either get on with installing the program or  some dmesg message
indicating a problem? Could you share dmesg output?

shirshen12, 2021-07-23:
Hi @maciejj-xilinx 

As I had filed a ticket #37, Onload does compile successfully and also I had pasted the dmesg output in there. There were no issues in compiling and also I was able to get Memcached offload to Onload successfully. The problem starts when I observe the poll stats (k_polls vs u_polls) and see u_polls to be zero.

Also, please mlx5_core driver version 5.4.1 has AF_XDP ZC functionality built in.
[image]

So, one thing is puzzling. onload_stackdump does show that stacks are being created but only when i check the u_polls vs k_polls, the counters for u_polls are zero.

maciejj-xilinx, 2021-07-26:
Onload definitely requires its XDP program being installed to intercept the traffic.
Since this the XDP program installation does not take effect for some reason the lack of acceleration is expected.

Results of your experiment show that there is an XDP program support in the system.
It puzzles me that Onload believes it succeeded registering the device, which includes attaching the program to the net device, while system state does not support this.
The BPF program itself is apparently loaded.

Since this seems to be the only thing that is missing I wonder if you could try manually `bpftool net attach` the loaded prog to your device.
Perhaps this would give us some clue.
Should this succeed would this result in accelerated traffic? Should this fail would this give some error message/code?

shirshen12, 2021-07-26:
Hello @maciejj-xilinx 

Please see the following the steps I have done to get xdpsock to be detected by the driver as a XDP program to be loaded.

Before manually attaching: (xdpsock program not listed)
```
[root@sriov-onload2 ~]#bpftool net list dev enP28143s1
xdp:

tc:

flow_dissector:
```

Manually attaching:
```
[root@sriov-onload2 ~]# bpftool net attach xdp name xdpsock dev enP28143s1 
[root@sriov-onload2 ~]#
```

Verify if program is being detected:
```
[root@sriov-onload2 ~]# bpftool net list dev enP28143s1
xdp:
enP28143s1(3) driver id 6

tc:

flow_dissector:
```

Verify using xdpdump:
```
[root@sriov-onload2 ~]# xdpdump -D
Interface        Prio  Program name      Mode     ID   Tag               Chain actions
--------------------------------------------------------------------------------------
lo                     <No XDP program loaded!>
eth0                   <No XDP program loaded!>
enP28143s1             xdpsock           native   6    278a08739491e2c9 
```

dmesg o/p corresponding to above action:
```
[411474.900787] Using feature eBPF/xdp.
[411591.274512] [onload] [4]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
```

Verify onload stack being created:
```
[root@sriov-onload2 ~]# onload_stackdump 
#stack-id stack-name      pids
4         -               -
```

Now, offload memcached:
```
[root@sriov-onload2 ~]# onload -p latency memcached -m 24576 -c 1024 -t 4 -u root -l 10.113.65.36:11211
oo:memcached[561541]: Using Onload <dev-snapshot> [4]
oo:memcached[561541]: Copyright 2019-present Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
```

Now, when we generate traffic for offloaded memcached, again no u_polls, only k_polls and hence no traffic being processed by onload stack and traffic bounced off to kernel.
```
[root@sriov-onload2 ~]# onload_stackdump stats | grep polls
k_polls: 318603
u_polls: 0
ioctl_evq_polls: 0
periodic_polls: 5533
interrupt_polls: 313069
deferred_polls: 0
timeout_interrupt_polls: 0
```

maciejj-xilinx, 2021-07-26:
More telling would be reported number of events.: `onload_stackdump stats | grep evs`  But I suppose this also shows 0 - this would mean no traffic hitting AF_XDP socket.

shirshen12, 2021-07-26:
Ok, so rx_evs stats are not zero:

```
[root@sriov-onload2 ~]# onload_stackdump stats | grep evs
rx_evs: 55990
tx_evs: 0
periodic_evs: 0
interrupt_evs: 0
```

This is perplexing. The u_polls are still zero.

```
[root@sriov-onload2 ~]# onload_stackdump stats | grep polls
k_polls: 1363836
u_polls: 0
ioctl_evq_polls: 0
periodic_polls: 7304
interrupt_polls: 1356531
deferred_polls: 0
timeout_interrupt_polls: 0
```

shirshen12, 2021-07-26:
I have some lead, its an issue with queue binding as seen by MLNX mlx5_core driver. I will give a detailed report shortly.

shirshen12, 2021-07-26:
So, @maciejj-xilinx this is the issue, please do take time to read this entire thread, its very informative.
https://www.spinics.net/lists/xdp-newbies/msg01252.html

maciejj-xilinx, 2021-07-26:
Onload makes available `EF_AF_XDP_ZEROCOPY=0` that allows disabling zerocopy, and
`EF_IRQ_CHANNEL=<channel_no>` to bind to the specific channel

maciejj-xilinx, 2021-08-02:
I was hoping that either:
 * `EF_AF_XDP_ZEROCOPY=0` (to disable zerocopy and use standard queue number), or alternatively
 * `EF_IRQ_CHANNEL=<high number>` (to make it work for zerocopy )
would make it work

Just to double check: the traffic is just standard tcp/ip - no encapsulation, right?

Can you check for presence of relevant hardware filters?

shirshen12, 2021-08-03:
Hi @maciejj-xilinx 

I see no hardware ntuple filters at all, when I do:
`[root@sriov-onload2 ~]# ethtool -n  enP28143s1
`

maciejj-xilinx, 2021-08-05:
We'd expect a hw filter there. Well at least with SFC and Intel NICs. Have you got any intidacation of filter insertion error in dmesg?

If the filter is not there, I'd suggest adding a HW filter manually. For memcached the case is simple a 3 tuple filter for listen socket.

`ethtool --config-ntuple  enP28143s1 flow-type tcp4  dst-ip 10.113.65.36 dst-port 11211 action 4`

Action is queue number. I'd suppose this need to match the queue number Onload is using and good guess would be using the high number for zerocopy.

shirshen12, 2021-08-05:
So, I have not added the hw filter rule, but on further digging for XDP_REDIRECT stats on Mellanox NiCs and I find a massive number of XDP redirects.

```
[root@sriov-onload2 ~]# ethtool -S enP28143s1 | grep redirect
     rx_xdp_redirect: 7175774
     rx_xsk_xdp_redirect: 0
     rx0_xdp_redirect: 7175777
     rx1_xdp_redirect: 0
     rx2_xdp_redirect: 0
     rx3_xdp_redirect: 0
     rx4_xdp_redirect: 0
     rx5_xdp_redirect: 0
     rx6_xdp_redirect: 0
     rx7_xdp_redirect: 0
```

So, this means that Onload's XDP handler insertion program does intercept the traffic and then redirects to kernel stack. 

Let me enable the hw filter and check now.

shirshen12, 2021-08-05:
So, no luck @maciejj-xilinx . I added the hardware filter and see only xdp redirects. I think it needs a deeper fix as mentioned in that earlier mailing list link, which I posted earlier.

maciejj-xilinx, 2021-08-09:
A bit puzzling is that the XDP program does not get attached to the interface nor does the filter.
Does some operation fail and Onload code somehow misses it? Hard to say would be good to print result of each call in `af_xdp_init`.
Let's say the above can be worked around with manual operations as suggested earlier and
we are left with the queue number issue.

From reading the ticket we should use `EF_IRQ_CHANNEL=8` (the same number for ethtool action)

Onload's XDP program uses BPF map to redirect traffic from the bound queue to its XSK socket.
However, Onload does not expect the queue number to be bigger than what it read from the NIC info.

It looks that this code should help.
It would allow queue number 8.
And would protect us in case XDP program gets misreported queue number (e.g.0) - I am not certain what to expect.

```
diff --git a/src/lib/efhw/af_xdp.c b/src/lib/efhw/af_xdp.c
index f3a0a5b56a..b6149eb04c 100644
--- a/src/lib/efhw/af_xdp.c
+++ b/src/lib/efhw/af_xdp.c
@@ -794,6 +794,10 @@ static int af_xdp_init(struct efhw_nic* nic, int instance,
   if( rc < 0 )
     goto out_free_user_offsets;

+  xdp_map_update(nic->af_xdp, instance % nic->vi_lim, file);
+  xdp_map_update(nic->af_xdp, instance + nic->vi_lim, file);
+
+
   /* TODO AF_XDP: currently instance number matches net_device channel */
   rc = xdp_bind(sock, nic->net_dev->ifindex, instance, vi->flags);
   if( rc == -EBUSY ) {
@@ -928,7 +932,7 @@ __af_xdp_nic_init_hardware(struct efhw_nic *nic,
        xdp->vi = (struct efhw_af_xdp_vi*) (xdp + 1);
        xdp->pd = (struct protection_domain*) (xdp->vi + nic->vi_lim);

-       rc = map_fd = xdp_map_create(sys_call_area, nic->vi_lim);
+       rc = map_fd = xdp_map_create(sys_call_area, nic->vi_lim * 2);
        if( rc < 0 )
                goto fail_map;
```

shirshen12, 2021-08-09:
@maciejj-xilinx so, let me generate a patch from the above fragment you posted and test again.

shirshen12, 2021-08-11:
So, @maciejj-xilinx no luck. The hardware filter that we apply in the NiC simply blackholes the traffic. I think, it will be a better idea to debug this in a Azure VM from your side.

maciejj-xilinx, 2021-09-03:
The problem then seems less obvious. If I was to progress on my part I'd need access to at least the NIC model. In some respects it might be preferable to actually try this on bare-metal with the NIC model first. It is fairly likely to have more to do with mlx5 drivers than Azure VM.

shirshen12, 2021-10-09:
@zhiyisun This is very helpful, this might mean something is off in Host side in Azure. As you can see from prior comments, I am trying on CentOS 8, 4.18+ kernel. Maybe I can try on Ubuntu and check and try to reproduce your steps.

shirshen12, 2021-10-14:
Also @maciejj-xilinx and @zhiyisun, to test your work you will need to run a VM with:

1. CentOS Linux 8 with 4.18+ kernel **OR**
2. Ubuntu 20.04 LTS

and upgrade the driver to Mellanox_OFED 5.4+ since this version onwards AF_XDP ZC support is enabled.

shirshen12, 2022-03-23:
@maciejj-xilinx / @zhiyisun currently MSFT Azure is implementing the XDP_REDIRECT clause in Hyper-V PCI driver, I think after that it will start working, because if the dummy driver issue.

shirshen12, 2022-04-14:
The patch has been implemented, https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/?id=1cb9d3b6185b

I will test and report and close this if it works.