Xilinx-CNS/onload #40: difference between   Xilinx-CNS/onload  and   majek/openonload

somewheve, 2021-08-25:
I notice that there are two repositories. and  `Xilinx-CNS/onload `  remove  `<zf/zf.h>` which is used by tcp_direct.   and   `<zf/zf.h>` can be found at  ` majek/openonload.`  

I am confused about this. 

and  https://support-nic.xilinx.com/wp/onload?sd=SF-116303-CD-9&pe=361  chapter6.1 told me to use `<zf/zf.h>`

EnterpriseOnload

EnterpriseOnload source RPMs are available for download below. Access to EnterpriseOnload packages requires an active support contract.
how can I get the RPMs?

somewheve, 2021-08-25:
I want to know how to use  tcp_direct in the repository.

closed 2021-08-25 by somewheve

abower-xilinx, 2021-08-25:
Hi @somewheve,

TCPDirect is not part of this repository - you need to obtain an OpenOnload release package from https://support-nic.xilinx.com/ to use TCPDirect.

The second repository you mention is not maintained by Xilinx.

If you need any help using the released versions of Onload please contact support-nic@xilinx.com

somewheve, 2021-08-25:
got it , thank you

C8LUKA, 2022-02-14:
> Hi @somewheve,
> 
> TCPDirect is not part of this repository - you need to obtain an OpenOnload release package from https://support-nic.xilinx.com/ to use TCPDirect.
> 
> The second repository you mention is not maintained by Xilinx.
> 
> If you need any help using the released versions of Onload please contact [support-nic@xilinx.com](mailto:support-nic@xilinx.com)

sorry to disturb you，It is very kindly to show that anyway or any example to use tcp like tcp_direct with ef_vi or something

somewheve, 2022-02-14:
I would like to ask if using tcp_direct  by rust  is possible ?

C8LUKA, 2022-02-14:
> 

it seem that this repo would not add tcp_direct @_@