Xilinx-CNS/onload #45: No kernel build at '/lib/modules/5.10.0-0.bpo.3-rt-amd64/build'

progettista6, 2021-09-28:
Hello everyone, 
I use a DELL Latitude 5520 computer (Intel i5), with operating system Ubuntu 18.04.5 LTS, Linux kernel 5.10.0-0.bpo.3-rt-amd64 and architecture x86-64. I tried to install OpenOnLoad, following the instructions in the paragraph "Building directly from repository" (https://github.com/Xilinx-CNS/onload/blob/master/DEVELOPING.md). 
In particular there seems to be some problems related to the drivers: as you can see from the figure, the kernel build is not found.

[image]

Instead, running the same instructions on a generic kernel, the whole installation process went smoothly.

Has anyone had the same problem as me? Can you help me?

Thank you,
Best regards,
Emma

ol-alexandra, 2021-09-28:
If you install `linux-image-5.10.0-0.bpo.3-rt-amd64` and want to build any module, you should also install `linux-headers-5.10.0-0.bpo.3-rt-amd64`.  It is organised this way for any Debian/Ubuntu kernel.

abower-amd, 2024-04-16:
This should be resolved with the advice above, thanks! Closing.

closed 2024-04-16 by abower-amd