Xilinx-CNS/onload #46: Openonload support for WSL2

asaiddd, 2021-09-28:
Hi, I am developing a server application for RHEL server with Solarflare NIC and this application runs with openonload. I want to test it with my windows computer on WSL2 to improve developer experience.  When I tried to call scripts/onload_install script it gives a clear error. I would like to ask if WSL2 support exists currently or is it in roadmap. 

onload_install: Building Onload.
mmakebuildtree: No kernel modules at '/lib/modules/4.19.84-microsoft-standard+'
onload_build: FAILED: mmakebuildtree --driver -d x86_64_linux-4.19.84-microsoft-standard+
onload_install: ERROR: Build failed.  Not installing

rhughes-xilinx, 2021-09-29:
I got it building and loading with the instructions at https://docs.darlinghq.org/wsl-build.html with the following tweaks:

- We require that the kernel config file be the usual `./.config`, so copy `Microsoft/config-wsl` there
- We do not require `CONFIG_EMBEDDED=n` but we do require `CONFIG_PPS=m` and `CONFIG_XDP_SOCKETS=y`

As I said, with that I got it loading, but it immediately panicked the kernel upon writing to `afxdp/register`. At least I assume it was a kernel panic - the window disappeared. My Google-fu seems to be failing to tell me how to get hold of the kernel console in WSL. There must be a way, but Microsoft don't seem to want to tell us what it is. Good luck!

rhughes-xilinx, 2021-09-30:
For future reference, WSL stores the last bit of the kernel log in eventvwr in Microsoft/Windows/Hyper-V Workers/Admin. Obviously.

I've pushed 44a4bc363f which moves things along a bit, but there's still more issues. This task isn't on our roadmap, so it's just getting background attention at best - if anybody else would like to move it along faster then feel free.

I believe what's happening is that WSL is putting everything in non-root namespaces (pid, mount, network, etc.), and that's not a configuration that we've previously encountered. I can't find the source of their `init` and my current interest level in reverse-engineering what they're playing at is limited.

My recommendation would be to use a proper Linux machine (or at least a proper VM). Have I mentioned how much I dislike Windows?

asaiddd, 2021-11-30:
@rhughes-xilinx  thank you for your explanation and help! I have go through the steps provided in the [document](https://docs.darlinghq.org/wsl-build.html) but I could not be able to load it. Clearly, even WSL2 is not mature enough to behave like a real linux kernel. But native linux it is not an option for my workplace IT environment. And even with a small amount of kernel knowledge I see that `pstree` output shows that init procedure is not like other linux kernels :) 

Again, thank you for your input. Let's wait for somebody to carry the task on!

sweemer, 2023-02-12:
It would be greatly helpful to me as well to be able to build, link, and run applications with OpenOnload on WSL2 for development purposes. Has there been any progress on this issue in the past year?