Xilinx-CNS/onload #51: correct way to enable hw timestamping

Tehada, 2021-11-12:
I am working with 2 servers with Solarflare NIC on the first one and Mellanox NIC on the second. I want to read hardware timestamps of received packets on both servers. On server with Mellanox NIC I can achieve this by setting appropriate options to setsockopt call and then read timestamps using cmsg mechanism. My application doesn't depend on libvma (Mellanox's alternative to onload) to enable and read hw timestamps. However, when I run the same application on first server with Solarflare NIC, application shows that hw timestamps are zero -- I assume that means that they are disabled. After reading some examles in this repo I added code from [file exchange.c](https://github.com/Xilinx-CNS/onload/blob/master/src/tests/trade_sim/exchange.c#L440):
```
#ifdef USE_ONLOAD
  if (onload_is_present()) {
    onload_stack_opt_set_int("EF_RX_TIMESTAMPING", 3);
    std::cout << "Enabled hw ts for onload.\n";
  }
#endif
```
And with this snippet and when I compile my app with `-lonload` option I get the correct hw timestamps, when running on server with Solarflare adapter.

I want to avoid dependency on onload lib, as it seems that it is possible -- my app works perfectly fine on Mellanox NIC without any external dependencies on libvma or anything. So my question: is it possible to read hw timestamps from application, which uses Solarflare NIC without compiling it with onload lib? If so, what is the correct way to achieve it? I assumed that I should enable hw timestamping on network interface with sfptpd, so I ran it with these config options:
```
timestamping_interfaces enp1s0f0np0
timestamping_disable_on_exit off
```
but it didn't help.

ol-alexandra, 2021-11-12:
Please `export EF_RX_TIMESTAMPING=1` before running your application instead of `onload_stack_opt_set_int()`.  It works without Onload extension API.

Tehada, 2021-11-12:
> Please `export EF_RX_TIMESTAMPING=1` before running your application instead of `onload_stack_opt_set_int()`. It works without Onload extension API.

I will try today, thanks for your help!

Tehada, 2021-11-12:
I checked, this way everything works. Thanks again!

closed 2021-11-12 by Tehada