Xilinx-CNS/onload #52: Could Someone post me private lab Benchmarks of openonload  and compare with VPP?

osevan, 2021-11-18:
Someone got great results on intel NICs with openonload?

For example vpp needs only 2 CPU cores to reach the network cars limit. 

Any Benchmarks available for openonload how efficient they works 

Thanks and

Best regards

osevan, 2021-11-22:
Another quesrion:

How we can manage two nics with openonload,when we want working together? 
,maybe 4 NICs too

rhughes-xilinx, 2021-11-22:
We do not publish benchmarks, sorry. The best performance results will be obtained with a Xilinx/Solarflare NIC, since the AF_XDP compatibility layer used for other NICs is quite costly.

Support for multiple NICs in Onload is transparent and automatic - it will just use the current Linux configuration, including routes, bonding, VLANs, etc.

shirshen12, 2021-11-29:
@osevan please see below:

I can answer this question: 

For memcached on Intel 82599 with Onload-on-AF_XDP
vs
memcached on VMA on Mellanox Connect X 4 Lx, the numbers are:

for 64 byte payload memcached on VMA is 795K TPS
for 64 byte payload memcached on Onload-on-AF_XDP on Intel 82599 is 744K TPS

So performance difference is just 5%. XDP is that good!

shirshen12, 2021-11-29:
> Another quesrion:
> 
> How we can manage two nics with openonload,when we want working together? ,maybe 4 NICs too

You cant do NiC bonding, NiC Teaming with Onload-on-AF_XDP.

ol-alexandra, 2021-11-30:
> You cant do NiC bonding, NiC Teaming with Onload-on-AF_XDP.

It is not true.  It works, and it is tested.
You add the NICs (not the team/bond interface, but the underlying NICs!) to `/sys/module/sfc_resource/afxdp/register`.  You configure team/bond.  (These 2 steps can go in any order, I believe.)  And Onload-on-AF_XDP accelerated traffic over team/bond.
Or you can use 2 NICs separately, without team/bond.  There is a limitation for the NIC number.  IIRC you can't use more than 8 NICs without modifying some constants in the source code.  4 NICs is definitely supported by default.

shirshen12, 2021-11-30:
oh ok, this is news to me @ol-alexandra . I will retest it and let you know. I think I tried adding the two NiCs independently and hoped the bond will pick up. 

From my experience of Onload-on-ef_vi you dont have to do anything and Onload automagically picks it up.