Xilinx-CNS/onload #53: onload is not behaving the same as kernel with signal handlers

sarosharif, 2021-11-19:
I have a program which captures SIGINT and does not close the application.
The application itself is a server which blocks on accept() call. 

While the application is blocking on accept() if it receives SIGINT:

**On kernel:**
 accept does not return 

**On Onload:**  
accept returns and the next statement is executed 

Machine specification are as follows::

    Solarflare Communications XtremeScale SFC9250 NIC
    Operating System: Ubuntu 20.04.1
    Kernel: Linux 5.8.0-55-generic

ol-alexandra, 2021-11-19:
How do you install the signal handler? `sigaction()` with `SA_RESTART` or some other obscure way?
Which Onload version are you using?  The master branch here or some released version?

sarosharif, 2021-11-19:
The handler is installed using this command::
`signal(SIGINT, handle_sigint);`
 
 onload --version gives the following output:
 
```
onload: WARNING: EF_TCP_SYNRECV_MAX=16384 and EF_MAX_ENDPOINTS=8192 are inconsistent.
onload: EF_TCP_SYNRECV_MAX is set to 16384 based on /proc/sys/net/ipv4/tcp_max_syn_backlog value and assuming up to 4 listening sockets in the Onload stack
onload: Too few endpoints requested: ~4 syn-receive states consume one endpoint. 
Onload <dev-snapshot>
Copyright 2019-present Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
Built: Jun 14 2021 05:44:33 (release)
Build profile header: <ci/internal/transport_config_opt_extra.h>
Kernel module: <dev-snapshot>
```

ol-alexandra, 2021-11-19:
I wasn't asking about `onload --version`, sorry.  I was asking where did you get the Onload sources from? The master branch here (may be a few days/weeks ago) or anything else?

Are you using BSD semantic or System V semantic of `signal()`?  See Portability section in [man 2 signal](https://man7.org/linux/man-pages/man2/signal.2.html).  How do you archive this semantic?  What's your distro, libc version?

Our tests show that Onload mimics Linux when an application uses `bsd_signal()` or `sysv_signal()` functions.  Normally, if you compile an application which uses the `signal()` function, it results in calling one of these two.  However some obscure applications may do different things, such as having a private function `signal()` which calls syscall `SYS_signal`.

Please provide a reproducer for your issue with full source code.

sarosharif, 2021-11-22:
It's from the master branch here, last updated on 11 March 2021. Commit number: `4bdaf3ce1a411610302bfae4b08f5f032389efec`

I am working on a linux system that internally uses BSD semantics for signal. The glibc version is as follows: `(Ubuntu GLIBC 2.31-0ubuntu9.2) 2.31`. When the application calls signal() internally this system call is made: `rt_sigaction(SIGINT, {sa_handler=0x55f03c63f409, sa_mask=[INT], sa_flags=SA_RESTORER|SA_RESTART, sa_restorer=0x7f99ab61e210}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0`

Here is the code at can be used to reproduce this problem:
[multi_server.c.txt](https://github.com/Xilinx-CNS/onload/files/7578634/multi_server.c.txt)

ol-alexandra, 2021-11-22:
Thank you for the reproducer, we are exploring it.

Please re-check with the recent Onload code, there were a few signal-releated fixes.  I do not hold my breath, though...

ol-alexandra, 2021-11-26:
@sarosharif Please retest with the patch above.

sarosharif, 2021-11-29:
Thank you
I re-tested with the patch provided. The issue is resolved now.

closed 2021-11-29 by sarosharif