Xilinx-CNS/onload #54: build error on linux 5.15 “va_copy” [-Werror]

davy2900, 2021-11-28:
Hello, i tried to build openonload on CentOS8(linux kernel 5.15, 12th intel). But it fail because of many errors like [-Werror]. How to fix it. Thank you.

ol-alexandra, 2021-11-29:
[Readme](https://github.com/Xilinx-CNS/onload/blob/master/README.md#compatible-linux-kernels-and-distributions-for-xilinx-network-adapters) does not declare compatibility with linux-5.15.  You can propose a patch or wait for my patch (hopefully this year). Currently linux-5.14 works, 5.15 does not.

davy2900, 2021-11-29:
If i install it on linux 4.18, does tcp unload--发自新浪邮箱客户端

&mdash;

davy2900, 2021-11-29:
If I install it on linux 4.18, does tcp onload work with sfn8522(not plus sku). Thanks--发自新浪邮箱客户端

davy2900, 2021-11-29:
If I install it on linux 4.18, does tcp onload work with sfn8522(not plus sku)? Thanks

ol-alexandra, 2021-11-30:
Onload basically works with some linux-5.15 kernels now.  However I believe that some kernels need more fixes.  Please provide the details about the kernel you use.  Is it CentOS 8 Stream default kernel?

And just to be sure I understand the problem correctly: `Unknown symbol in module` condition prints all the meaningful messages to dmesg.  Please show me these messages.

davy2900, 2021-11-30:
CentOS 8 2111. When I try it in vmware, it works well. But When I try it in 12th core Machine, it failures with [failed to find syscall table]. It failures both in linux-5.14 and linux 5.15. Buy it works in linux 4.18.
dmesg:
[ 1062.924897] Solarflare NET driver unloading
[ 1062.966309] Solarflare driverlink driver unloading
[ 1062.994321] Solarflare driverlink driver v5.3.12.1008 API v33.0
[ 1062.996251] Solarflare NET driver v5.3.12.1008
[ 1063.001360] [sfc efrm] init_sfc_resource: ERROR: failed to find syscall table

ol-alexandra, 2021-11-30:
I'll have a look at it when I can get to such a CentOS.
Vmware usually uses a bit different kernel, something with "cloud" in the name.  Can you provide `uname -a` for both working and non-working cases?

davy2900, 2021-11-30:
in 12th machine(failure): Linux localhost.localdomain 5.15.5-1.el8.elrepo.x86_64 #1 SMP Thu Nov 25 08:22:50 EST 2021 x86_64 x86_64 x86_64 GNU/Linux
in vmware (success):Linux localhost.localdomain 5.15.5-1.el8.elrepo.x86_64 #1 SMP Thu Nov 25 08:22:50 EST 2021 x86_64 x86_64 x86_64 GNU/Linux

ol-alexandra, 2021-11-30:
So it is really the same kernel!  I'm surprised, thank you.  I'll have a look when I have a chance.

davy2900, 2021-11-30:
Can I use onload in centos8 and linux 4.18 with sfn8522(not plus sku)? When i use linux 4.18, it says :his error can occur if:
 - no Solarflare network interfaces are active/UP, or they are running
   packed stream firmware or are disabled, and
 - there are no AF_XDP interfaces registered with sfc_resource
Please check your configuration.
And I have type: echo enp5s0 > /sys/module/sfc_resource/afxdp/register    But i don't now is it work. Thanks.

davy2900, 2021-12-01:
I  discover: non-working cases has not "/sys/module/sfc_resource"

rhughes-xilinx, 2021-12-14:
I hope that 0c32bd3893 will fix this for you

ol-alexandra, 2021-12-28:
@davy2900 Do you have any more issues with linux-5.15?  If yes, then please reopen.

closed 2021-12-28 by ol-alexandra