Xilinx-CNS/onload #55: cannot write anything into "/sys/module/sfc_resource/afxdp/register“

davy2900, 2021-11-30:
When i want to use AF_XDP, i follow the next instruction:
"For each interface on which Onload is to use AF_XDP, execute the following:
echo ens2f0 > /sys/module/sfc_resource/afxdp/register“

But i cannot write anything into "/sys/module/sfc_resource/afxdp/register“. I try echo and vim. They have failed.
The file is 777, and i am root. I have "setenforce 0". But I also cannot write anything into  "/sys/module/sfc_resource/afxdp/register“.
[centos 8, linux 5.14.21]
Thanks

shirshen12, 2021-11-30:
Can you let me know for which NiC you are attempting to use Onload-on-AF_XDP.

It works for Intel 82599, ixgbe driver (tested in bare metal by me)
It works for Mellanox Connect X 4 Lx and Connect X 6 X, mlx5_core driver (tested in bare metal by me and also Openstack VMs with KVM hypervisor with PCI passthrough)

davy2900, 2021-12-01:
sfn8522 (not plus), centos 8-2111, linux 4.18

shirshen12, 2021-12-01:
If you are using SFC NiC, then AFAIK know the ef_vi is used directly and no AF_XDP. @ol-alexandra can comment more. AF_XDP interface is only for non SFC NiCs like Intel/Mellanox etc., who have XDP and AF_XDP ZC support in their drivers.

shirshen12, 2021-12-01:
For Intel i225, can you give the detailed instructions of installation and configuration, then maybe I can help.

ol-alexandra, 2021-12-02:
You can pass `enable_driverlink=0` parameter to `sfc_resource` module to disable direct use of any SFC NICs by ef_vi & Onload.  Then you can use those NICs with Onload-AF_XFP.

shirshen12, 2021-12-02:
@ol-alexandra can we have the updated manuals of Onload with Onload-on-AF_XDP. The Onload manuals are extremely comprehensive.

closed 2023-03-14 by abower-amd