Xilinx-CNS/onload #57: Can not reload onload after reboot

asaiddd, 2021-12-08:
Hi,

After rebooting my device, `onload_tool reload` script is not working. Then I tried to uninstall and install it. But I couldnt do it either. You can see the error log below. Seems like my kernel version is changed from `4.18.0-193` to` 4.18.0-305` and some of the kernel function names are changed in the latter one. I checked it in RedHat's website and `4.18.0-305` is the latest available Linux release for RHEL8. 

My question is, is there any compatibility issue in the latest RHEL8 Linux release? How can I solve this compatibility issue on my own? If you have any quick patch to fix it, can you please share it with me?

Thanks in advance,
Ahmet Sait Tekkurt

Picture:
[image]

[onload_install_error.txt](https://github.com/Xilinx-CNS/onload/files/7675817/onload_install_error.txt)

ol-alexandra, 2021-12-08:
Which Onload version are you talking about?  This gitbug repo?  It does not have this issue, it supports both 4.18.0-305 from RHEL 8.4 and 4.18.0-348 from RHEL 8.5.  Or is it about some releases Onload version?  Then please ask for help from Xilinx support.

closed 2021-12-08 by ol-alexandra

asaiddd, 2021-12-11:
@ol-alexandra  thank you for your comment. I realized that I have cloned onload from another repository.