Xilinx-CNS/onload #58: Conan package for user libraries

crucio55, 2021-12-14:
Would you consider adding packages to the [Conan repository](https://github.com/conan-io/conan-center-index) for building the user libraries?
The building process in not particularly complicated, so not much work will be required.

This would greatly simplify our work - and might help many others, too, with easy standardizing of their build system :relaxed:

ol-alexandra, 2021-12-15:
In my humble opinion, building UL and kernel parts separately is a road to hell.

crucio55, 2021-12-15:
Actually, we've been doing that for years, and it's been fine so far (:

It's probably a common use-case - building the executables remotely, and not on the server, that is.

ol-alexandra, 2021-12-15:
We are talking about this github repository, right?  And how do you guarantee that modules and UL come from the same revision?

crucio55, 2021-12-15:
Yeah, of course!
They might be compiled on different GCC versions though

ol-alexandra, 2021-12-15:
Gcc version shouldn't be an issue.  In this case you are free to propose a patch.  I'm almost sure that nobody from Xilinx Onload team is familiar with the Conan.

crucio55, 2021-12-15:
The thing is that the Conan recipe is pushed to their repository that I linked - and I don't think I can propose a patch there on behalf of Xilinx Onload... :sweat: 

Even if I do end up doing it, I'll probably have questions along the way. For example, get links for the source tarballs - there are no Github releases to take their links, and on the [official page](https://support-nic.xilinx.com/wp/onload) I couldn't find links that would work with `wget` and the like

Anyway, even if I build the initial Conan support, I won't exactly stay to maintain it, so that's something to consider for the Onload Xilinx team.

abower-xilinx, 2021-12-15:
Thanks for the suggestion @crucio55! It's not realistic for us to maintain such a contribution either so I'm going to close this issue.

However, we would be happy to consider pull requests on this repository that helped anyone else working on packaging or integration of Onload.

closed 2021-12-15 by abower-xilinx