Xilinx-CNS/onload #59: multicast: packets go through the kernel (onload_stackdump - empty)

0xdd7fq, 2022-01-10:
Why multicast apps doesn't work throuth onload? Any help welcome.

```
net@lizard:~$ onload --version
Onload 7.1.2.141
Copyright 2019-2021 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
Built: Jan  2 2022 09:12:03 (release)
Build profile header: <ci/internal/transport_config_opt_extra.h>
Kernel module: 7.1.2.141

onload /.mcast-track -i bond0 233.26.38.5:6005
2022/01/10 08:39:52.584457 Subscribed to 233.26.38.5 at bond0
2022/01/10 08:39:53.584329 | >233.26.38.5	| No datagrams for last 1s
2022/01/10 08:39:54.584478 | >233.26.38.5	| datagrams for last 1s: 43

net@lizard:~$ onload nc -l 6666
oo:nc.openbsd[96276]: Using OpenOnload 7.1.2.141 [6]
oo:nc.openbsd[96276]: Copyright 2019-2021 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
oo:nc.openbsd[96276]: Importing OpenOnload 7.1.2.141 [7,c96276-c0]
oo:nc.openbsd[96276]: Copyright 2019-2021 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks

```

0xdd7fq, 2022-01-10:
@ol-alexandra can you help ?

abower-xilinx, 2022-01-10:
Hi 0xdd7fq,
Please e-mail support-nic@xilinx.com for help with a released Onload version on a Xilinx NIC, providing as much information as possible including an 'sfreport' - see tools available from support-nic.xilinx.com .

closed 2022-01-23 by 0xdd7fq