Xilinx-CNS/onload #62: AF XDP with Amazon ENA driver doesn't seem to work

ShayAgros, 2022-02-15:
Hi,
thank you for the work you're putting into this project, and I'm sorry in advance if I'm missing the whole point somewhere here.

I'm testing a beta version of Amazon's Linux ENA driver which introduces support for AF XDP zero-copy functionality. I installed onload using the following steps:

```bash
$ # launch an Ubuntu 20.04 instance (VM) in AWS EC2 service. The distribution comes with AWS flavored Ubuntu
$ uname -r
  5.11.0-1022-aws
$ # Compile and install Linux kernel from source, version 5.12. This time made MTD to be part of kernel and not module. Otherwise this module doesn't seem to actually come as module (though config file claims does)
$ git clone https://github.com/Xilinx-CNS/onload.git && cd onload
$ sudo scripts/onload_install
$ sudo onload_tool reload
  onload_tool: /sbin/modprobe -r onload
  onload_tool: /sbin/modprobe -r sfc
  onload_tool: /sbin/modprobe sfc
  onload_tool: /sbin/modprobe onload
$ dmesg
  ...
  [ 3627.823124] Efx driverlink unregistering resource driver
  [ 3627.891398] Solarflare NET driver unloading
  [ 3627.975052] Solarflare driverlink driver unloading
  [ 3628.034403] Solarflare driverlink driver v5.3.12.1010 API v33.0
  [ 3628.040754] Solarflare NET driver v5.3.12.1010
  [ 3628.062379] Efx driverlink registering resource driver
  [ 3628.138805] [onload] Onload 01899428 2022-02-14 master 
  [ 3628.142841] [onload] Copyright 2019-2022 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
  [ 3628.253613] onload_cp_server[52143]: Spawned daemon process 52172
$ echo ens6 | sudo tee /sys/module/sfc_resource/afxdp/register
  ens6
$ dmesg
  [ 3700.143341] [sfc efrm] efrm_nondl_register_device: register ens6
  [ 3700.147870] [sfc efrm] ens6 type=4:
  [ 3700.171485] ena 0000:00:06.0 ens6: Local page cache is disabled when using XDP
  [ 3700.191012] ena 0000:00:06.0 ens6: XDP program is set, changing the max_mtu from 9216 to 3498
  [ 3700.191059] [sfc efrm] ens6 index=0 ifindex=11
  [ 3700.191070] [onload] oo_nic_add: ifindex=11 oo_index=0
```

Looking at the logs and checking that the ENA interface indeed has an XDP program loaded seems to indicate that the operations succeeded. However when running a simple iperf command, I get the following output:
```bash
$ onload iperf -c 172.31.48.72 -M 64 -t 5000 -i 1
  onload: WARNING: EF_TCP_SYNRECV_MAX=16384 and EF_MAX_ENDPOINTS=8192 are inconsistent.
  onload: EF_TCP_SYNRECV_MAX is set to 16384 based on /proc/sys/net/ipv4/tcp_max_syn_backlog value and assuming up to 4   listening sockets in the Onload stack
  onload: Too few endpoints requested: ~4 syn-receive states consume one endpoint. 
  oo:iperf[54103]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-1)
  See kernel messages in dmesg or /var/log/syslog for more details of this failure
  WARNING: attempt to set TCP maxmimum segment size to 64 failed.
  Setting the MSS may not be implemented on this OS.
```

and running `dmesg` yields:
```kmsg
[ 4136.527928] [onload] [5]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
[ 4136.557765] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
[ 4136.562146] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
```

The `iperf` traffic *does* go through but looking at the driver's stats I see that all packets received XDP_PASS instruction which means that they don't use AF XDP (which requires XDP_REDIRECT instruction).
If I run the same iperf command as super-user (e.g.
`onload sudo iperf -c 172.31.48.72 -M 64 -t 5000 -i 1`
the 'FIXME' and WARNING message above don't show up, but otherwise it looks like still no traffic is being handled by AF XDP.

Same behavior was observed using netcat client-server and a simple client-server C program. I'm guessing the DMA flush is called from an AF XDP.

Can you tell me if I'm doing something wrong here ? Or if you spot some missing XDP behavior in the ENA driver ?

ol-alexandra, 2022-02-15:
ena does not support n-tuple filtering, which is required for Onload-over-AF_XDP.

https://github.com/Xilinx-CNS/onload/issues/28#issuecomment-1016760006

And Onload should document it...

maciejj-xilinx, 2022-02-28:
Hello [ShayAgros](https://github.com/ShayAgros)

We have updated the code base with likely workaround see #28

arieb, 2022-03-06:
hey @ShayAgros, i just read this after posting https://github.com/Xilinx-CNS/onload/issues/64 your comment about 
```
$ # Compile and install Linux kernel from source, version 5.12. This time made MTD to be part of kernel and not module. Otherwise this module doesn't seem to actually come as module (though config file claims does)
```
seems to be the fix for my problem. any chance you have an AMI you can share with the proper kernel? (or a guide on recompiling the ubuntu kernel with all the aws customizations)

ShayAgros, 2022-03-08:
Hi @arieb 

I'm doing the following:
```shell
$ git clone https://github.com/torvalds/linux.git ~/linux && cd ~/linux
$ git checkout --branch 5.12 v5.12
$ cp /boot/config-`uname -r` ./.config
$ ./scripts/config --set-str CONFIG_SYSTEM_TRUSTED_KEYS ""
$ ./scripts/config -e CONFIG_MTD
$ make -j $(getconf _NPROCESSORS_ONLN)
$ make -j $(getconf _NPROCESSORS_ONLN) modules_install
$ make -j $(getconf _NPROCESSORS_ONLN) install
$ sudo cp .config /boot/config-$(make kernelversion)
$ reboot
```
This should boot with the newly compiled Linux version

@maciejj-xilinx Thanks for doing it. I'll test it when I get some time for it. Resolving for now

closed 2022-03-08 by ShayAgros

wrfeewrtgaqwfrwaq, 2022-09-23:
So I'm also trying to get onload to work on AWS. Everything seems to work but no traffic goes through and dmesg shows:
```
eth0: Command parameter 46 is not supported
[sfc efhw] rxclass_get_dev_info: rxclass: Cannot get RX class rule count
[onload] oof_local_port_addr_fixup_wild: 1:2047 ERROR: FILTER TCP 172.31.0.132:45907 0.0.0.0:0 failed (-95)
```

aurosglobal, 2023-07-05:
> hey @ShayAgros, i just read this after posting #64 your comment about
>
> ```
> $ # Compile and install Linux kernel from source, version 5.12. This time made MTD to be part of kernel and not module. [...]

For anyone else coming across this thread, the `mtd.ko` is typically in `linux-modules-extra` package. There should be no need to recompile your kernel.