Xilinx-CNS/onload #64: onload_install reload fails on unknown symbol 'mtd_device_unregister' 'mtd_device_parse_register'

arieb, 2022-03-06:
Hello,

Im building the latest release (onload-7.1.3.202) from source on ubuntu 20.04 LTS on AWS C4 instance (with intel nic).
```
$ uname -a
Linux ip-60-0-46-192 5.11.0-1028-aws #31~20.04.1-Ubuntu SMP Fri Jan 14 14:37:50 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```
```
$ ethtool -i ens3
driver: ixgbevf
version: 5.11.0-1028-aws
firmware-version:
expansion-rom-version:
bus-info: 0000:00:03.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: yes
```

Building completes fine, but when running
`sudo onload_tool reload`
it fails with 
```
ubuntu@hostname:~/onload-7.1.3.202$ sudo onload_tool reload
onload_tool: /sbin/modprobe onload
modprobe: ERROR: could not insert 'sfc': Unknown symbol in module, or unknown parameter (see dmesg)
modprobe: ERROR: ../libkmod/libkmod-module.c:990 command_do() Error running install command '/sbin/modprobe --ignore-install sfc  &&             if [ -e /lib/modules/$(uname -r)/extra/sfc_aoe.ko
]; then               /sbin/modprobe sfc_aoe;             fi' for module sfc: retcode 1
modprobe: ERROR: could not insert 'onload': Invalid argument
onload_tool: FAILED: /sbin/modprobe onload
```

dmesg
```
[  671.578225] sfc: loading out-of-tree module taints kernel.
[  671.578497] sfc: module verification failed: signature and/or required key missing - tainting kernel
[  671.578581] sfc: Unknown symbol mtd_device_unregister (err -2)
[  671.578683] sfc: Unknown symbol mtd_device_parse_register (err -2)
```

i tried with previous release (Onload-7.1.2.141) but no luck either

rhughes-xilinx, 2022-03-07:
I see from that other bug that you've now got it loaded, but I'd still like to explore easier ways.

Did you trying manually `modprobe`ing `mtd` before `onload_tool reload` (i.e. could it be a problem with dependencies?). Does `mtd.ko` even exist in `/lib/modules`? If you do modprobe it, do `mtd_device_unregister` and `mtd_device_parse_register` show up in `/proc/kallsyms`?

arieb, 2022-03-07:
> I see from that other bug that you've now got it loaded, but I'd still like to explore easier ways.
>
> Did you trying manually `modprobe`ing `mtd` before `onload_tool reload` (i.e. could it be a [...]

i actually dont have it working yet - im building the kernel from source with MTD now included and not a module.
to answer you question, there is no `mtd.ko` in /lib/modules

kieranm-xilinx, 2022-03-07:
I have a machine running  Ubuntu 20.04.3 LTS (GNU/Linux 5.4.0-92-generic x86_64) and have no problems accessing the mtd module without needing to rebuild the kernel:

$ lsmod | grep mtd
mtd                    65536  13 sfc

$ find /lib/modules/5.4.0-92-generic/kernel/ -name mtd.ko
/lib/modules/5.4.0-92-generic/kernel/drivers/mtd/mtd.ko

$ dpkg-query -S /lib/modules/5.4.0-92-generic/kernel/drivers/mtd/mtd.ko
linux-modules-extra-5.4.0-92-generic: /lib/modules/5.4.0-92-generic/kernel/drivers/mtd/mtd.ko

arieb, 2022-03-07:
` dpkg-query -S /lib/modules/5.4.0-92-generic/kernel/drivers/mtd/mtd.ko`
this is what is was missing!
i installed this package and mtd.ko module now exists.
`onload_tool reload` seems to work now :)
thank you!