Xilinx-CNS/onload #66: Kubernetes support

variantf, 2022-03-07:
Hi, I'm wondering if I can use tcpdirect/efvi inside kubernetes cluster with multus? I can't find any document or example for now. 

I found there was a device plugin for solareflare nics here https://github.com/vikaschoudhary16/sfc-device-plugin. However, I read through the codebase and found that looks like dedicating one nic to one pod.  I'm hopping if there's a way the pod can share the same nic, keep the performance on k8s host without using `hostNetwork: true`

abower-xilinx, 2022-03-07:
Hi @variantf, there is some multus support available under "Kubernetes Onload" at https://support-nic.xilinx.com/wp/onload but I would advise contacting support-nic@xilinx.com if you want to use Onload and Kubernetes for a project.

variantf, 2022-03-07:
Thank you @abower-xilinx I'll look into it or may be buy a commercial license for further support.

closed 2022-03-07 by variantf