Xilinx-CNS/onload #67: latest master branch: /sys/module/sfc_resource/afxdp/register: Permission denied

arieb, 2022-03-14:
failed to register ens3 nic while using latest master branch on AWS c4 instance running intel NIC
```
$ ifconfig
ens3: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9001
```

```
$ ethtool -i ens3
driver: ixgbevf
version: 5.13.0-1014-aws
firmware-version:
expansion-rom-version:
bus-info: 0000:00:03.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: yes
```

```
$ sudo echo ens3 > /sys/module/sfc_resource/afxdp/register
-bash: /sys/module/sfc_resource/afxdp/register: Permission denied
```

nothing on dmesg
```
[  661.374381] [onload] Onload
[  661.376093] [onload] Copyright 2019-2022 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
[  661.484810] onload_cp_server[1098]: Spawned daemon process 1117
```

rhughes-xilinx, 2022-03-23:
That redirection is being done by your shell, not by `sudo`. Here's an appropriate explanation of what's going on, and some alternatives: https://stackoverflow.com/questions/20895619/why-cant-i-echo-contents-into-a-new-file-as-sudo

arieb, 2022-03-23:
Thanks @rhughes-xilinx - i completely missed that one. 
for what its worth, i registered my NIC properly this time using 
`sudo sh -c 'echo ens3f0 > /sys/module/sfc_resource/afxdp/register'`

but NIC init failed with
```
[  199.588305] [sfc efrm] nondl_register_store: cannot detect number of channels for device ens3 assuming 1
[  199.588309] [sfc efrm] efrm_nondl_register_device: register ens3
[  199.591295] [sfc efrm] ens3 type=4:
[  199.591497] [sfc efrm] ?: ERROR: hardware init failed rc=-22
[  199.594015] [sfc efrm] ens3 index=0 ifindex=2
[  199.594041] [onload] oo_nic_add: ifindex=2 oo_index=0
```

rhughes-xilinx, 2022-03-23:
AFAICT ixgbevf has no support for AF_XDP zero-copy, thus cannot work with Onload.

It would be technically possible to make Onload work with drivers without zero-copy support, but that's not something we're currently looking at. If you're feeling (very) ambitious, patches welcome...