Xilinx-CNS/onload #72: Please add the feature of handling transparently multiple threads binding to multiple RSS queues on non-Solarflare NiCs

shirshen12, 2022-04-02:
Hello,

Please add the feature of where Onload in AF_XDP mode can transparently create multiple stacks bound to multiple RSS queues for multiple threads transparently while maintaining all the networking semantics like SO_REUSEPORT etc., 

This issue is already in https://github.com/Xilinx-CNS/onload/issues/70, in detail.

Any help is appreciated.

shirshen12, 2022-04-06:
Hello @maciejj-xilinx can you take a look at it ?

maciejj-xilinx, 2022-04-11:
I had another look into the specifics of the implementation.
Currently, it seems there are two relatively contained changes needed:
 * enable use of RSS/Clustering for AF_XDP (it is disabled currently)
 * suppress use of hardware for non-RSS/non-Clustered stacks
   as currently only single stack per channel can be accelerated and network driver channels need to be saved for the cluster

It looks just these two should be enough to enable both passive and active side RSS scalable filters.
The active side would depend on configurable Toeplitz RSS hashing on the Interface. It looks both Intel and Mellanox NICs allow this.

shirshen12, 2022-04-11:
sounds good @maciejj-xilinx . How do we enable, RSS/Clustering for AF_XDP ?

maciejj-xilinx, 2022-04-13:
These two points list required code changes. I reckon relatively simple ones though. This is on my AF_XDP TODO list.

shirshen12, 2022-04-13:
sounds good @maciejj-xilinx

shirshen12, 2022-04-26:
Hi @maciejj-xilinx any progress on the patch ?

nharua, 2025-10-20:
Hi @maciejj-xilinx any update for this ticket?