Xilinx-CNS/onload #74: why oo:sfnt-pingpong[7699]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-22) ?

760985933, 2022-04-13:
oo:sfnt-pingpong[7699]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-22)
See kernel messages in dmesg or /var/log/syslog for more details of this failure

shirshen12, 2022-04-13:
Can you give me more details of your environment ?

760985933, 2022-04-13:
[image]

[image]

760985933, 2022-04-13:
[image]

760985933, 2022-04-13:
onload -p latency  ./sfnt-pingpong --affinity "1;1"   
I dont understand --affinity "1;1"       1;1 waht does that mean?
thanks!

shirshen12, 2022-04-13:
I think for SFC NiCs you must contact their support. The "1;1" is actually about core pinning. You might want to read the manual downloadable from here: https://support-nic.xilinx.com/wp/onload. Also mentioned here: https://github.com/Xilinx-CNS/onload/issues/47

760985933, 2022-10-11:
您好，我已收到您发送的邮件，等待我查看内容后给您回复。