Xilinx-CNS/onload #78: EC2 ENA Onload error

mikhail-lapshin, 2022-05-17:
Hi Onload team,

I have an error when starting a process with onload
**ERROR: ld.so: object 'libonload.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.**

I'm running EC2 node with latest Ubuntu 20.04 with amazon ENA NIC, rebuilt kernel 5.16, installed the latest Onload. Did the same as #62
Eventually I have 
```
$ sudo dmesq
[  +0.075612] Solarflare driverlink driver v5.3.12.1023 API v33.0
[  +0.004554] Solarflare NET driver v5.3.12.1023
[  +0.011961] Efx driverlink registering resource driver
[  +0.033999] [onload] Onload 7309f876 2022-05-13 master 
[  +0.003706] [onload] Copyright 2019-2022 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
[ +33.346988] [sfc efrm] efrm_nondl_register_device: register ens5
[  +0.004034] [sfc efrm] ens5 type=4:\x000 ifindex=2
[  +0.002978] ena 0000:00:05.0 ens5: XDP program is set, changing the max_mtu from 9216 to 3498
[  +0.000015] [sfc efrm] ens5 index=0 ifindex=2
[  +0.000045] [onload] oo_nic_add: ifindex=2 oo_index=0
```
Also syslog has 
`onload_cp_server[354964]: Accelerating ens5: RX 1 TX 1`

```
$ ethtool -i ens5 
driver: ena
version: 5.16.0
firmware-version: 
expansion-rom-version: 
bus-info: 0000:00:05.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

```
$ ethtool -S ens5 | grep xdp
     queue_0_rx_xdp_aborted: 0
     queue_0_rx_xdp_drop: 0
     queue_0_rx_xdp_pass: 2644
     queue_0_rx_xdp_tx: 0
     queue_0_rx_xdp_invalid: 0
     queue_0_rx_xdp_redirect: 0
     queue_1_xdp_tx_cnt: 0
     queue_1_xdp_tx_bytes: 0
     queue_1_xdp_tx_queue_stop: 0
     queue_1_xdp_tx_queue_wakeup: 0
     queue_1_xdp_tx_dma_mapping_err: 0
     queue_1_xdp_tx_linearize: 0
     queue_1_xdp_tx_linearize_failed: 0
     queue_1_xdp_tx_napi_comp: 1
     queue_1_xdp_tx_tx_poll: 1
     queue_1_xdp_tx_doorbells: 0
     queue_1_xdp_tx_prepare_ctx_err: 0
     queue_1_xdp_tx_bad_req_id: 0
     queue_1_xdp_tx_llq_buffer_copy: 0
     queue_1_xdp_tx_missed_tx: 0
     queue_1_xdp_tx_unmask_interrupt: 1
```

But when I start iperf it reports an error and fallback to default stack
```
$ onload sudo iperf -c 172.31.28.95 -M 64 -t 20 -i 1
ERROR: ld.so: object 'libonload.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
WARNING: attempt to set TCP maxmimum segment size to 64 failed.
Setting the MSS may not be implemented on this OS.
```
Any idea on why that might happen?

My build steps for Onload installation
```
$ git clone https://github.com/Xilinx-CNS/onload.git && cd onload
$ sudo scripts/onload_install
$ sudo onload_tool reload

$ sudo ip link set dev ens5  mtu 3498
$ sudo ethtool -L ens5 combined 1

$ echo ens5 | sudo tee /sys/module/sfc_resource/afxdp/register
```

kieranm-xilinx, 2022-05-17:
Does `sudo onload iperf...` work (ie. change the order of sudo and onload)

mikhail-lapshin, 2022-05-17:
Yep, that works! Thanks.

closed 2022-05-17 by mikhail-lapshin