Xilinx-CNS/onload #82: support for CentOS Stream 9

ajschorr, 2022-06-22:
Hi,

I downloaded onload-7.1.3.202 to install on a CentOS Stream 9 system, but the installation fails.
So I grabbed the git tree to see whether it yet has support for the RHEL 5.14 kernel, but the build
also fails. I have attached a log of the output from "make -k" to show where it's crapping out.
It looks like most of the trouble is coming from the lack of a declaration or definition of PDE_DATA.
I'm not certain, but it seems as if the fix for that may simply be to rename PDE_DATA to pde_data:
https://lore.kernel.org/linux-mm/20220122061423.HMq1py5Q0%25akpm@linux-foundation.org/

In addition, there's an issue with [make.log](https://github.com/Xilinx-CNS/onload/files/8953833/make.log) being undefined. 
That one I think can just be replaced with netif_rx:
https://lore.kernel.org/netdev/20220306215753.3156276-10-bigeasy@linutronix.de/

Is there a plan to patch these and issue a new release soon with RHEL 9 (CentOS Stream 9) support?

Thanks,
Andy

ajschorr, 2022-06-22:
Oops, that got botched a bit. The attachment "make.log" shows the build errors.
But it's netif_rx_ni that needs to be replaced with netif_rx.

Regards,
Andy

ajschorr, 2022-07-14:
With these 2 patches, I can compile successfully. Unfortunately, I don't see an obvious way to use #ifdef to make these conditional, since it's the same kernel version in RHEL 9 and CentOS Stream 9 (5.14.0). Is there a way to test the patch level?

[pde_data.patch.txt](https://github.com/Xilinx-CNS/onload/files/9113396/pde_data.patch.txt)
[netif_rx.patch.txt](https://github.com/Xilinx-CNS/onload/files/9113397/netif_rx.patch.txt)

ol-alexandra, 2022-07-14:
A proper version of your first patch is here: #86

ol-alexandra, 2022-07-14:
@ol-sergeyn  The netif_rx.patch.txt patch above comes from linux-5.18, which can't be supported properly yet, because sfc net driver breaks for now.

ajschorr, 2022-07-14:
Thanks for the feedback. When I look a the diff between the RHEL 9 and CS 9  kernel source trees, I see that they have replaced netif_rx_ni with netif_rx everywhere. So why wouldn't that work? They seem to have backported some patches from 5.17 and later. Couldn't kernel_compat.sh be patched to test for the existence of netif_rx_ni?

abower-xilinx, 2022-07-21:
Thanks for

closed 2022-11-20 by abower-amd