Xilinx-CNS/onload #83: AF_XDP on ConnectX-5

boyracer2020, 2022-06-22:
Using onload (clone from repo on 06/20) I am seeing : 
```
oo:tcp_pingpong[2241]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-1)
See kernel messages in dmesg or /var/log/syslog for more details of this failure
oo:tcp_pingpong[2241]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-1)
See kernel messages in dmesg or /var/log/syslog for more details of this failure
Bound
```
and 
```
Jun 22 10:33:43 ubuntu-thinkcentre2 kernel: [ 3354.109987] [onload] [0]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
Jun 22 10:33:43 ubuntu-thinkcentre2 kernel: [ 3354.128545] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
Jun 22 10:33:43 ubuntu-thinkcentre2 kernel: [ 3354.128552] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
```
looking thru the issues, I have enabled ntuples. No changes. 

What's missing ? Do I need to install special MLNX drivers ? 

OS: Ubuntu 20

thanks, M

shirshen12, 2022-06-26:
@boyracer2020 mention steps to reproduce the issue, what is your environment ? Bare-metal, Azure VMs or normal VMs running on KVM with PCI Passthrough with Mellanox ?

boyracer2020, 2022-06-27:
hi @shirshen12 , it's a bare-metal host running standard Ubuntu 20.04

shirshen12, 2022-06-27:
Ok, please see exact instructions to setup, please let me know if its works:

```
# upgrade to latest OS kernel
sudo apt update -y
sudo apt upgrade -y
sudo apt full-upgrade -y

# reboot into new kernel
reboot

sudo apt install build-essential net-tools unzip libcap-dev linux-tools-common linux-tools-generic netperf libevent-dev libnl-route-3-dev tk bison tcl libnl-3-dev flex libnl-route-3-200 dracut python2 libpcap-dev -y
sudo apt install initramfs-tools -y

# build the Mellanox driver, mlx5_core
wget --no-check-certificate https://content.mellanox.com/ofed/MLNX_OFED-5.6-1.0.3.3/MLNX_OFED_LINUX-5.6-1.0.3.3-ubuntu20.04-x86_64.tgz
tar zxf MLNX_OFED_LINUX-5.6-1.0.3.3-ubuntu20.04-x86_64.tgz
cd MLNX_OFED_LINUX-5.6-1.0.3.3-ubuntu20.04-x86_64
sudo ./mlnxofedinstall --force --add-kernel-support --vma --tmpdir /mnt/
dracut -f

# reboot into new driver (optional)
reboot

# Install Onload:
git clone https://github.com/Xilinx-CNS/onload.git
cd onload
scripts/onload_mkdist --release
cd onload-<version>/scripts/
./onload_install
./onload_tool reload

# register the NiC with AF_XDP driver interface
echo enp2s0f0 > /sys/module/sfc_resource/afxdp/register
echo enp2s0f1 > /sys/module/sfc_resource/afxdp/register
```

boyracer2020, 2022-06-28:
did all of that , see below , still no luck :
```
Welcome to Ubuntu 20.04.4 LTS (GNU/Linux 5.13.0-51-generic x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

0 updates can be applied immediately.

Your Hardware Enablement Stack (HWE) is supported until April 2025.
Last login: Mon Jun 27 21:51:53 2022 from 192.168.0.102
fischer@munich-thinkcentre1:~$ sudo apt update -y
[sudo] password for fischer:
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Hit:2 http://de.archive.ubuntu.com/ubuntu focal InRelease
Get:3 http://de.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:4 http://security.ubuntu.com/ubuntu focal-security/main amd64 DEP-11 Metadata [40,7 kB]
Get:5 http://de.archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]
Get:6 http://security.ubuntu.com/ubuntu focal-security/universe amd64 DEP-11 Metadata [66,3 kB]
Get:7 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 DEP-11 Metadata [2.464 B]
Get:8 http://de.archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [1.935 kB]
Get:9 http://de.archive.ubuntu.com/ubuntu focal-updates/main amd64 DEP-11 Metadata [278 kB]
Get:10 http://de.archive.ubuntu.com/ubuntu focal-updates/main amd64 c-n-f Metadata [15,6 kB]
Get:11 http://de.archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [924 kB]
Get:12 http://de.archive.ubuntu.com/ubuntu focal-updates/universe i386 Packages [677 kB]
Get:13 http://de.archive.ubuntu.com/ubuntu focal-updates/universe amd64 DEP-11 Metadata [391 kB]
Get:14 http://de.archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 DEP-11 Metadata [944 B]
Get:15 http://de.archive.ubuntu.com/ubuntu focal-backports/main amd64 DEP-11 Metadata [7.964 B]
Get:16 http://de.archive.ubuntu.com/ubuntu focal-backports/universe amd64 DEP-11 Metadata [30,5 kB]
Fetched 4.706 kB in 3s (1.619 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
fischer@munich-thinkcentre1:~$ sudo apt upgrade -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libfwupdplugin1 libpcre2-32-0
Use 'sudo apt autoremove' to remove them.
The following packages will be upgraded:
  ubuntu-advantage-tools
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 876 kB of archives.
After this operation, 21,5 kB of additional disk space will be used.
Get:1 http://de.archive.ubuntu.com/ubuntu focal-updates/main amd64 ubuntu-advantage-tools amd64 27.9~20.04.1 [876 kB]
Fetched 876 kB in 1s (1.079 kB/s)
Preconfiguring packages ...
(Reading database ... 176920 files and directories currently installed.)
Preparing to unpack .../ubuntu-advantage-tools_27.9~20.04.1_amd64.deb ...
Unpacking ubuntu-advantage-tools (27.9~20.04.1) over (27.8~20.04.1) ...
Setting up ubuntu-advantage-tools (27.9~20.04.1) ...
Installing new version of config file /etc/ubuntu-advantage/uaclient.conf ...
Created symlink /etc/systemd/system/multi-user.target.wants/ubuntu-advantage.service → /lib/systemd/system/ubuntu-advantage.service.
Processing triggers for man-db (2.9.1-1) ...
fischer@munich-thinkcentre1:~$ sudo apt full-upgrade -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libfwupdplugin1 libpcre2-32-0
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
fischer@munich-thinkcentre1:~$ sudo apt install build-essential net-tools unzip libcap-dev linux-tools-common linux-tools-generic netperf libevent-dev libnl-route-3-dev tk bison tcl libnl-3-dev flex libnl-route-3-200 dracut python2 libpcap-dev -y
Reading package lists... Done
Building dependency tree
[... 370 lines cut ...]
root@munich-thinkcentre1:/home/fischer# echo enp6s0f0np0 > /sys/module/sfc_resource/afxdp/register
root@munich-thinkcentre1:/home/fischer# echo enp6s0f1np1 > /sys/module/sfc_resource/afxdp/register
root@munich-thinkcentre1:/home/fischer# exit
exit
fischer@munich-thinkcentre1:~$ sudo /sbin/ifconfig enp6s0f0np0 10.0.0.1 netmask 255.255.255.0 up
fischer@munich-thinkcentre1:~$ ./tcp_pingpong  -s -l 10.0.0.1
^C
fischer@munich-thinkcentre1:~$ onload ./tcp_pingpong  -s -l 10.0.0.1
oo:tcp_pingpong[401175]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-1)
See kernel messages in dmesg or /var/log/syslog for more details of this failure
oo:tcp_pingpong[401175]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-1)
See kernel messages in dmesg or /var/log/syslog for more details of this failure
^C
fischer@munich-thinkcentre1:~$ dmesg | tail
[ 3340.826170] [onload] [2]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
[ 3340.872019] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
[ 3340.872025] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
[ 3340.872029] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
[ 3340.872030] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
[ 3340.880341] [onload] [3]: WARNING: huge pages are incompatible with AF_XDP. Disabling hugepage support.
[ 3340.893582] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
[ 3340.893587] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
[ 3340.893591] [sfc efhw] af_xdp_flush_rx_dma_channel: FIXME AF_XDP
[ 3340.893592] [sfc efhw] af_xdp_flush_tx_dma_channel: FIXME AF_XDP
```

shirshen12, 2022-06-29:
Check if the driver version is indeed, 5.6.x for Mellanox

`ethtool -i <ifname>`

boyracer2020, 2022-06-29:
yep. it shows version: 5.6-1.0.3 

Note that this also happens with SFC and the error hint refers to sfc efhw

can someone try to reproduce? the same happens on a 2nd system. Ubuntu 20.04 , mlx or sfc does not matter.

boyracer2020, 2022-06-29:
full output:
```
 ethtool -i enp6s0f0np0
driver: mlx5_core
version: 5.6-1.0.3
firmware-version: 16.33.1048 (MT_0000000080)
expansion-rom-version:
bus-info: 0000:06:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes
```

ol-alexandra, 2022-06-29:
> fischer@munich-thinkcentre1:~$ onload ./tcp_pingpong  -s -l 10.0.0.1
> oo:tcp_pingpong[401175]: netif_tcp_helper_alloc_u: ERROR: Failed to allocate stack (rc=-1)

You should be root to use Onload on AF_XDP.  `-1 == -EPERM`.

boyracer2020, 2022-06-29:
magic :) that was the missing part. thanks a bunch. Not seeing this in the docs.

shirshen12, 2022-06-29:
yay!!! @ol-alexandra is back!

shirshen12, 2022-06-29:
Yes you have to be root as of now.

shirshen12, 2022-06-29:
BTW @boyracer2020 any non-trivial mult-threaded app will have trouble running in full performance since Onload-on-AF_XDP lacks transparent binding to multiple queues.

boyracer2020, 2022-06-29:
thanks. since the subject is about AF_XDP on ConnectX-5, do we know how to further reduce latency? some IRQ/Napi settings?

this is what we are seeing in a latency benchmark (round trip / 2):
standard tcp/ip:
```
fischer@ubuntu-thinkcentre2:~$  ./tcp_pingpong  -h 10.0.0.1 -l 10.0.0.2 -E 128 -i 1000000
Bound
Connected
Half Round-Trip latency benchmark [tcp]:
  Size 1:        9.20 us         (mean=11.06 median=11.04, 99%=12.09 max=219.84, std_dev=0.64)
  Size 2:        9.09 us         (mean=11.02 median=11.00, 99%=11.94 max=345.58, std_dev=0.84)
  Size 4:        8.98 us         (mean=11.04 median=11.01, 99%=11.99 max=120.43, std_dev=0.61)
  Size 8:        9.08 us         (mean=11.08 median=11.05, 99%=12.05 max=432.24, std_dev=0.82)
  Size 16:       9.05 us         (mean=11.03 median=11.01, 99%=12.09 max=150.62, std_dev=0.66)
  Size 32:       9.01 us         (mean=11.03 median=11.01, 99%=11.93 max=284.35, std_dev=0.72)
  Size 64:       9.25 us         (mean=11.12 median=11.08, 99%=12.14 max=651.26, std_dev=0.99)
  Size 128:      9.52 us         (mean=11.27 median=11.23, 99%=12.17 max=138.19, std_dev=0.63)
```
and now with AF_XPD enabled:
```
fischer@ubuntu-thinkcentre2:~$ sudo onload --profile=latency ./tcp_pingpong  -h 10.0.0.1 -l 10.0.0.2 -E 128 -i 1000000
oo:tcp_pingpong[2745]: Using Onload   [6]
oo:tcp_pingpong[2745]: Copyright 2019-2022 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
Bound
Connected
Half Round-Trip latency benchmark [tcp]:
  Size 1:        6.06 us         (mean=8.86 median=8.76, 99%=11.10 max=162.18, std_dev=1.20)
  Size 2:        5.94 us         (mean=8.85 median=8.75, 99%=11.11 max=122.13, std_dev=1.20)
  Size 4:        5.94 us         (mean=8.60 median=8.58, 99%=10.81 max=269.04, std_dev=1.14)
  Size 8:        6.02 us         (mean=8.52 median=8.45, 99%=10.74 max=160.91, std_dev=1.09)
  Size 16:       5.97 us         (mean=8.55 median=8.48, 99%=10.78 max=268.14, std_dev=1.12)
  Size 32:       5.92 us         (mean=8.59 median=8.54, 99%=10.80 max=134.33, std_dev=1.10)
  Size 64:       6.43 us         (mean=8.68 median=8.61, 99%=10.88 max=159.15, std_dev=1.11)
  Size 128:      6.30 us         (mean=8.80 median=8.72, 99%=11.21 max=127.79, std_dev=1.21)
  ```
Onload without AF_XDP shows a nice 1.1usecs here.

Q: is there a way to reduce RX / TX operations with AF_XDP ?

maciejj-xilinx, 2022-06-29:
Hi @boyracer2020 ,

So how low latency you could get would be device specific.
There are some standard ways to improve:

sudo ethtool -c <intf> tx-usecs 0 rx-usecs 0

Another item is adjusting interrupt affinity... I'd be vague here as we did most of the work
with throughput in mind in regards to AF_XDP.

P.S.
Thanks @shirshen12 and @ol-alexandra for your support.

closed 2023-03-14 by abower-amd