Xilinx-CNS/onload #94: "insmod: ERROR: could not insert module..... Unknown symbol in module"

cpp77, 2022-08-20:
I get this problem on two different machines (one virtual and one physical), both running Ubuntu 20.04 LTS.

I've followed the instructions to build from source. Then I load the drivers in the kernel using this:

`"$(mmaketool --toppath)/build/$(mmaketool --driverbuild)/driver/linux/load.sh" onload`

I get this error:

```
insmod: ERROR: could not insert module 
onload/build/x86_64_linux-5.15.0-46-generic/driver/linux/sfc_resource.ko: Unknown symbol in module
```

I checked `dmesg | tail` and it had this:

`init_sfc_resource: ERROR: failed to find syscall table`

Not sure if it matters but neither machine had a Solarflare NIC.

Unfortunately really unsure how to proceed with this issue.

rhughes-xilinx, 2022-08-22:
For the syscall table thing, please retest with the freshly-pushed 673463c07a. If that still doesn't work then there's debug logging which you can get at by toggling the `#if` at the top of `syscall_x86.c`; please do that and post the dmesg.

cpp77, 2022-08-23:
Many thanks for that. It fixed the original issue but unfortunately I don't think Onload is working as expected.

I continued to load the drivers in to the kernel:

`"$(mmaketool --toppath)/build/$(mmaketool --driverbuild)/driver/linux/load.sh" onload`

The output was:
```
unload.sh: /sbin/rmmod onload
unload.sh: /sbin/rmmod sfc_char
unload.sh: /sbin/rmmod sfc_resource
unload.sh: /sbin/rmmod sfc
unload.sh: /sbin/rmmod virtual_bus
unload.sh: /sbin/rmmod sfc_driverlink
NET_OPT is    
CHAR_OPT is 
sfc is a DEBUG driver
RESOURCE_OPT is 
CHAR_OPT is 
ONLOAD_OPT is
```
**('sfc is a DEBUG driver' doesn't look good?)**

I then ran my app using:

`<path_to_onload>/scripts/onload <app>`

I'm measuring the performance inside openssl `bss_sock.c`, where it writes to the socket:

```
uint64_t t0 = __rdtsc();
ret = writesocket(b->num, in, inl);
uint64_t t1 = __rdtsc();
```
The socket is non-blocking. **Without** Onload it takes 95,000 cycles. I'm on an old laptop, no proper NIC but I'm hoping to see a speed improvement using the Onload stack instead of the Linux kernel/driver.

**With** Onload it's around 110,000 cycles. This performance degradation is reproducible.

However, at the very beginning of my app starting under Onload I noticed this:

```
oo:my_app[26514]: netif_tcp_helper_alloc_u: ENODEV.
This error can occur if:
 - no Solarflare network interfaces are active/UP, or they are running
   packed stream firmware or are disabled, and
 - there are no AF_XDP interfaces registered with sfc_resource
Please check your configuration.
oo:my_app[26514]: __citp_netif_alloc: failed to construct netif (19)
oo:my_app[26514]: citp_netif_alloc_and_init: failed to create netif (19)
oo:my_app[26514]: citp_tcp_socket: failed (errno:19) - PASSING TO OS
```

rhughes-xilinx, 2022-08-23:
Onload works by talking directly to the NIC. If you do not have a NIC that it supports then figuring that out is going to cost overhead before it falls back to doing exactly what would have happened without Onload. Read the readme for instructions on registering arbitrary NICs for use with Onload, but this requires support be included in the NIC's driver, which not all do.

Also note that with the AF_XDP backend Onload gives you significantly better throughput by amortising the cost of syscalls; the improvement for an individual operation is something, but isn't great.

cpp77, 2022-08-24:
Thanks. Can't say I'm surprised. Without starting another discussion thread may I ask if you know the current status of using Onload within AWS? I'm new to AWS, using Ubuntu 20.04 and I saw you commenting in another discussion about ENA and XDP. Is it now possible to get good improvement with Onload in AWS, or complete pot luck with the NIC on the vm?

rhughes-xilinx, 2022-08-24:
Sorry, it's not something I've ever looked at. If you go for 'enhanced networking' you get to pick the NIC that's in the VM, so pot luck can be avoided, but I don't know anything about tuning for their systems.

abower-amd, 2024-04-16:
The issue as raised is resolved with advice. Closing. Thanks for taking an interest in the technology!

closed 2024-04-16 by abower-amd