Xilinx-CNS/onload #108: When I start two threads to recv and send at the same time, the sending delay will rise？

ASaltyFishBoy, 2022-10-26:
onload --version 201811
built sep 12 2002
kernel module 201811
(I don't think this is the version number, but this command only returns these to me)

sf  x2522

If close recv thread,the send penetration delay will be reduced by 2-300ns
I want to know whether the problem originates from the design of onload or sf card, or from external causes

ol-alexandra, 2022-10-26:
Disclaimers:
1. OpenOnload 201811 predates this git repository.
2. I am not working for Level 5 Networks, Solarflare Communications, Xilinx and/or AMD.  The following is my personal opinion.

The main unit of Onload is an Onload stack.  And an accelerated application performs best if each Onload stack is used from one execution thread.  Usually, one process uses one Onload stack in all its thread.  If you close one thread which uses Onload, then another thread performs better.

If your "sending" and "receiving" threads use the same socket, then my answer is "don't do it".  If the threads are using different sockets, then you can play with `EF_STACK_PER_THREAD` variable to put your sockets into different Onload stacks.  `onload_stackdump all` can help to find out your sockets/stacks layout.

ASaltyFishBoy, 2022-10-27:
> Disclaimers:
>
> 1. OpenOnload 201811 predates this git repository.
> 2. I am not working for Level 5 Networks, Solarflare Communications, Xilinx and/or AMD.  The following is my personal opinion.
>
> The [...]

Thanks,but i can't change our net thread structure.
Now I can only try to use ef_vi directly implements the tcp protocol stack with low data competition.Thank you for your library and answers.