Xilinx-CNS/onload #124: sfc build is broken on CentOS9 5.14.0-207: too many arguments to function ‘netif_napi_add’

ol-sergeyn, 2022-12-20:
CentOS 9 with `5.14.0-207.rt14.208.el9.x86_64`

Build command:
```
mmakebuildtree --driver
make -C build/x86_64_linux-5.14.0-207.rt14.208.el9.x86_64/
```
Output:
```
...
  CC [M]  /home/nikitins/work/v5/build/x86_64_linux-5.14.0-207.rt14.208.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/efx_channels.o
/home/nikitins/work/v5/build/x86_64_linux-5.14.0-207.rt14.208.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/efx_channels.c: In function ‘efx_init_napi_channel’:
/home/nikitins/work/v5/build/x86_64_linux-5.14.0-207.rt14.208.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/efx_channels.c:2267:9: error: too many arguments to function ‘netif_napi_add’
 2267 |         netif_napi_add(channel->napi_dev, &channel->napi_str,
      |         ^~~~~~~~~~~~~~
In file included from /home/nikitins/work/v5/build/x86_64_linux-5.14.0-207.rt14.208.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/net_driver.h:24,
                 from /home/nikitins/work/v5/build/x86_64_linux-5.14.0-207.rt14.208.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/efx_channels.c:10:
./include/linux/netdevice.h:2514:1: note: declared here
 2514 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
```

That comes from Linux 6.1
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=b48b89f9c189d24eb5e2b4a0ac067da5a24ee86d

ol-alexandra, 2023-02-20:
Alternatively, if you are using non-SFC NIC, than you can use `make HAVE_SFC=0` and avoid building `sfc.ko` at all.

ivatet-amd, 2023-03-20:
@ol-sergeyn We updated the SFC driver in 1215b3a0e88606465a7bc86db287d06e53620fcf last week, and it must address the mentioned build time error. I'm build-testing Onload in Ubuntu 22.04 with the kernel 6.1.19 from the mainline PPA https://kernel.ubuntu.com/~kernel-ppa/mainline/v6.1.19/amd64, where it builds and runs OK. In other words, the latest `master` should work. 👍

I've also tried to build Onload from dc8773c7d57d4bcf89ff03c35340cbff4e878ab1 on the same setup, and it fails the same way as yours. This sounds convincing enough to me to close the issue. However, please reopen if it's still the issue on your setup.

closed 2023-03-20 by ivatet-amd