Xilinx-CNS/onload #125: Closing the wrong socket returns errno as a result and sets errno to zero

ol-damirm, 2023-01-05:
Here is a C-program for demonstration:
```c
int
main(int argc, char *argv[])
{
    int s = -1; 
    int ret = close(s);
    printf("close(%d) -> %d, errno: %d (%s)\n", s, ret, errno, strerror(errno));
    exit(0);
}
```

The results:
```shell
# Pure Linux:
$ ./bad-socket 
close(-1) -> -1, errno: 9 (Bad file descriptor)

# Under Onload:
$ onload ./bad-socket 
close(-1) -> -9, errno: 0 (Success)
```
This behavior started with commit 15aa41eb6f "ON-14454: Remove close syscall ioctl.".

@ol-yurijp I see that the `sockapi-ts/level5/fd_caching/fd_cache_shutdown_reuse` test fails because of the same commit.

tcrawley-xilinx, 2023-04-28:
Hi @ol-damirm
This should be resolved with 2317118053769fe2338e60a9560e9e99560bcfc6

If I repeat your testing I get:
```
$ ./bad-socket
close(-1) -> -1, errno: 9 (Bad file descriptor)

$ onload ./bad-socket
close(-1) -> -1, errno: 9 (Bad file descriptor)
```
I'm going to close this issue now, if you see this issue again could you re-open the issue.

closed 2023-04-28 by tcrawley-xilinx

ol-damirm, 2023-05-05:
@tcrawley-xilinx Thank you, I do not see this issue anymore.