Xilinx-CNS/onload #128: Build onload on kernel > 6.0

YungeG, 2023-02-20:
lastest onload build will fail on kernel > 6.0, prompting 
`No rule to make target 'vmlinux.o', needed by 'modpost'.`

ol-alexandra, 2023-02-20:
It is very strange, `make HAVE_SFC=0` works for us with linux-6.0 and linux-6.1.  @ol-sergeyn do you remember such an issue?

ol-alexandra, 2023-02-20:
If I recall correctly, than it is an issue with some Linux kernels and with some Linux distributions.  And you get this message when you try to build any module, onload or not.

ol-sergeyn, 2023-02-20:
> It is very strange, `make HAVE_SFC=0` works for us with linux-6.0 and linux-6.1. @ol-sergeyn do you remember such an issue?

Yes, we've fixed it once in #114

YungeG, 2023-02-21:
Thank you very much !

ivatet-amd, 2023-03-20:
@YungeG Is there still an issue in your setup? I believe Onload should build on 6.1 after the recent SFC driver update.

ivatet-amd, 2023-03-24:
Assuming it is a no, closing. Please re-open if the issue is still the case.

closed 2023-03-24 by ivatet-amd

YungeG, 2023-04-02:
No Problem anymore, after modifying Makefile according to https://github.com/Xilinx-CNS/onload/issues/114 .
Sorry for late response .