Xilinx-CNS/onload #129: Linux-6.1: HugeTLB: vma_lock already exists in hugetlb_vm_op_open

ol-sergeyn, 2023-02-24:
Linux prints this warning when creating an onload stack (non-AF_XDP):
`HugeTLB: vma_lock already exists in hugetlb_vm_op_open.`

How to reproduce:
- system: debian-testing with `6.1.0-2-amd64` kernel
- build and load Onload `mmakebuildtree && make && load.sh onload`
- run `onload nc -l 6666`

NB: sfc netdriver build is broken due to #124, so you need a patch to successfully build it and reproduce the issue:
```diff
diff --git a/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.h b/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.h
index a58c7f2e7f..e55a3d17a5 100644
--- a/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.h
+++ b/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.h
@@ -2681,4 +2681,9 @@ typedef atomic_t refcount_t;
 #define refcount_dec_and_test	atomic_dec_and_test
 #endif
 
+#ifdef EFX_HAVE_NAPI_ADD_WEIGHT
+/* linux >= 6.1 */
+#define netif_napi_add netif_napi_add_weight
+#endif
+
 #endif /* EFX_KERNEL_COMPAT_H */
diff --git a/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.sh b/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.sh
index 2e0eb7472d..49b2ba6e71 100755
--- a/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.sh
+++ b/src/driver/linux_net/drivers/net/ethernet/sfc/kernel_compat.sh
@@ -180,6 +180,7 @@ EFX_HAVE_SKB_FRAG_TRUESIZE		symtype	skb_add_rx_frag		include/linux/skbuff.h	void
 EFX_HAVE_INNER_IP_HDR			symbol	inner_ip_hdr		include/linux/ip.h
 EFX_HAVE_INNER_TCP_HDR			symbol	inner_tcp_hdr		include/linux/tcp.h
 EFX_HAVE_INDIRECT_CALL_WRAPPERS		file				include/linux/indirect_call_wrapper.h
+EFX_HAVE_NAPI_ADD_WEIGHT		symbol	netif_napi_add_weight	include/linux/netdevice.h
 
 # Stuff needed in code other than the linux net driver
 EFX_HAVE_NEW_KFIFO			symbol kfifo_out		include/linux/kfifo.h

```

ivatet-amd, 2023-03-20:
Thanks, @ol-sergeyn. SFC should be OK on 6.1 with the latest `master` similar to https://github.com/Xilinx-CNS/onload/issues/124. Please reopen if there's still a problem with Onload in your setup.

ivatet-amd, 2023-03-20:
Re-opening because I do see this issue in testing:
```
< > 16:23:05 DUT [  235.834009] HugeTLB: vma_lock already exists in hugetlb_vm_op_open.
```
Interestingly, you have bisected the kernel to the same hash as in https://github.com/Xilinx-CNS/onload/issues/131. I'll take a closer look this week.

ol-sergeyn, 2023-03-21:
I've rechecked `master` 54958ab612, and the issue still exists.

> Interestingly, you have bisected the kernel to the same hash as in https://github.com/Xilinx-CNS/onload/issues/131

Yes, that Linux commit causes #131 and this one.

ivatet-amd, 2023-04-24:
Hello @ol-sergeyn, I hope this and https://github.com/Xilinx-CNS/onload/issues/131 are no longer the case after the merge of https://github.com/Xilinx-CNS/onload/pull/138 today. The latest Onload doesn't operate on VMAs and must resolve both issues. May I ask you to try verifying at your convenience?

ol-sergeyn, 2023-04-24:
Hello @ivatet-amd! Neither this issue nor #131 are present after your fix.
Thanks!

closed 2023-04-25 by ivatet-amd