Xilinx-CNS/onload #131: 6.1.8-100.fc36: kernel BUG at mm/hugetlb.c:1123

ol-sergeyn, 2023-03-09:
Onload causes kernel BUG with call trace on Fedora with Linux 6.1:
```
[  391.824180] vma ffff9cf544af3af0 start 0000000000000000 end 0000000000000000 mm 0000000000000000                                                                                                                
[  391.824180] prot 0 anon_vma 0000000000000000 vm_ops 0000000000000000                                                                                                                                            
[  391.824180] pgoff 0 file ffff8a794a9f8e00 private_data 0000000000000000                                                                                                                                         
[  391.824180] flags: 0x0()                         
[  391.851261] ------------[ cut here ]------------                                                      
[  391.856419] kernel BUG at mm/hugetlb.c:1123!                                                          
[  391.861198] invalid opcode: 0000 [#1] PREEMPT SMP PTI                                                 
[  391.866834] CPU: 2 PID: 40582 Comm: nc Tainted: G           OE      6.1.8-100.fc36.x86_64 #1                                                                                                                    
[  391.876253] Hardware name: Dell Inc. PowerEdge R230/0FRVY0, BIOS 2.3.2 11/16/2017                                                                                                                               
[  391.884605] RIP: 0010:hugetlb_vm_op_close+0x174/0x1b0                                                 
[  391.890245] Code: 8b 80 d8 00 00 00 48 8b 00 4c 8b b0 30 02 00 00 e9 f8 fe ff ff 48 f7 de 48 89 df e8 c6 fc ff ff eb b24                                                                                        
[  391.911201] RSP: 0018:ffff9cf544af3a98 EFLAGS: 00010296                                               
[  391.917031] RAX: 00000000000000d6 RBX: ffffffff90cdf7e8 RCX: 0000000000000000                                                                                                                                   
[  391.924995] RDX: 0000000000000001 RSI: ffffffff8f74befb RDI: 00000000ffffffff                                                                                                                                   
[  391.932957] RBP: 0000000000000000 R08: 0000000000000000 R09: ffff9cf544af3908                                                                                                                                   
[  391.940919] R10: 0000000000000003 R11: ffffffff901465a8 R12: ffff9cf544af3af0                                                                                                                                   
[  391.948882] R13: 0000000000000002 R14: ffff8a79acfb0000 R15: ffff8a79acfb0d30                                                                                                                                   
[  391.956843] FS:  0000000000000000(0000) GS:ffff8a7a67d00000(0000) knlGS:0000000000000000                                                                                                                        
[  391.965874] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033                                                                                                                                                   
[  391.972286] CR2: 00007fb1e80141e8 CR3: 000000013a77c003 CR4: 00000000003706e0                                                                                                                                   
[  391.980248] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000                                                                                                                                   
[  391.988210] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400                                                                                                                                   
[  391.996175] Call Trace:                          
[  391.998903]  <TASK>                              
[  392.001245]  shm_close+0x20/0x30                                                                      
[  392.004846]  oo_bufpage_huge_free+0x59/0x110 [onload]                                                 
[  392.010512]  ? release_pkts+0x10a/0x210 [onload]                                                      
[  392.015686]  ? tcp_helper_dtor+0x1c9/0x990 [onload]                                                   
[  392.021149]  ? thr_release_base+0x1fd/0x660 [onload]                                                  
[  392.026709]  ? oo_fop_release+0xb7/0x160 [onload]                                                     
[  392.031977]  ? __fput+0x8e/0x250                                                                      
[  392.035570]  ? task_work_run+0x56/0x90                                                                
[  392.039753]  ? do_exit+0x338/0xb00                                                                    
[  392.043549]  ? kmem_cache_free+0x181/0x3b0                                                            
[  392.048121]  ? do_group_exit+0x2d/0x80                                                                
[  392.052305]  ? get_signal+0x90f/0x950                                                                 
[  392.056393]  ? arch_do_signal_or_restart+0x44/0x770                                                   
[  392.061839]  ? exit_to_user_mode_prepare+0x117/0x1f0                                                  
[  392.067380]  ? syscall_exit_to_user_mode+0x17/0x40                                                    
[  392.072728]  ? do_syscall_64+0x67/0x80                                                                
[  392.076912]  ? do_send_specific+0x8a/0xa0                                                             
[  392.081388]  ? do_tkill+0x95/0xc0                                                                     
[  392.085077]  ? syscall_exit_to_user_mode+0x17/0x40                                                    
[  392.090424]  ? do_syscall_64+0x67/0x80                                                                
[  392.094606]  ? exc_page_fault+0x70/0x170                                                              
[  392.098983]  ? entry_SYSCALL_64_after_hwframe+0x63/0xcd                                               
[  392.104818]  </TASK>     
```
It can be reproduced easily just with:
```bash
# Run netcat with onload
$ onload nc -l 6666
oo:ncat[40582]: Using Onload 6c55327703 2023-03-06 github-master  [0]
oo:ncat[40582]: Copyright 2019-2023 Xilinx, 2006-2019 Solarflare Communications, 2002-2005 Level 5 Networks
# Type CTRL-C to close the socket
^C
```

I've bisected Linux:
```
b6305049f30652f1efcf78d627fc6656151a7929 is the first bad commit
commit b6305049f30652f1efcf78d627fc6656151a7929
Author: Mike Kravetz <mike.kravetz@oracle.com>
Date:   Mon Nov 14 13:00:18 2022 -0800

    ipc/shm: call underlying open/close vm_ops

    Shared memory segments can be created that are backed by hugetlb pages.
[... 25 lines cut ...]
    Cc: Manfred Spraul <manfred@colorfullife.com>
    Cc: Matthew Wilcox <willy@infradead.org>
    Cc: Miaohe Lin <linmiaohe@huawei.com>
    Cc: Michal Hocko <mhocko@suse.com>
    Signed-off-by: Andrew Morton <akpm@linux-foundation.org>

 ipc/shm.c | 34 +++++++++++++++++++++++++---------
 1 file changed, 25 insertions(+), 9 deletions(-)
```
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=b6305049f30652f1efcf78d627fc6656151a7929

This is a fix to make kernel not to panic:
```diff
diff --git a/src/lib/efthrm/iobufset.c b/src/lib/efthrm/iobufset.c
index 1b03e0c99f..d1d25470ad 100644
--- a/src/lib/efthrm/iobufset.c
+++ b/src/lib/efthrm/iobufset.c
@@ -210,6 +210,7 @@ static void oo_bufpage_huge_free(struct oo_buffer_pages *p)
    * assumptions about this interface more apparent. */
   memset(&vma, 0, sizeof(vma));
   vma.vm_file = p->shm_map_file;
+  vma.vm_flags = VM_HUGETLB;
   p->close(&vma);
   fput(p->shm_map_file);
```

ol-sergeyn, 2023-03-21:
The issue still exists on `master` https://github.com/Xilinx-CNS/onload/commit/54958ab61230c2624a3ffa329fa8aba06d5061c2

ivatet-amd, 2023-03-21:
Thanks for checking that promptly.