Xilinx-CNS/onload #132: onload-8.0.2.51 does not build Centos 9 Stream with 5.14.0-285.el9.x86_64

Ziemien, 2023-03-15:
Kernel: 5.14.0-285.el9.x86_64
Onload: onload-8.0.2.51 

```
ripts/Makefile.build:43: /root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/siena/Makefile: No such file or directory
make[9]: *** No rule to make target '/root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/siena/Makefile'.  Stop.
make[8]: *** [scripts/Makefile.build:607: /root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/siena] Error 2
make[8]: *** Waiting for unfinished jobs....
  UPD     /root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/autocompat.h
make[7]: *** [Makefile:1923: /root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc] Error 2
make[7]: Leaving directory '/usr/src/kernels/5.14.0-285.el9.x86_64'
make[6]: *** [/root/rpmbuild/BUILD/onload-8.0.2.51/src/driver/linux_net/drivers/net/ethernet/sfc/Makefile:274: modules] Error 2
make[6]: Leaving directory '/root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc'
make[5]: *** [/root/rpmbuild/BUILD/onload-8.0.2.51/src/driver/linux_net/drivers/net/ethernet/mmake.mk:8: all] Error 2
make[5]: Leaving directory '/root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net/ethernet'
make[4]: *** [/root/rpmbuild/BUILD/onload-8.0.2.51/src/driver/linux_net/drivers/net/mmake.mk:8: all] Error 2
make[4]: Leaving directory '/root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net'
make[3]: *** [/root/rpmbuild/BUILD/onload-8.0.2.51/src/driver/linux_net/drivers/mmake.mk:8: all] Error 2
make[3]: Leaving directory '/root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers'
make[2]: *** [/root/rpmbuild/BUILD/onload-8.0.2.51/src/driver/linux_net/mmake.mk:8: all] Error 2
make[2]: Leaving directory '/root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net'
make[1]: *** [/root/rpmbuild/BUILD/onload-8.0.2.51/src/driver/mmake.mk:19: all] Error 2
make[1]: Leaving directory '/root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver'
make: *** [/root/rpmbuild/BUILD/onload-8.0.2.51/src/mmake.mk:25: all] Error 2
make: Leaving directory '/root/rpmbuild/BUILD/onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64'
```
the missing file in siena dir, `onload-8.0.2.51/build/x86_64_linux-5.14.0-285.el9.x86_64/driver/linux_net/drivers/net/ethernet/sfc/siena/` is empty.

Ziemien, 2023-03-15:
the master does compile, commit 1c45de39d43c682ee85c489d2ed624fb17fa653b, however the onload segfaults when I am trying to run any command i .e

```
$ onload echo 123
Segmentation fault (core dumped)
```

abower-amd, 2023-03-20:
Thanks for raising this @Ziemien - I assume you are happy with this issue now - closing!

closed 2023-03-20 by abower-amd