Xilinx-CNS/onload #135: application running with onload crashes

hdlee, 2023-04-06:
i'm running Alma 9.1 and onload 8.0.2.51. (i also have run Centos stream 9 and onload-master, problem is the same)

after successfully build and install onload kernel modules, i just ran a simple command like: onload -p latency echo "abc", and it crashes:

onload: onload_import: latency (/usr/libexec/onload/profiles/latency.opf)
onload: onload_set: EF_POLL_USEC=100000
onload: onload_set: EF_TCP_FASTSTART_INIT=0
onload: onload_set: EF_TCP_FASTSTART_IDLE=0
onload: onload_import-done: latency
onload: WARNING: EF_TCP_SYNRECV_MAX=16384 and EF_MAX_ENDPOINTS=8192 are inconsistent.
onload: EF_TCP_SYNRECV_MAX is set to 16384 based on /proc/sys/net/ipv4/tcp_max_syn_backlog value and assuming up to 4 listening sockets in the Onload stack
onload: Too few endpoints requested: ~4 syn-receive states consume one endpoint. 
onload: WARNING: EF_TCP_SYNRECV_MAX=16384 and EF_MAX_ENDPOINTS=8192 are inconsistent.
onload: EF_TCP_SYNRECV_MAX is set to 16384 based on /proc/sys/net/ipv4/tcp_max_syn_backlog value and assuming up to 4 listening sockets in the Onload stack
onload: Too few endpoints requested: ~4 syn-receive states consume one endpoint. 
oo:env[49532]: CRITICAL: mprotect(glibc exec) = 13. Process will likely crash now
oo:bash[49533]: CRITICAL: mprotect(glibc exec) = 13. Process will likely crash now
onload: WARNING: EF_TCP_SYNRECV_MAX=16384 and EF_MAX_ENDPOINTS=8192 are inconsistent.
onload: EF_TCP_SYNRECV_MAX is set to 16384 based on /proc/sys/net/ipv4/tcp_max_syn_backlog value and assuming up to 4 listening sockets in the Onload stack
onload: Too few endpoints requested: ~4 syn-receive states consume one endpoint. 
oo:sed[49534]: CRITICAL: mprotect(glibc exec) = 13. Process will likely crash now
/usr/bin/onload: line 79: 49532 Segmentation fault      (core dumped) env
     49533                       (core dumped) | egrep "^EF_|^LD_PRELOAD=|^LD_LIBRARY_PATH=|^LD_PRELOAD_64="
     49534                       (core dumped) | sed "s/^/$me: env: /"
onload: invoking: echo ab
onload: WARNING: EF_TCP_SYNRECV_MAX=16384 and EF_MAX_ENDPOINTS=8192 are inconsistent.
onload: EF_TCP_SYNRECV_MAX is set to 16384 based on /proc/sys/net/ipv4/tcp_max_syn_backlog value and assuming up to 4 listening sockets in the Onload stack
onload: Too few endpoints requested: ~4 syn-receive states consume one endpoint. 
oo:echo[49509]: CRITICAL: mprotect(glibc exec) = 13. Process will likely crash now
Segmentation fault (core dumped)

i've looked into the bash script ,and it is weird the "env" returns "Segmentation fault      (core dumped) env"

hdlee, 2023-04-06:
the problem is inside the libonload.so, even i export LD_PRELOAD=libonload.so manually, it also crashes.

however if i set export LD_PRELOAD=libonload which is obviously incorrect. the output:
[root@localhost ~]# env
ERROR: ld.so: object 'libonload' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.

ld.so always try to load $LD_PRELOAD and therefore it seems libonload.so is the reason of the crash.

msherman13, 2023-07-06:
+1 having the same issue built from master branch

jbondpre-amd, 2023-07-07:
re @hdlee 
This is caused by SELinux in enforcing mode, namely the `execmod` permission. We write to some parts of glibc code in memory at runtime, but this SELinux permission denies `mprotect()`ing the memory back to an executable region.
If you don't need SELinux anyway, you can disable it and the crash will go away.

@msherman13 are you experiencing the same error at start up, or a different one? What OS are you using?

msherman13, 2023-07-07:
This fixes it thank you. Is this documented somewhere?