Xilinx-CNS/onload #144: Installing/using OpenOnload on Amazon Linux 2023?

jmgpeeters, 2023-05-04:
I am trying to build from source, using master, on Amazon Linux 2023 with kernel 6.1.25-37.47.amzn2023.x86_64. Has anyone tried this, or is there a better way of getting OpenOnload to work on these machines, e.g. through the tarballs?

I'm stuck at

`make -C "$(mmaketool --toppath)/build/$(mmaketool --driverbuild)"
...
ERROR: modpost: "pps_register_source" [/work/onload-master/build/x86_64_linux-6.1.25-37.47.amzn2023.x86_64/driver/linux_net/drivers/net/ethernet/sfc/sfc.ko] undefined!
ERROR: modpost: "pps_event" [/work/onload-master/build/x86_64_linux-6.1.25-37.47.amzn2023.x86_64/driver/linux_net/drivers/net/ethernet/sfc/sfc.ko] undefined!
ERROR: modpost: "pps_unregister_source" [/work/onload-master/build/x86_64_linux-6.1.25-37.47.amzn2023.x86_64/driver/linux_net/drivers/net/ethernet/sfc/sfc.ko] undefined!
make[8]: *** [scripts/Makefile.modpost:126: /work/onload-master/build/x86_64_linux-6.1.25-37.47.amzn2023.x86_64/driver/linux_net/drivers/net/ethernet/sfc/Module.symvers] Error 1
`
On this kernel it seems CONFIG_PPS is unset. Not sure to what extent PPS is a requirement for OpenOnload to work, but other people seem to have gotten OpenOnload running on AWS EC2, so I presume there is a way.

jmgpeeters, 2023-05-05:
I downloaded a zipped up master from github yesterday, which corresponds to 097d57b9a12ba32bf1726031fa74f0c793601dc1.

Cheers!
-J

abower-amd, 2023-05-05:
Thanks @jmgpeeters.

This is not something we have seen but you are welcome to see if you can help fix it! Onload includes a version of the `sfc` net driver for relevant (Solarflare) AMD Ethernet adapters under `src/driver/linux_net`. It is possible that there is an error either (1) in the kernel compat code for the `sfc` driver build or (2) in the way that the Onload build process initiates the build of the net driver component.

On the other hand if the underlying network adapter does not require this particular driver, then you can see these instructions for building without the `sfc` driver: https://github.com/Xilinx-CNS/onload#building-without-xilinx-nics-for-af_xdp-only .
The official OpenOnload release packages would not be suitable in your case because (a) these do not yet support kernel 6.1 and (b) these do not support AF_XDP.

moscovium115, 2024-03-21:
@jmgpeeters were you able to get onload working on amazon linux?