Xilinx-CNS/onload #149: rhel/rocky 9.2 - onload 8.0.2.51 (v.40) - can't install

Ziemien, 2023-06-15:
https://github.com/Xilinx-CNS/onload/issues/132

Recently RHEL 9.1 got an update - the latest kernel 5.14.0-284.11.1.el9_2.x86_64 breaks with installing the rpms package. The same problem as linked issue 132 - siena dir is empty.

Is there a new official release that will support this kernel?

alexiusflavius, 2023-07-07:
Can we please have an updated version of onload(tar/dkms) that would work on RHEL/RL9.2? It's been almost 2 month  since 9.2 is out
Building master is great and it's working - but official package is needed for multiple reasons

abower-amd, 2023-07-14:
The OpenOnload-8.1.0.15 release is now published. Thanks for waiting, @Ziemien and others!

closed 2023-07-14 by abower-amd