Xilinx-CNS/onload #158: writel aliasing choice

gitrj95, 2023-06-30:
hi, isn't this ub? i've found a few analogous instances throughout the code.

eg in src/lib/ciul/sysdep_unix.h
```
ef_vi_inline void __raw_writel(uint32_t data, volatile void *addr)
{
  *((volatile uint32_t *) addr) = data;
}
```
was the above a conscious choice over, say, memcpy? are you just relying on volatile, here?

gitrj95, 2023-06-30:
tracing through a code path, i noticed that addr may be a char*?

gitrj95, 2023-06-30:
ie is this intentional, "safe" ub?

rhughes-xilinx, 2023-07-04:
This function is for writing to a hardware device, i.e. the original intended use for `volatile`. See https://stackoverflow.com/questions/72552/why-does-volatile-exist (and other Google results) for an introduction. It's therefore defined behaviour.

memcpy wouldn't be correct because it doesn't guarantee that the write is performed as a single operation (as opposed to as 4 1-byte writes, for example). The hardware spec for the devices we are talking to requires that the write arrive as a single transfer over PCIe where this function is used (this requirement saves transistors, and hence power).

closed 2023-07-04 by rhughes-xilinx

gitrj95, 2023-07-04:
also, why is going via uint32_t guaranteed to be a single write? isn't this platform dependent?