Xilinx-CNS/onload #162: Unloading kernel modules causes the message kmemleak: 858 new suspected memory leaks (see /sys/kernel/debug/kmemleak)

ol-damirm, 2023-07-26:
Unloading kernel modules causes the message `kmemleak: 858 new suspected memory leaks (see /sys/kernel/debug/kmemleak)`
I see this message on the kernel with KMEMLEAK enabled in the kernel config.

How to reproduce:
```shell
$ git clone https://github.com/Xilinx-CNS/onload
$ cd onload
$ ./scripts/onload_build --debug --kernel
$ cd build/x86_64_linux-6.1.31-1-debug-amd64/driver/linux
$ sudo ./load.sh
$ sudo ./unload.sh
$ sleep 15m
# from console: [ 2468.287909][ T4562] kmemleak: 858 new suspected memory leaks (see /sys/kernel/debug/kmemleak)
$ sudo cat /sys/kernel/debug/kmemleak > kmemleak-onload-ba03aac0.txt
```

I see it after the commit ba03aac044 "onload_mktest_net_driver: Replacing net driver with v5_3_14_1016" on:
- Distro: Debian-12
- Linux 6.1.31-1-debug-amd64
- NIC: Solarflare Communications SFC9120 10G Ethernet Controller

I attached the output from the last command from the reproducer: [kmemleak-onload-ba03aac0.txt](https://github.com/Xilinx-CNS/onload/files/12171858/kmemleak-onload-ba03aac0.txt)

ol-alexandra, 2023-07-26:
[linux-6.1.31-debug-config.txt](https://github.com/Xilinx-CNS/onload/files/12171891/linux-6.1.31-debug-config.txt)
This is the config file used to build the kernel.

rhughes-xilinx, 2023-07-26:
I'm assuming those undecoded symbols in the backtraces are from within our drivers. Can you grab a copy of /proc/kallsyms while the driver is loaded so that we can decode the backtraces retroactively?

ol-damirm, 2023-07-26:
@rhughes-xilinx Yes, here are the files:
[proc-kallsyms.txt](https://github.com/Xilinx-CNS/onload/files/12173045/proc-kallsyms.txt)
[kmemleak.txt](https://github.com/Xilinx-CNS/onload/files/12173059/kmemleak.txt)