Xilinx-CNS/onload #164: Missing symbols from the built loadable kernel module for `sfc_resource`

bergentruckung, 2023-08-07:
Hello,

We're trying to test out onload 8.1.0.15 by building from source on RHEL 8.8 (kernel version: 4.18.0-477.15.1) and when we try to build with `onload_build --kernelver $(uname -r)`, the build process completes, but we see warnings about missing symbols:

```
WARNING: "efrm_syscall_table" [/var/tmp/onload-8.1.0.15/build/x86_64_linux-4.18.0-477.15.1.el8_8.x86_64/driver/linux_onload/onload.ko] undefined!
```
... if we try to insert the newly built kernel module for `sfc_resource`, it fails with:

```
insmod: ERROR: could not insert module build/x86_64_linux-4.18.0-477.15.1.el8_8.x86_64/driver/linux/sfc_resource.ko: Unknown symbol in module
```
Since `sfc_resource` is a dependency for `onload`, `onload` doesn't work either.

What should we do to proceed?

abower-amd, 2023-08-07:
Hi @bergentruckung - so far as I can tell our test for this kernel builds and installs the modules fine. Could you raise a support ticket for this, please? Could you also try building from the latest tree here on github?

bdellegrazie, 2024-04-15:
Hi,

I'm also experiencing this with the latest Debian Stable kernel - 6.1.0.20 and latest openonload (v8.1.2.26) - using DKMS:
Output from systemd-modules-load:
```
Apr 14 15:34:50 hydra0 systemd-modules-load[530]: modprobe: ERROR: ../libkmod/libkmod-module.c:895 kmod_module_insert_module() could not find module by name='onload'
Apr 14 15:34:50 hydra0 systemd-modules-load[530]: modprobe: ERROR: could not insert 'onload': Unknown symbol in module, or unknown parameter (see dmesg)
```
dmesg output is like this (the first part shows the OS bundled driver unloading):
```
Apr 14 07:22:18 hydra0 kernel: Solarflare NET driver unloading
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_mcfw:2a":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_exp_rom:07":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_uefi:00":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_exp_rom_cfg:00":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_dynamic_cfg:00":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_license:00":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_mumfw:0a":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_dynamic_cfg_dflt:00":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_exp_rom_cfg_dflt:00":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_bundle_metadata:00":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_bundle:07":
Apr 14 07:22:18 hydra0 kernel: Deleting MTD partitions on "ens3f0np0 sfc_status:00":
Apr 14 07:22:19 hydra0 kernel: sfc_driverlink: loading out-of-tree module taints kernel.
Apr 14 07:22:19 hydra0 kernel: sfc_driverlink: module verification failed: signature and/or required key missing - tainting kernel
Apr 14 07:22:19 hydra0 kernel: Solarflare driverlink driver v5.3.16.1004 API v33.1
Apr 14 07:22:19 hydra0 kernel: Solarflare NET driver v5.3.16.1004
Apr 14 07:22:19 hydra0 kernel: sfc 0000:65:00.0: Solarflare NIC detected: device 1924:0b03 subsys 1924:8022
Apr 14 07:22:19 hydra0 kernel: pps pps0: new PPS source ptp2
Apr 14 07:22:19 hydra0 kernel: sfc 0000:65:00.0 (unnamed net_device) (uninitialized): MC command 0xb inlen 12 failed rc=-114 (raw=114) arg=0
Apr 14 07:22:19 hydra0 kernel: sfc 0000:65:00.0 eth0: driverlink registering nic
Apr 14 07:22:19 hydra0 kernel: sfc 0000:65:00.1: Solarflare NIC detected: device 1924:0b03 subsys 1924:8022
Apr 14 07:22:19 hydra0 kernel: sfc 0000:65:00.0 ens3f0: renamed from eth0
Apr 14 07:22:19 hydra0 kernel: sfc 0000:65:00.1 eth1: driverlink registering nic
Apr 14 07:22:19 hydra0 kernel: sfc 0000:65:00.1 ens3f1: renamed from eth1
Apr 14 07:22:19 hydra0 kernel: [sfc efrm] init_sfc_resource: ERROR: failed to find syscall table
```

abower-amd, 2024-04-26:
The original issue is unconfirmed; the second one is fixed by a9d697b42d3563ce53787159f420c0ae1138a062.

closed 2024-04-26 by abower-amd