Xilinx-CNS/onload #173: assertion in splice()

ol-alexandra, 2023-09-21:
I'm using `splice()` in following way: OS socket -> OS pipe -> Onload socket.  Sometimes I see following assertion:
```
[11220.108079] [onload] ci_assert((offset + size)<=(4096))
[11220.114533] [onload] where [offset + size=100c]
[11220.120205] [onload] and [4096=1000] 
[11220.124798] [onload] at /var/tmp/session_2022.05.07/onload-6840772e686a4c345e662596ff14e43d98cdb759-5.14.0-70.el9.x86_64-debug/build/x86_64_linux-5.14.0-70.el9.x86_64/driver/linux_onload/tcp_sendpage.c:57 
[11220.146305] [onload] FAIL at /var/tmp/session_2022.05.07/onload-6840772e686a4c345e662596ff14e43d98cdb759-5.14.0-70.el9.x86_64-debug/build/x86_64_linux-5.14.0-70.el9.x86_64/driver/linux_onload/tcp_sendpage.c:57 
[11220.168316] [onload] from /var/tmp/session_2022.05.07/onload-6840772e686a4c345e662596ff14e43d98cdb759-5.14.0-70.el9.x86_64-debug/build/x86_64_linux-5.14.0-70.el9.x86_64/driver/linux_onload/tcp_sendpage.c:57
```
I see this issue with linux>=6.1 and with recent CentOS kernels.
Everything works with NDEBUG build, or with DEBUG build when I remove this assertion.

I have an explanation.  Modern network drivers (including SFC) use compound pages as receive buffers; these big pages are split into smaller buffers of MTU size (mtu + some headers, etc ).  As a result `skb->data` of a received packet is continuous, but it may cross the bounds or an ordinary page.  This chunk of data is passed to OS pipe, and then to `linux_tcp_helper_fop_sendpage()`, where the assertion is fired.

I'll propose a patch removing the guilty assertion.

Please feel free to comment my considerations above.

ol-alexandra, 2023-09-25:
It was seen with 5.4.0-155-lowlatency from Ubuntu 20.04.  We have not seen this with linux-5.4 before.  I assume there is some "security fix" which is getting added to all the kernels.

ivatet-amd, 2023-12-04:
Closing as the PR is merged. Thanks.

closed 2023-12-04 by ivatet-amd