Xilinx-CNS/onload #184: onload-8.1.1 installation failing on Arch Linux

sshukla23, 2023-11-08:
I am unable to install onload version 8.1.1 in my arch linux system. 
While runnning `./onload_install` I am facing the error 
`/tmp/onload-8.1.1/build/x86_64_linux-6.5.9-arch2-1/driver/linux_resource/af_xdp.c:100:8: error: too many arguments to function ‘pin_user_pages’`
  `100 |   rc = pin_user_pages(area->user_addr, 1, FOLL_WRITE, &area->page, NULL);` as shown in the image 
[image]

My kernel version is `6.5.9-arch2-1`

ivatet-amd, 2023-11-08:
Thank you, @sshukla23. I believe @ol-sergeyn has addressed it in https://github.com/Xilinx-CNS/onload/pull/169, and we need to merge it soon.

@sshukla23 Meanwhile, could you give it a go and see if it addresses this Arch Linux issue?

ivatet-amd, 2023-11-15:
We merged the above PR, and I'm tempted to close this issue, but please re-open it if the build time error persists.