Xilinx-CNS/onload #185: Encryption with SolarFlare's TCPDirect

prashant-nhbt, 2023-11-10:
While doing connections with standard non-TCPDirect sockets, it's quite simple to load the CA certificate and exchange data with SSL_read and SSL_write using the <openssl/ssl.h> module in C++. On trying to do something similar with TCPDirect, I couldn't find any help from their documentation or any source on the internet. Please help me with this :(

pkbcell, 2024-01-09:
I am also facing same issue.  If someone solution regarding same. Please let me know. Thanks

rhughes-xilinx, 2024-01-09:
Sorry, this repository only contains ef_vi and OpenOnload - we can't offer help with TCPDirect. You can e-mail support-nic@amd.com to get support for proprietary libraries. If this is your first time doing this, you should receive an automated email with the details to sign up.

I can suggest, however, that you look in to writing a custom BIO for OpenSSL, which is their abstraction over the sockets API. Maybe start reading at https://www.openssl.org/docs/man3.2/man3/BIO_new_bio_pair.html . The Internet doesn't seem replete with examples of how to do this, but maybe your search-fu is better than mine.